espressif/arduino-esp32 #2060: BLEScan fails after esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)

czuvich, 2018-11-15:
I'm trying to release classic BT memory since I won't be using for my application.  I'm currently using the Arduino IDE with the latest pull (as of this morning).

Before I call the BLEDevice::init() call I invoke the `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)` method with success.

Afterwards, I start the BLEScanner with the following (the code blocks aren't wrapping correctly):

this->_scan = BLEDevice::getScan();
this->_scan->setActiveScan(true); //active scan uses more power, but get results faster

  if (!this->_scan->start(durationInSeconds, handleScanCompleteCallback)) {
    ESP_LOGE(LOG_TAG, "<< [false] BLScan.start() failed.");
    this->_scan = NULL;
    return false;
  }

The scan->start ends up failing with the following message:

`08:17:46.704 -> [E][BLEScan.cpp:206] start(): esp_ble_gap_set_scan_params: err: 259, text: Invalid state`

Is it possible to use the `esp_bt_controller_mem_release` call and still use a scanner?

chegewara, 2018-11-15:
@czuvich This error log means that something is wrong with variable in this line:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/src/BLEScan.cpp#L203

I dont know why or what, but could you try to debug it somehow in your code? It requires that you change BLEScan.cpp before this line:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/src/BLEScan.cpp#L208
and print log values:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/src/BLEScan.cpp#L31-L33
and `m_scan_params.scan_interval`, `m_scan_params.scan_window`.

czuvich, 2018-11-15:
So I added the following line before the scan call:

`ESP_LOGI(LOG_TAG, "czuvich before_esp_ble_gap_set_scan_params >> scan_type:[%d] own_addr_type:[%d] scan_filter_policy:[%d]", m_scan_params.scan_type, m_scan_params.own_addr_type, m_scan_params.scan_filter_policy);`

I also cranked up the log output to Verbose
Here's the output:

13:57:43.865 -> [D][BLEScan.cpp:196] start(): >> start(duration=5)

13:57:43.865 -> [I][BLEScan.cpp:203] start(): czuvich before_esp_ble_gap_set_scan_params >> scan_type:[1] own_addr_type:[0] scan_filter_policy:[0]

13:57:43.865 -> [E][BLEScan.cpp:207] start(): esp_ble_gap_set_scan_params: err: 259, text: Invalid state

czuvich, 2018-11-15:
Including the scan_window and scan_interval as requested here:

14:03:53.028 -> [D][BLEScan.cpp:196] start(): >> start(duration=5)

14:03:53.028 -> [I][BLEScan.cpp:203] start(): czuvich before_esp_ble_gap_set_scan_params >> scan_type:[1] own_addr_type:[0] scan_filter_policy:[0] scan_interval:[160] scan_window:[160]

14:03:53.028 -> [E][BLEScan.cpp:207] start(): esp_ble_gap_set_scan_params: err: 259, text: Invalid state

chegewara, 2018-11-15:
Nicely done, but all logs looks good. All parameters are initialized correctly. There is last that i mentioned:
`scan_duplicate`
Could you try to print log `m_scan_params.scan_duplicate`?

czuvich, 2018-11-15:
Here ya' go.

start(): czuvich before_esp_ble_gap_set_scan_params >> scan_type:[1] own_addr_type:[0] scan_filter_policy:[0] scan_window:[160] scan_interval:[160] scan_duplicate:[0]

chegewara, 2018-11-15:
All parameters are perfectly fine. There is something wrong with arduino-esp32and bt stack.

czuvich, 2018-11-15:
Anything else I need to do on my end?

czuvich, 2018-11-16:
Ok dokie.  Hopefully, the esp-idf team will fix this bug.  It would be great to be able to free up some RAM for my app :)

chegewara, 2018-11-24:
Hi @czuvich 
there is very simple explanation of this error. As you know arduino is precompiled set of esp-idf libraries. Unfortunately it is precompiled with some bluetooth settings just like this one:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L194
Because classic bluetooth is turned on, when you release memory from classic bt then bluetooth controller cant be initialized:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-bt.c#L43-L47
`[E][esp32-hal-bt.c:45] btStart(): BT Enable failed`

Sorry it took so long to find such simple explanation.

czuvich, 2018-11-25:
Thanks @chegewara!  So, it sounds like I have a few options if I want to take advantage of `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)`:

1)  Drop Arduino and program directly to the ESP32 customizing as needed.
2) Recompile the Arduino ESP32 binaries disabling classic BT; however, I assume it may be some trial and error since I'm not too familiar with the exact side effects of changing different configurations for Arduino.
3) Deal with it until an official fix is released.

For me,  the second option doesn't make a lot of sense unless it's a quick fix  (I might as well go native).  Do those options sound about right?

chegewara, 2018-11-25:
4. Use arduino as component in esp-idf.

closed 2019-08-15 by stale[bot]