espressif/arduino-esp32 #2063: Some functions defined in esp32-hal-uart.c are not declared in esp32-hal-uart.h

AlessandroAU, 2018-11-16:
Some functions defined in esp32-hal-uart.c are not declared in esp32-hal-uart.h,

They are
```
void uartDetachRx(uart_t* uart)

void uartDetachTx(uart_t* uart)

void uartAttachRx(uart_t* uart, uint8_t rxPin, bool inverted)

void uartAttachTx(uart_t* uart, uint8_t txPin, bool inverted)
```

Is this intentional? 

Regards,
Alessandro

stickbreaker, 2018-11-16:
I would say yes, these functions are just helper functions for `uartBegin()` and `uartEnd()`.  They are not designed to be used by the applications programmer.  

Chuck.

AlessandroAU, 2018-11-16:
Hi Chuck 

Hmm fair enough, I need to implement bi-directional serial over 1 wire, my plan was to tie RX/TX together with an appropriate resistor and just Detach/Attach the TX pin when needed. Would there be an issue with using those functions to do so?

Regards,
Alessandro

AlessandroAU, 2018-11-16:
I don't mind if my output shows back up in the input buffer. I can just ignore it in software.
I am interfacing with a RC transmitter which talks over one wire serial.

It sends a command packet every 4ms, and will accept a response in the window between them. Therefore there should not be a risk of both devices sending at once if timings are maintained. 

If I do the following:

detachTX as default
wait for and process command Packet,
attachTX, send response, detachTX,
Flush RX buffer
Wait for next Command Packet. 

Should it not work?

Regards,
Alessandro

closed 2019-08-15 by stale[bot]

positron96, 2020-05-24:
So, I need to have tx pin inverted and rx pin non-inverted. What are my options, since uartBegin can only change both inversions and separate functions for uartAttachRx\Tx are not exported?

lbernstone, 2020-05-24:
Please don't hijack issues.  If you want help, man up and open an issue.

positron96, 2020-05-24:
Hm, sorry, I didn't mean you to get offended. I just think that my problem stems from this exact issue, so I'm not hijacking anything, just expanding.

lbernstone, 2020-05-24:
It has absolutely nothing to do with the OP issue, other than using uart and esp32.