espressif/arduino-esp32 #2065: Esp32 rebooting after power up

anandpanchal, 2018-11-16:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			2018
IDE name:							Arduino IDE? 
Flash Frequency:					       80Mhz
PSRAM enabled:                                             no
Upload Speed:						115200
Computer OS:                                                  Windows 7

Can anybody help me in understanding that why my esp is rebooting at startup ? 

Thanks

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4015af74  PS      : 0x00060034  A0      : 0x80080f2e  A1      : 0x3ffc0bb0  
A2      : 0x00060a23  A3      : 0x00000000  A4      : 0x3ffcccdc  A5      : 0x000000ff  
A6      : 0x3ffc3fc4  A7      : 0x3ffcccdc  A8      : 0x80081282  A9      : 0x3ffc0b90  
A10     : 0x000f313f  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060a23  SAR     : 0x00000016  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082687  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4015af74

Backtrace: 0x4015af74:0x3ffc0bb0 0x40080f2b:0x3ffc0bd0 0x40080fbd:0x3ffc0bf0 0x4008176d:0x3ffc0c10 0x40082684:0x00000000

Rebooting...
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4015af74  PS      : 0x00060034  A0      : 0x80080f2e  A1      : 0x3ffc0bb0  
A2      : 0x00060e23  A3      : 0x00000000  A4      : 0x8008d6a3  A5      : 0x3ffc0ba0  
A6      : 0x00000008  A7      : 0x00000001  A8      : 0x80081282  A9      : 0x3ffc0b90  
A10     : 0x000ece86  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060e23  SAR     : 0x00000012  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082681  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4015af74

Backtrace: 0x4015af74:0x3ffc0bb0 0x40080f2b:0x3ffc0bd0 0x40080fbd:0x3ffc0bf0 0x4008176d:0x3ffc0c10 0x4008267e:0x00000000

Rebooting...

stickbreaker, 2018-11-16:
Decode your back trace.
Use [me-no-dev's arduino plugin](https://github.com/espressif/arduino-esp32/blob/master/README.md#decoding-exceptions)

Chuck.

anandpanchal, 2018-11-16:
Decoding 5 results
0x4015af74: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
0x40080f2b: fall_isr2() at C:\Users\Asus\Desktop\esp32_13/esp32_13.ino line 80
0x40080fbd: __onPinInterrupt at C:\Users\Asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/esp32-hal-gpio.c line 233
0x4008176d: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/xtensa_vectors.S line 1105
0x40082684: spi_flash_op_block_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 82 (discriminator 1)

i decoded my backtraces but i am not able to extract information out of it. can you please help/guide me in this regard ? @stickbreaker

anandpanchal, 2018-11-16:
Decoding 5 results
0x4015af74: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
0x40080f2b: fall_isr2() at C:\Users\Asus\Desktop\esp32_13/esp32_13.ino line 80
0x40080fbd: __onPinInterrupt at C:\Users\Asus\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/esp32-hal-gpio.c line 233
0x4008176d: _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/xtensa_vectors.S line 1105
0x4008267e: spi_flash_op_block_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 81

anandpanchal, 2018-11-16:
Here is my ISR sketsh : 

/* ISR IR1, IRAM_ATTR , to store the isr variables in ram */
void IRAM_ATTR fall_isr1()
{
  unsigned long current_ts =  millis();
  if ((current_ts - fall_ts_ir1) > INTERRUPT_FILTER_DURATION)
  {
    fall_flag_ir1 = true;
    fall_ts_ir1 = current_ts;
  }
}

/* isr2 */
void IRAM_ATTR fall_isr2()
{
  unsigned long current_ts = millis();
  if((current_ts - fall_ts_ir2) > INTERRUPT_FILTER_DURATION)
  {
    fall_flag_ir2 = true;
    fall_ts_ir2 = current_ts;
  }
}

stickbreaker, 2018-11-16:
Define `INTERRUPT_FILTER_DURATION`

Chuck.

anandpanchal, 2018-11-16:
It is defined in the code already. I have only posted the isr part .

stickbreaker, 2018-11-16:
>0x4015af74: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228

`__dividi3` is an longInt divide function.
This leads me to believe your #define is complex.  Every bit go code in/called by an isr must be IRAM_ATTR.  BUT, memory available for IRAM is small.

Chuck.

stickbreaker, 2018-11-16:
Precalculate the value, so the isr can just use it.

anandpanchal, 2018-11-16:
Variables which are shared by isr and loop block , are declared as
volatile. and isr variables are stored in IRAM by declaring isr as
IRAM_ATTR. What else I can do to avoid this rebooting situation?

anandpanchal, 2018-11-16:
Fall flag ir1 and fall ts ir1 , fall flag ir2, fall ts ir2 are declared as
volatile

stickbreaker, 2018-11-16:
What core are you currently using? Are you using the latest development core (downloaded directly from GitHib?).

Chuck.

stickbreaker, 2018-11-16:
@me-no-dev this crash looks like it is happening in the calls to `millis()` [here](https://github.com/espressif/arduino-esp32/blob/85032b226c7775bae510e954112823a2ae32181a/cores/esp32/esp32-hal-misc.c#L49)

It doesn't make any sense.

Chuck.

anandpanchal, 2018-11-17:
#include <WiFi.h>
#include <NTPClient.h>
#include <WiFiUdp.h>
#include <HTTPClient.h>

/* poll wifi status every 500 ms*/
#define WIFI_TEST_INTERVAL  500

/* wifi ssid and password */
const char* ssid     = "xxxxxx";
const char* password = "xxxxxxx";

/* Define NTP Client to get time */
WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP);

/* Variables to save date and time */
String formattedDate;
String dayStamp;
String timeStamp;
String timeStamp_off;
String Date;
String Time;
String cutime;
String day;
String data_post;

/* ISR parameters and flags */
volatile unsigned long fall_ts_ir1 = 0;
volatile unsigned long fall_ts_ir2 = 0;
volatile bool fall_flag_ir1 = false;
volatile bool fall_flag_ir2 = false;

/* relay parameters */
bool relay_on_flag = false;
unsigned long relay_on_ts = 0;
unsigned long relay_on_duration = 3000; //600000 for 10 minutes
const int relay = 5;
bool relay_default_flag = false;

/* ir1, ir2 sensors and wifi indication led pins */
const int ir1 = 33;
const int ir2 = 35;
const int tr = 23;
const int wifi_led = 22;
const int relay_switch = 26;
const int relay_output = 23;

const int mode_led = 19;
const int mode_switch = 27;
int prev_mode_switch_state;
char server_ip[16] = "1.1.1.1";
int server_port = 1111;

bool send_post_flag;

/* ignore interrupts if occurs within 1.5 seconds */
#define INTERRUPT_FILTER_DURATION 1500

/* ISR IR1, IRAM_ATTR , to store the isr variables in ram */
void IRAM_ATTR fall_isr1()
{
  unsigned long current_ts =  millis();
  if ((current_ts - fall_ts_ir1) > INTERRUPT_FILTER_DURATION)
  {
    fall_flag_ir1 = true;
    fall_ts_ir1 = current_ts;
  }
}

/* isr2 */
void IRAM_ATTR fall_isr2()
{
  unsigned long current_ts = millis();
  if((current_ts - fall_ts_ir2) > INTERRUPT_FILTER_DURATION)
  {
    fall_flag_ir2 = true;
    fall_ts_ir2 = current_ts;
  }
}

Divide-And-Conquer, 2018-12-08:
I'm having a similar problem to this, my code  contains the necessary IRAM_ATTR declarations, and builds/runs fine using Arduino IDE release 1.8.3. But I get similar crash (__divdi3) from 1.85 and 1.8.8. I do notice that the build size (RAM and Program) has increased with the later builds.

When running I display

3 Relay Unified Switch - Switch Number = 1
Module: UnifiedSwitch3Relay.ino - Version: 1.0.10
 - Compiled on: Dec  8 2018 at 10:08:58
Initialise ST7735
DS18B20 - Dallas Temperature IC Control startup
ESP32 Chip ID = 9C0074A4AE30

Connecting to wifi: 
scan start
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40169a30  PS      : 0x00060734  A0      : 0x80080f76  A1      : 0x3ffc0b70  
A2      : 0x00000020  A3      : 0x00000000  A4      : 0x80082e9e  A5      : 0x3ffceab0  
A6      : 0x3ff000dc  A7      : 0x00000001  A8      : 0x8008138e  A9      : 0x3ffc0b50  
A10     : 0x0035b607  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000000  
A14     : 0x00000002  A15     : 0x3ffcec2c  SAR     : 0x00000016  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x400834f3  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40169a30

Backtrace: 0x40169a30:0x3ffc0b70 0x40080f73:0x3ffc0b90 0x40080f68:0x3ffc0bb0 0x40080f5b:0x3ffc0bd0 0x40081465:0x3ffc0bf0 0x4008195d:0x3ffc0c10 0x40082874:0x00000000

Rebooting...

Exception decoder shows

PC: 0x40169a30: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
EXCVADDR: 0x00000000

Decoding stack results
0x40169a30: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
0x40080f73: ClickEncoder::service() at C:\Users\Martin\Documents\Arduino\libraries\encoder-arduino\ClickEncoder.cpp line 72
0x40080f68: timerIsr() at C:\Users\Martin\AppData\Local\Temp\arduino_build_216843\sketch\menus.cpp line 86
0x40080f5b: onTimer2() at H:\Projects 2\117 - UnifiedSwitch3Relay\UnifiedSwitch3Relay\UnifiedSwitch3Relay/UnifiedSwitch3Relay.ino line 284
0x40081465: __timerISR at C:\Users\Martin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\esp32-hal-timer.c line 88
0x40082874: spi_flash_op_block_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 82

I appear to be calling in my same libraries for both old and new builds, so I'm thinking that it is something to do with newer Arduino code base. 

My code is quite large, I guess my my next option is to strip it down to a minimal single file.

If anyone has any suggestions I'd be interested.

Martin

Divide-And-Conquer, 2019-02-04:
I've taken anandpanacal's code, and attempted to get it down to a small self contained demonstration of the problem.

The code has been seen to demonstrate the fault on various releases of the IDE between 1.8.2 and 1.8.8.
As the fault manifests itself when wi-fi is enabled, I did think it might have something to do with power
consumption, but I've tested the code with two different ESP32 boards, with different power supplies,
so I don't believe its hardware related.

The exception decoder details are as follows:

PC      : 0x40137ea8  PS      : 0x00060034  A0      : 0x80080f36  A1      : 0x3ffc0bb0  
A2      : 0x00060e23  A3      : 0x00000000  A4      : 0x3ffcc6e4  A5      : 0x000000ff  
A6      : 0x3ffc3e7c  A7      : 0x3ffcc6e4  A8      : 0x80081226  A9      : 0x3ffc0b90  
A10     : 0x004ac23d  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060e23  SAR     : 0x00000012  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x4008270d  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40137ea8

Backtrace: 0x40137ea8:0x3ffc0bb0 0x40080f33:0x3ffc0bd0 0x400812fd:0x3ffc0bf0 0x400817f5:0x3ffc0c10 0x4008270a:0x00000000

PC: 0x40137ea8: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
EXCVADDR: 0x00000000

Decoding stack results
0x40137ea8: __divdi3 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/libgcc2.c line 1228
0x40080f33: fall_isr2() at H:\Projects 2\117 - UnifiedSwitch3Relay\anandpanchal/anandpanchal.ino line 76
0x400812fd: __timerISR at C:\Users\Martin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\esp32-hal-timer.c line 88
0x4008270a: spi_flash_op_block_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/cache_utils.c line 82

The code is:
**
#include <WiFi.h>
#include <WiFiUdp.h>

/* ISR parameters and flags */
volatile unsigned long fall_ts_ir1 = 0;
volatile unsigned long fall_ts_ir2 = 0;
volatile bool fall_flag_ir1 = false;
volatile bool fall_flag_ir2 = false;

extern void IRAM_ATTR fall_isr1();
extern void IRAM_ATTR fall_isr2();

/* create a hardware timers */
hw_timer_t * timer1 = NULL;
hw_timer_t * timer2 = NULL;

void SetTimerInterrupt1_1000ms()
{
  //
  /* Use 1st timer of 4 */
  /* 1 tick take 1/(80MHZ/80) = 1us so we set divider 80 and count up */
  timer1 = timerBegin(0, 80, true);

  /* Attach onTimer function to our timer */
  timerAttachInterrupt(timer1, &fall_isr1, true);

  /* Set alarm to call onTimer function every second 1 tick is 1us
  => 1 second is 1000000us */
  /* Repeat the alarm (third parameter) */
  timerAlarmWrite(timer1, 1000000, true);

  /* Start an alarm */
  timerAlarmEnable(timer1);
}

// Function: SetTimerInterrupt2_1ms
// Set Timer 2 to interrupt every millisecond
//
void SetTimerInterrupt2_1ms()
{
  //
  /* Use 2nd timer of 4 */
  /* 1 tick take 1/(80MHZ/80) = 1us so we set divider 80 and count up */
  timer2 = timerBegin(1, 80, true);

  /* Attach onTimer function to our timer */
  timerAttachInterrupt(timer2, &fall_isr2, true);

  /* Set alarm to call onTimer function every 1mSec 1 tick is 1us
  => 1 mSec is 1000us */
  /* Repeat the alarm (third parameter) */
  timerAlarmWrite(timer2, 1000, true);

  /* Start an alarm */
  timerAlarmEnable(timer2);
}

/* ignore interrupts if occurs within 1.5 seconds */
#define INTERRUPT_FILTER_DURATION 1500

/* ISR IR1, IRAM_ATTR , to store the isr variables in ram */
void IRAM_ATTR fall_isr1()
{
unsigned long current_ts = millis();
if ((current_ts - fall_ts_ir1) > INTERRUPT_FILTER_DURATION)
{
fall_flag_ir1 = true;
fall_ts_ir1 = current_ts;
}
}

/* isr2 */
void IRAM_ATTR fall_isr2()
{
unsigned long current_ts = millis();
if((current_ts - fall_ts_ir2) > INTERRUPT_FILTER_DURATION)
{
fall_flag_ir2 = true;
fall_ts_ir2 = current_ts;
}
}

IPAddress ip(192,168,1,112); 
IPAddress gateway(192, 168, 1, 1); 
IPAddress subnet(255, 255, 255, 0);
IPAddress DNS( 192,168,1,1);

void setup() {
// put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Start");

  delay(1000);                    // Give WiFi a chance to connect

  WiFi.config(ip, gateway, subnet, DNS); // Use static IP
  WiFi.begin("XXXXXXXXXXXX", "YYYYYYYYYYYY"); // Fill in login and password
  
  delay(1000);                    // Give WiFi a chance to connect

  SetTimerInterrupt1_1000ms();
  SetTimerInterrupt2_1ms();
}

void loop() {
  // put your main code here, to run repeatedly:
 //         Serial.print(n,DEC);
          
  Serial.print(fall_ts_ir1,DEC);
  Serial.print(" - ");
  Serial.print(fall_ts_ir2,DEC);
  Serial.println("");
}
**
If the 
  SetTimerInterrupt1_1000ms();
  SetTimerInterrupt2_1ms();
statements are moved in front of the 
  WiFi.config(..); 
statement, the WiFi system as it does not get a chance to complete before the divdi3 crash occurs.

Martin

closed 2019-08-15 by stale[bot]