espressif/arduino-esp32 #2066: Timed out waiting for PHY register. MH-ET Live DevKit32?] & LAN 8720

X-WL, 2018-11-16:
> I have a problem with esp-Wroom-32 with ETH_Lan8720 board. I'am using an arduino example code from here :
> 
> https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino
> 
> I did the wireing like this:
> 
> GPIO17 - PHY_POWER : NC - Osc. Enable - 4k7 Pulldown
> GPIO22 - EMAC_TXD1 : TX1
> GPIO19 - EMAC_TXD0 : TX0
> GPIO21 - EMAC_TX_EN : TX_EN
> GPIO26 - EMAC_RXD1 : RX1
> GPIO25 - EMAC_RXD0 : RX0
> GPIO27 - EMAC_RX_DV : CRS
> GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
> GPIO23 - SMI_MDC : MDC
> GPIO18 - SMI_MDIO : MDIO
> GND : GND
> 3V3 : VCC
> 
> from the site :
> https://sautter.com/blog/ethernet-on-esp32-using-lan8720/
> 
> I soldered the NC PIN on the lan8720 board to the enable pin of the oscillator. I added pull down resistor to nc pin and pull up to gpio00.
> 
> LAN 8720 board : https://fr.aliexpress.com/item/Smart-Electronics-LAN8720-module-network-module-Ethernet-transceiver-RMII-interface-development-board-for-arduino/32825173408.html?spm=a2g0s.9042311.0.0.40696c370I7nRH
> 
> ESP32: https://fr.aliexpress.com/item/ESP-WROOM-32-WiFi-Bluetooth-4-2-dual-core-CPU-MCU-low-power-Bluetooth-240MHZ-ESP32/32829367382.html?spm=a2g0s.9042311.0.0.15636c37H4la82
> 
> I get the following errors:
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:952
> load:0x40078000,len:6084
> load:0x40080000,len:7936
> entry 0x40080310
> E (2256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
> E (3256) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0 (mask 0xfff0). Current value 0x0000
> E (4256) emac: Timed out waiting for PHY register 0x2 to have value 0x0007 (mask 0xffff). Current value 0x0000
> 
> I dont know what to do , can anyone help me?

Hi I use MH-ET Live board. I have the same problem. What is the solution? I am using the latest version of the library in the PlatformIO.

_Originally posted by @X-WL in https://github.com/espressif/arduino-esp32/issue_comments#issuecomment-439553359_

yanzixiang, 2019-01-23:
https://github.com/espressif/arduino-esp32/issues/1938

X-WL, 2019-08-01:
The problem is still relevant for the latest version.
#1938 doesn't solve the problem. Apparently, the Board can't get an IP address. Maybe I'm doing something wrong. Can I get detailed instructions on what to do? Thanks!

X-WL, 2019-09-30:
up

closed 2019-12-13 by stale[bot]