espressif/arduino-esp32 #2069: ESP32 I2C Write.read just read 0

arnold-b, 2018-11-18:
Hello Together, 

Wire.beginTransmission(0x5D);
 
  uint8_t w1 = Wire.write(0x81);
   uint8_t w2 = Wire.write(0x40);
   int error = Wire.endTransmission(false);
 
  Wire.requestFrom(0x5D, 11);
  
  uint8_t data = 0; 
    while (Wire.available()) {
      data = Wire.read();
      Serial.println(data );
    }
   Wire.endTransmission(true);

Result is jus a a lot 0. In logic analyser i get the right values.

Is there anyone who succeed

arnold-b, 2018-11-18:
I forgot. Wire speed is 400000

stickbreaker, 2018-11-18:
Known problem in Release 1.0.0.

Either install the dev version from this repo, or just replace the 4 i2c files.

See #1962 

Chuck.

arnold-b, 2018-11-18:
Thanks i will try. I searched a few hours and didn’t find any solution. 

 

And I didn’t think that’s a problem. Because in eclipse it work

stickbreaker, 2018-11-18:
@arnold-b if you installed using the Arduino Board Manager, it installed the faulty release. The dev repo version is correct and working.

Chuck.

arnold-b, 2018-11-18:
Hey, 

 

do you mean arduino-esp32-1.0-rc4 ?

because I tried this one and it still not working

stickbreaker, 2018-11-18:
@arnold-b the best working files are in this repo.  Between RC1 and Release 1.0.0 there were multiple changes to I2C, Between RC4 and Release 1.0.0 I  resolved a problem with interrupt saturation.  During this major rewrite, I made a mistake.  It manifested when using a Write ReSTART operation. (`Wire.endTransmission(false);`)  as you noticed, the subsequent Read operation does not return any data.

just follow the instruction in #1962, copy the four files from this repo.  replace the four same your Arduino installation with those file.  

*   \cores\esp32\esp32-hal-i2c.h
*  \cores\esp32\esp32-hal-i2c.c
*  \libraries\Wire\src\Wire.h
*  \libraries\Wire\src\Wire.cpp

You have a couple of problems with your code:
```c++
#define _sda 21 //16
#define _scl 22 // 17

//Wire.setClock(400000);  //Wire.begin() needs to be first
Wire.begin(_sda, _scl,400000);
Wire.beginTransmission(0x5D);

uint8_t w1 = Wire.write(0x81); // w1 will be 0 if there is no room in the output buffer (128 max), or 1 as success
uint8_t w2 = Wire.write(0x40); // same as w1
int error = Wire.endTransmission(false); // will return error code

if(error != I2C_ERROR_CONTINUE){ // because of ReSTART operation, Normally I2C_ERROR_OK
   Serial.printf(" I2C write terminated with error: %d(%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
}

uint8_t count =Wire.requestFrom(0x5D, 11);

if(Wire.lastError() != I2C_ERROR_OK){ 
   Serial.printf(" I2C Read returned %d characters, terminated with error: %d(%s)\n", count, Wire.lastError(), Wire.getErrorText(Wire.lastError()));
}

uint8_t data = 0;
while (Wire.available()) {
  data = Wire.read();
  Serial.println(data );
  }
//Wire.endTransmission(true); // must be paired with Wire.beginTransmission()
```
Chuck.

arnold-b, 2018-11-18:
Hey ,

 

thanks

Just now i try to install the whole repo and I have different problem. 

 

Collect2.exe error with uart

 

C:\Users\Braun\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib\libdriver.a(uart.o): In function `uart_flush_input':

 

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/uart.c:1246: undefined reference to `vRingbufferReturnItem'

 

It looks like there is some fixed user path

stickbreaker, 2018-11-18:
@arnold-b 

>Wait for git to pull any changes and close Git GUI
>Step 4 Open [ARDUINO_SKETCHBOOK_DIR]/hardware/espressif/esp32/tools and double-click get.exe

Did you follow all the instructions?

Make sure you run the `get.exe` from the tools directory.

Shutdown and restart the Arduino IDE,

You will have to reselect your board inside the Arduino IDE to tell it to use this newly installed core.

Adding the entire dev Repo adds a second set of boards to the boards list. Scroll completely down the boards list, you will notice a second **ESP32 Arduino** list of boards.  too ease confusion I change the title text in my `platform.txt` files.  The first line
>name=EPS32 Arduino

to:
>name=ESP32 Github Version

Under Windows 7:
* Board Manager version (release 1.0.0) is located in:
C:\Users\\{your user name}\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\
* GitHub Dev version:
C:\Users\\{your user name}\Documents\Arduino\hardware\espressif\esp32

Chuck.

arnold-b, 2018-11-18:
Sorry, 

 

i didnt. I just copied the Master .zip 

 

then I changed the SDK to my old one. 

 

I2C works now. 

 

But I will change again and follow your instruction.

 

Thanks Again

arnold-b, 2018-11-18:
Problem solved with https://github.com/espressif/arduino-esp32/issues/1962

closed 2018-11-18 by arnold-b