espressif/arduino-esp32 #2073: libriver.a (uart) collect2.error

arnold-b, 2018-11-18:
AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib\libdriver.a(uart.o):(.literal.uart_flush_input+0x8): undefined reference to `vRingbufferReturnItem'

AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib\libdriver.a(uart.o):(.literal.uart_flush_input+0xc): undefined reference to `xRingbufferReceive'

Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib\libdriver.a(uart.o):(.literal.uart_flush_input+0x10): undefined reference to `xRingbufferSend'

Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib\libdriver.a(uart.o): In function `uart_flush_input':

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/uart.c:1246: undefined reference to `vRingbufferReturnItem'

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/uart.c:1253: undefined reference to `xRingbufferReceive'

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/uart.c:1270: undefined reference to `vRingbufferReturnItem'

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/uart.c:1271: undefined reference to `xRingbufferSend'

collect2.exe: error: ld returned 1 exit status

ArTyGill, 2019-05-14:
+

ArTyGill, 2019-05-14:
I have some problem

ArTyGill, 2019-05-15:
Solved the problem by reinstalling the library.

closed 2019-08-14 by stale[bot]