espressif/arduino-esp32 #2075: Is it posibble using BLE to do OTA

syk543277780, 2018-11-18:
Board:							ESP32 Dev Module
Core Installation/update date:		Sep/2018
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:				        115200
Computer OS:                                          Windows 10

### Description:
Is it posibble using BLE to do OTA. I am going to do the ota via ble. however i cannot find any example related to this.

chegewara, 2018-11-19:
I will say in theory yes, i even saw info that someone is doing this with esp-idf, but i didnt try it yet (its on my list todo).

lbernstone, 2018-11-21:
No security needed, no boot partition changes. You need to stream the file from BLE to the Update class in a similar way to https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/AWS_S3_OTA_Update

syk543277780, 2018-11-21:
Thanks Ibernstone, this what I am looking for.

closed 2018-12-03 by syk543277780

chegewara, 2018-12-04:
I have to say i already did OTA over BLE example code, but its not available. So answering OP question, yes, it is possible.

syk543277780, 2018-12-04:
Thanks chegewara, I already got answer from Ibernstone. 
Ibernstone shared the example link above.
I can use the SD card OTA example.
I will transfer the firmware to ESP32 via BLE and save it to SPIFFS as a bin file, and then follow the SD card OTA example. it works for me.

huynh213, 2018-12-05:
@chegewara: I am interested to learn how you do the OTA over BLE on the ESP32? Is there any guideline/ example code that I can have access to for reference?

chegewara, 2018-12-06:
Hi,
I didn't do too many tests with it, but here is my thinking:
- use indicate or write to characteristic (depend you want to use client or server), max speed about 120kbits
- using notifications you will speed up about 4 times, but it's not as reliable and you need at least to write code to check if all packets (bytes) were received, max speed about 500-600kbits
- all test I performed with esp-idf, but should be applicable to arduino, at least when I fix all issues.

chegewara, 2018-12-14:
@huynh213 Its a part responsible for OTA in my app. You have to remember its not complete code.:
```
#include <string.h>
// #include <string>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_ota_ops.h"

#include "nvs.h"
#include "nvs_flash.h"
#include "sdkconfig.h"

static int binary_file_length = 0;
static char* TAG = "TAG";
esp_ota_handle_t update_handle = 0 ;
const esp_partition_t *update_partition = NULL;

static void task_fatal_error()
{
    ESP_LOGE(__func__, "FATAL ERROR");
}

extern void ota_begin()
{
    esp_err_t err;
    /* update handle : set by esp_ota_begin(), must be freed via esp_ota_end() */

    ESP_LOGE(TAG, "Starting OTA example...");

    const esp_partition_t *configured = esp_ota_get_boot_partition();
    const esp_partition_t *running = esp_ota_get_running_partition();

    if (configured != running) {
        ESP_LOGW(TAG, "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
                 configured->address, running->address);
        ESP_LOGW(TAG, "(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
    }
    ESP_LOGI(TAG, "Running partition type %d subtype %d (offset 0x%08x)",
             running->type, running->subtype, running->address);

    ESP_LOGI(TAG, "Connect to Wifi ! Start to Connect to Server....");

    update_partition = esp_ota_get_next_update_partition(NULL);
    assert(update_partition != NULL);
    ESP_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
             update_partition->subtype, update_partition->address);

    err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_begin failed, error=%d", err);
        task_fatal_error();
    }
    ESP_LOGI(TAG, "esp_ota_begin succeeded");
}
#define BUFFSIZE 517
#define TEXT_BUFFSIZE 517
static char ota_write_data[BUFFSIZE + 1] = { 0 };

extern void ota_chunk(uint8_t* text, size_t buff_len) 
{
    ESP_LOG_BUFFER_HEX_LEVEL(__func__, text, 32, ESP_LOG_INFO);
    esp_err_t err = esp_ota_write( update_handle, (const void *)text, buff_len);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Error: esp_ota_write failed! err=0x%x", err);
        task_fatal_error();
    }
    binary_file_length += buff_len;
    // ESP_LOGI(TAG, "Have written image length %d", binary_file_length);
    // }
}

extern void ota_finalize() 
{
    ESP_LOGI(TAG, "Total Write binary data length : %d", binary_file_length);

    if (esp_ota_end(update_handle) != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_end failed!");
        task_fatal_error();
    }
    esp_err_t err = esp_ota_set_boot_partition(update_partition);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_set_boot_partition failed! err=0x%x", err);
        task_fatal_error();
    }
    ESP_LOGI(TAG, "Prepare to restart system!");
    esp_restart();
    return ;
}
```

And here is ble logic. I am using indications in this case:
```
static void notifyCallback(
	BLERemoteCharacteristic* pBLERemoteCharacteristic,
	uint8_t* pData,
	size_t length,
	bool isNotify) {
		// ESP_LOGE(LOG_TAG, "Notify callback for characteristic %s of data %s length %d",
		// 		pBLERemoteCharacteristic->getUUID().toString().c_str(), ((char*) pData), length);
        ESP_LOGI(__func__, "stack highwater--> %d", uxTaskGetStackHighWaterMark(NULL));

        ota_chunk(pData, length);
        if(length < 500)
            ota_finalize();
}
```
You also need to find a way to call `void ota_begin()`. I have this simplified because i am starting OTA process by pressing GPIO_0 on client, then client send write request to BLE server and in `onWrite()` callback i am starting full process, at the same time i have already called `ota_begin()` before first indication arrived.

This is task responsible to send binary file over ble:
```
void task(void *p){
    uint64_t bytes = 0;
    size_t total_size = file->size;
step = 500;
    ESP_LOGE(__func__, "size: %d", total_size);
    ESP_LOG_BUFFER_HEX_LEVEL(__func__, file->response, 32, ESP_LOG_INFO);
    while(connected && bytes < total_size)
    {
        size_t size = (file->size - bytes < step) ? file->size - bytes : step;
        current = rtc_time_get();
        pChar->setValue(file->response + bytes, size);
        pChar->notify(false);
        last = current;
        bytes += size;
        ESP_LOGI(__func__, "Bytes: %lld, speed: ", bytes);
    }
    ESP_LOGE(__func__, "size: %lld", bytes);
    vTaskDelete(NULL);
}
```

bionascu, 2020-01-18:
@syk543277780 were you able to get BLE OTA updates to work using the SD card OTA example? Would you be able to share how you did it for reference? Thank you!

bahalouati, 2020-08-08:
Can anyone provide a full-tutorial ? I am new to ESP32 and OTA via BLE is something I wanna do!

EmanueleFeola, 2020-09-10:
Is there an arduino code example of OTA via BLE for esp32 available?

fbiego, 2021-02-14:
Here is an example [ESP32_BLE_OTA_Arduino](https://github.com/fbiego/ESP32_BLE_OTA_Arduino) with an Android app code available

The process is very slow and not guaranteed to work since the packets are not verified

TVAVAE, 2021-02-17:
Someone has an example for ESP-IDF BLE OTA?

krupis, 2021-09-10:
I see this is an old discussion. I am also wondering if there is an ESP-IDF BLE OTA example code available?