espressif/arduino-esp32 #2082: FreeRTOS V8.2.0

Enderdead, 2018-11-19:
Hello,
It's possible to use FREERTOS V10  with your project than the current version (8.2) ?
If it's possible, how can I do this? 
Thank you.

lbernstone, 2018-11-19:
FreeRTOS is included as part of the [ESP-IDF](https://github.com/espressif/esp-idf).
Short answer- not without a major amount of integration work.

Enderdead, 2018-11-19:
Yes of course, I have to replace all .h files but I don't know how compile .a file ='(. 
This one : https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/lib/libfreertos.a

lbernstone, 2018-11-19:
See #1142 and https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
Be careful of the windmill, Don Quixote.

closed 2019-08-02 by Enderdead