espressif/arduino-esp32 #2083: build error freertos INT_MAX not defined

AlmightyFrog, 2018-11-19:
### Hardware:
Board:							ESP-32S
PLATFORM: Espressif 32 > NodeMCU-32S
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)

IDE name:						Platform.io
Both tried stage and current version in platform.io
```
Platform Espressif 32
--------
Updating espressif32                     @ 1.5.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10000.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]

Platform Espressif 32 (Stage)
--------
Updating espressif32                     @ 4edbe9b        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20500.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ c3ec91f        [Up-to-date]
Updating tool-espotapy                   @ 1.1.2          [Up-to-date]
```

### Description:
Trying to build https://github.com/bdring/Grbl_Esp32 with platform.io but besides warnings about redefines (can be ignored for beginning I guess) there is also an issue with building freertos. Including the limit.h might solve it, but not sure how to do that without changing Arduino.h file:
```
In file included from /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:0,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 
from /tmp/tmpHPIv_p:1:                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        
/home/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/task.h: In function 'BaseType_t xTaskCreate(TaskFunction_t, cons                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            
/home/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/task.h:96:24: error: 'INT_MAX' was not declared in this scope                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               
#define tskNO_AFFINITY INT_MAX             
```
Any Idea how to solve it? Best of cause would be not to change the Grbl_Esp32 project files, but if it is necessary, I'd prefer that over not working at all of cause. I already tried to define it in a overall header which is included by almost all code files in the project but that didn't work, too.

Here is my platformio.ini
```
[env:nodemcu-32s]
platform = espressif32
#platform = https://github.com/platformio/platform-espressif32.git#feature/stage

board = nodemcu-32s
framework = arduino

[platformio]
src_dir=Grbl_Esp32
lib_dir=libraries
```

And finally the full build output if it might be helpful:
```
**************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
Obsolete PIO Core v3.6.1 is used (previous was 3.6.2a3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
**************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************************
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Converting Grbl_Esp32.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 38 compatible libraries
Scanning dependencies...
Dependency Graph
[... 177 lines cut ...]
from Grbl_Esp32/grbl.h:36,
from Grbl_Esp32/eeprom.cpp:21:
Grbl_Esp32/serial.h: At global scope:
Grbl_Esp32/serial.h:40:21: warning: 'serialCheckTaskHandle' defined but not used [-Wunused-variable]
static TaskHandle_t serialCheckTaskHandle = 0;
^
*** [.pioenvs/nodemcu-32s/src/eeprom.cpp.o] Error 1
In file included from Grbl_Esp32/grbl.h:55:0,
from Grbl_Esp32/report.h:24,
from Grbl_Esp32/grbl.h:54,
from Grbl_Esp32/motion_control.h:30,
from Grbl_Esp32/grbl.h:50,
from Grbl_Esp32/eeprom.h:24,
from Grbl_Esp32/grbl.h:47,
from Grbl_Esp32/system.h:23,
from Grbl_Esp32/grbl.h:43,
from Grbl_Esp32/settings.h:28,
from Grbl_Esp32/grbl.h:42,
from Grbl_Esp32/tdef.h:4,
from Grbl_Esp32/grbl.h:39,
from Grbl_Esp32/config.h:41,
from Grbl_Esp32/grbl.h:36,
from Grbl_Esp32/gcode.cpp:25:
Grbl_Esp32/serial.h: At global scope:
```

lbernstone, 2018-11-19:
INT_MAX is defined in newlib/sys/config.h. You should be able to `#include <sys/config.h>`, it is odd that it is not getting included elsewhere.

AlmightyFrog, 2018-11-19:
@lbernstone thank you for your quick response. I already tried that adding in grbl.h file before including something else with both direct path ```#include "/home/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/limits.h"```
and what you suggested
```#include <sys/config.h>```
but it did not work with as looks for me same error message. Couldn't see an change. Maybe can you see something? Here is full output with verbose:

```
Processing nodemcu-32s (platform: espressif32; board: nodemcu-32s; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
xtensa-esp32-elf-g++ -o "/home/user/Bastelprojekte/3rd/Grbl_Esp32/Grbl_Esp32/Grbl_Esp32.ino.cpp" -x c++ -fpreprocessed -dD -E "/tmp/tmpkB_tBI"
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Framework incompatible library /home/user/Bastelprojekte/3rd/Grbl_Esp32/.piolibdeps/arduino_ID2171
Framework incompatible library /home/user/.platformio/lib/ble-nrf51822_ID378
Framework incompatible library /home/user/.platformio/lib/ble_ID296
Framework incompatible library /home/user/.platformio/lib/nrf51-sdk_ID328
Collected 38 compatible libraries
Scanning dependencies...
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Dependency Graph
|-- <EEPROM> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM)
|-- <ESPmDNS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <WebSockets> 2.1.2 (/home/user/Bastelprojekte/3rd/Grbl_Esp32/libraries/arduinoWebSockets)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <SPI> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |-- <WiFiClientSecure> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure)
|   |   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <ESP32SSPD> 1.0 (/home/user/Bastelprojekte/3rd/Grbl_Esp32/libraries/ESP32SSDP)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <ArduinoOTA> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA)
|   |-- <Update> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/Update)
|   |-- <ESPmDNS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS)
|   |   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <SD(esp32)> 1.0.5 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/SD)
|   |-- <FS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|   |-- <SPI> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <DNSServer> 1.1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/DNSServer)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|-- <Preferences> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/Preferences)
|-- <WebServer> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WebServer)
|   |-- <WiFi> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/WiFi)
|   |-- <FS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|-- <BluetoothSerial> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/BluetoothSerial)
|-- <Update> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/Update)
|-- <FS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/FS)
|-- <SPI> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <SPIFFS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/SPIFFS)
|   |-- <FS> 1.0 (/home/user/.platformio/packages/framework-arduinoespressif32/libraries/FS)
xtensa-esp32-elf-g++ -o .pioenvs/nodemcu-32s/src/Grbl_Esp32.ino.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Os -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -DPLATFORMIO=30601 -DARDUINO_NodeMCU_32S -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DARDUINO=10805 -DARDUINO_AR [... cut ...]
from Grbl_Esp32/grbl.h:49,
from Grbl_Esp32/system.h:23,
from Grbl_Esp32/grbl.h:45,
from Grbl_Esp32/settings.h:28,
from Grbl_Esp32/grbl.h:44,
from Grbl_Esp32/tdef.h:4,
from Grbl_Esp32/grbl.h:41,
from Grbl_Esp32/config.h:41,
from Grbl_Esp32/grbl.h:38,
from Grbl_Esp32/gcode.cpp:25:
Grbl_Esp32/serial.h: At global scope:
Grbl_Esp32/serial.h:40:21: warning: 'serialCheckTaskHandle' defined but not used [-Wunused-variable]
static TaskHandle_t serialCheckTaskHandle = 0;
^
*** [.pioenvs/nodemcu-32s/src/gcode.cpp.o] Error 1
```