espressif/arduino-esp32 #2087: Maybe an Important Bug found in WifiClient

lzw7603, 2018-11-20:
-
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/NOV/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
the code in /libraries/WiFi/src/WiFiClient.cpp line 74
"_fill += res;" run after that the value "_fill" will be -1 sometime.
because the code in /libraries/WiFi/src/WiFiClient.cpp line 70
"if(res < 0 && errno != EWOULDBLOCK) {" the res will be -1 sometime

and it will make the code in /libraries/WiFi/src/WiFiClient.cpp line 106
"*dst = _buffer[_pos];" the value "_pos" will larger than "_size" (default 1436) where is the size of the _buffer.
and then the code will go crash and output an info like "Guru Meditation Error: Core  1 panic'ed (LoadStoreError)...."

now, i fix it by adding some code like
"if(res < 0) return res;"
before
"_fill += res;"  in /libraries/WiFi/src/WiFiClient.cpp line 74
an it running running stable for a long time.

so can it be a right fix?
what is the meaning of the "if(res < 0 && errno != EWOULDBLOCK) {" in /libraries/WiFi/src/WiFiClient.cpp line 70 ?

cv007, 2018-11-23:
Since there are bool checks for fillBuffer() in this same file, maybe something like this would be better-
```
if(res < 0) {
    if(errno != EWOULDBLOCK) _failed = true;
    return 0;
}
```
it will return 0 on any error (since no bytes were read), and set _failed only when an error other than ewouldblock
if you return -1, those fillBuffer() checks may seem as true (-1 != false, so is true)

Jeroen88, 2019-01-02:
@lzw7603 you are right, I found the same problem and created PR #2259 to solve it

closed 2019-08-15 by stale[bot]