espressif/arduino-esp32 #2091: Cant change the manufacturing data while advertising Arduino-Ble

wan420, 2018-11-21:
I am trying to change the manufacturing data everytime the ble advertises, the code for ble server goes like this : 
`void setup() {`
  `Serial.begin(115200);`
  `Serial.println("Starting BLE work!");`
  `BLEDevice::init("Transmitter1");`
  `BLEServer *pServer = BLEDevice::createServer();`
  `//BLEService *pService = pServer->createService(SERVICE_UUID);`
  `pAdvertising = pServer->getAdvertising();`
  `Serial.println("Characteristic defined! Now you can read it in your phone!");`
`}`

`void loop() {`
 ` // put your main code here, to run repeatedly:`
  `Serial.println("started Advertisement");`
  `pAdvertising->start();`
  `delay(2000);`
  `pAdvertising->stop();`
  `Serial.println("Stopped Advertisement");`
  `setadvdata();`
  `delay(500);`
`}`

`void setadvdata(){`
  `cardname = cardname+String(val);`
  `servicedata = cardname.c_str();`
  `Serial.print("Serivcedata :");`
  `Serial.println(servicedata.c_str());`
  `advdata.setManufacturerData(servicedata);`
  `pAdvertising->setAdvertisementData(advdata);`
  `Serial.println("Cardname :"+cardname);`
  `if(val == 999){`
    `val = 956;`
  `}`
  `else{`
    `val++;`
  `}`
  `cardname  = "MiniBeacon_00";`
`}`

The ble client I am trying to check if a device have a ble name as well as a manufacturing data or not, if so then print it : 
`class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {`
    `void onResult(BLEAdvertisedDevice advertisedDevice) {`
      `name = advertisedDevice.getName().c_str();`
      `data = advertisedDevice.getManufacturerData().c_str();`
      `Serial.println("Name :"+name+"Manfc data :"+data);`
      `delay(1000);`
    `}`
`};`

`void setup() {`
  `Serial.begin(115200);`
  `BLEDevice::init("");`
  `pBLEScan = BLEDevice::getScan(); //create new scan`
  `pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(),true);`
  `pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster`
  
`}`

`void loop() {`
  `// put your main code here, to run repeatedly:`
  `Serial.println("Scanning ..");`
  `foundDevices = pBLEScan->start(scanTime);`
  `Serial.print("Devices found: ");`
  `Serial.println(foundDevices.getCount());`
  `Serial.println("Scan done!");`
  `delay(2000);`
`}`

Is there something with the code? 
The serial monitor output shows the cardname as "MiniBeacon_00956", it doesnt change to anything

chegewara, 2018-11-21:
This code looks good. You stop advertising, then you are changing value and you start advertising. It should work.

PS to paste code try to use 3 ` before code and 3 after code

wan420, 2018-11-21:
[image]
So i added the changes, instead of sending "Minibeacon_00956" , I changed it to just "956", and as you can see above, it worked only for a while, after 960, it stops incrementing I think, cuz after that, all I am getting is 961, and not further than that

wan420, 2018-11-21:
Ok, I think the lines : 

```advdata.setManufacturerData(servicedata.c_str());```
  ```pAdvertising->setAdvertisementData(advdata);```

adds the data and not overrides it. Cuz when I run the beacon server `Transmitter1` , the value increments for 5 and it stays like that, no more increment after that.

chegewara, 2018-11-21:
Code looks good, but i will investigate it:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/src/BLEAdvertising.cpp#L341-L344

But it can take some time, i am busy now. Also if you could open issue in this repository it will help me to track it:
https://github.com/nkolban/esp32-snippets

wan420, 2018-11-21:
ok, I will move the post over there, Thanks a lot.
Ya I come here often since I am learning BLE.
Sorry I am not that great in this so I cant help you.
Take your time its fine. In the meantime I will try to find a work-around of this.

chegewara, 2018-11-21:
You are helping a lot, because you are providing necessary data.

wan420, 2018-11-21:
I should probably close this

closed 2018-11-21 by wan420