espressif/arduino-esp32 #2092: Strange error from new httpclient

mouridis, 2018-11-21:
### Hardware:
Board: WEMOS LOLIN32
Core Installation/update date:	 Commit c3ec91f9683954e567ebddc77312c004ef42c933
IDE name: Arduino IDE 1.8.6
Flash Frequency: 80Mhz
PSRAM enabled: NoUpload Speed: 921600
Computer OS: Windows 10 x64

### Description:
Updating to the latest commit (c3ec91f9683954e567ebddc77312c004ef42c933) yesterday broke my app in a very strange way. After hours trying to figure out what's wrong, it seems that when a String object is passed in a function that uses httpclient it results in a Guru Meditation Error.

You should have no problem reproducing the issue using the following sketch:
```cpp
#include <WiFi.h>
#include <HTTPClient.h>

const String APSSID = "xxxxxxx";
const String APPassphrase =  "yyyyyyy";
const String unavailableString = "unavailable";
const String serverURL = "http://example.com/";
const unsigned int timeoutInSeconds = 8;

void setup() {
  Serial.begin(115200);
  delay(1000);
  WiFi.mode(WIFI_STA);
  WiFi.begin(APSSID.c_str(), APPassphrase.c_str());
  Serial.print("Connecting.");
  int i = 0;
  while ((WiFi.status() != WL_CONNECTED) && (WiFi.status() != WL_NO_SSID_AVAIL) && (WiFi.status() != WL_CONNECT_FAILED) && (i < (timeoutInSeconds))) {
          delay(1000);
          Serial.print(".");
          i++;
  }
  Serial.println();
  if ((WiFi.status() == WL_CONNECTED)) {
    Serial.println("Connected to the WiFi network");
    Serial.println(getPayloadFromServer("nikos"));
      } else {
    Serial.println("Failed to connect to the WiFi network");
  }
  WiFi.disconnect();
}

void loop() {
}

String getPayloadFromServer(String message) {
  HTTPClient http;
  Serial.println("Attempting server connection.");
  http.begin(serverURL);
  Serial.println("Sending GET request.");
  int httpCode = http.GET();
  if (httpCode > 0) {
    Serial.println("Server reached and responded to GET request.");
    if (httpCode == HTTP_CODE_OK) {
      Serial.print("Server responded with requested payload: ");
      String payload = http.getString();
      Serial.println(payload);
      http.end();
      return payload;
    } else {
      Serial.println("Server error: Server did not provide the requested payload.");
    }
  } else {
    Serial.println("Server could not be reached or server did not reply to GET request.");
  }
  http.end();
  return unavailableString;
}
```

Just edit APSSID and APPassphrase to your situation and this will cause a Guru Meditation Error with this trace:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4015e078  PS      : 0x00060e30  A0      : 0x800d3130  A1      : 0x3ffb1e50  
A2      : 0x3ffb1ecc  A3      : 0x00000000  A4      : 0x00000013  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d4d42  A9      : 0x3ffb1e50  
A10     : 0x3ffb1f00  A11     : 0x3ffb7d84  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffffb  

Backtrace: 0x4015e078:0x3ffb1e50 0x400d312d:0x3ffb1e70 0x400d40c9:0x3ffb1e90 0x400d1bf6:0x3ffb1ec0 0x400d1d5e:0x3ffb1f60 0x4013f7cb:0x3ffb1fb0 0x4008ba0d:0x3ffb1fd0
```

Using ESP Exception Decoder with the above trace, I get this:
```
PC: 0x4015e078: HTTPClient::connected() at C:\Users\Nikos\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp line 381
EXCVADDR: 0x00000010

Decoding stack results
0x4015e078: HTTPClient::connected() at C:\Users\Nikos\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp line 381
0x400d312d: HTTPClient::disconnect() at C:\Users\Nikos\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp line 347
0x400d40c9: HTTPClient::begin(String) at C:\Users\Nikos\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp line 336
0x400d1bf6: getPayloadFromServer(String) (C:\Users\Nikos\OneDrive\Workspace\Electronics\Projects\TimeSquare2\Code\DeviceBeforeVersioning\LOLIN32 at ESP32)\HelpingScripts\BasicHttpClientMod/BasicHttpClientMod.ino line 40
0x400d1d5e: setup() (C:\Users\Nikos\OneDrive\Workspace\Electronics\Projects\TimeSquare2\Code\DeviceBeforeVersioning\LOLIN32 at ESP32)\HelpingScripts\BasicHttpClientMod/BasicHttpClientMod.ino line 25
0x4013f7cb: loopTask(void*) at C:\Users\Nikos\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15
0x4008ba0d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141
```

Now... somehow this problem doesn't happen if the String type parameter is removed from the getPayloadFromServer function. Allow me to offer the same sketch with only two changes in two lines to reflect the parameter removal from the getPayloadFromServer function:
```cpp
#include <WiFi.h>
#include <HTTPClient.h>

const String APSSID = "NikosOnePlus5";
const String APPassphrase =  "nikosnikos";
const String unavailableString = "unavailable";
const String serverURL = "http://example.com/";
const unsigned int timeoutInSeconds = 8;

void setup() {
  Serial.begin(115200);
  delay(1000);
  WiFi.mode(WIFI_STA);
  WiFi.begin(APSSID.c_str(), APPassphrase.c_str());
  Serial.print("Connecting.");
  int i = 0;
  while ((WiFi.status() != WL_CONNECTED) && (WiFi.status() != WL_NO_SSID_AVAIL) && (WiFi.status() != WL_CONNECT_FAILED) && (i < (timeoutInSeconds))) {
          delay(1000);
          Serial.print(".");
          i++;
  }
  Serial.println();
  if ((WiFi.status() == WL_CONNECTED)) {
    Serial.println("Connected to the WiFi network");
    Serial.println(getPayloadFromServer());
  } else {
    Serial.println("Failed to connect to the WiFi network");
  }
  WiFi.disconnect();
}

void loop() {
}

String getPayloadFromServer() {
  HTTPClient http;
  Serial.println("Attempting server connection.");
  http.begin(serverURL);
  Serial.println("Sending GET request.");
  int httpCode = http.GET();
  if (httpCode > 0) {
    Serial.println("Server reached and responded to GET request.");
    if (httpCode == HTTP_CODE_OK) {
      Serial.print("Server responded with requested payload: ");
      String payload = http.getString();
      Serial.println(payload);
      http.end();
      return payload;
    } else {
      Serial.println("Server error: Server did not provide the requested payload.");
    }
  } else {
    Serial.println("Server could not be reached or server did not reply to GET request.");
  }
  http.end();
  return unavailableString;
}
```

This only changes lines 25 and 36 from the previous sketch and it executes fine!

Another way to avoid the issue is to fallback to commit b70737d276e4f815edd28422ee76c77bb2167cc1, right before httpclient lib was updated to 1.2.

Can anybody offer any possible explanation why this happens?

I hope it's clear that I'm not looking for workarounds. In the above example sketch passing the String object as a parameter is useless (the message variable is not even used anywhere in the function). But in my full app I really need to pass a String object to the function. I am well aware that there are other ways to access a piece of text from the function (setting a global var, using char arrays etc).

I really can't understand how passing a String object that is not even used affects httpclient...

Thanks

chegewara, 2018-11-21:
Did you try to initialize `String var = "nicos";` before you pass it to function? Maybe `nicos` passed as argument is treated as const char* instead of string.

mouridis, 2018-11-21:
@chegewara Yes, I did. And just for triple-checking's sake, I tried again after your suggestion. Replacing
```
Serial.println(getPayloadFromServer("nikos"));
```
with this
```
String var = "nikos";
Serial.println(getPayloadFromServer(var));
```
makes no difference. I still get the Guru Meditation Error.

I also tried to declare var as a const String. Same result.

From the Exception Decoder output and the fact that the sketch works with the older commit just before httpclient update, it seems the reason for the error has to do with the new httpclient. But I cannot figure out why the new httpclient has a problem with a String that is not even passed to it.

Maybe @Jeroen88 who has clear understanding of the library inner workings can shed some light...

mouridis, 2018-11-22:
@Jeroen88 The response from my server is a 60 byte JSON. It should not be a problem. Besides, when I do not pass an irrelevant String to the function, this sketch works fine for payloads bigger than 1Kb. So, this is not the issue.

I suppose you suggest the StreamHttpClient example to handle a possibly big payload, but this is not the case as my payload is very small. Besides, the StreamHttpClient example is identical to my sketch. Its changes start after calling the GET() method, while the crash happens earlier in begin(). If I run the example as-is, I'm sure it will have no problem running. But  as soon as I convert the example to a function that accepts a String parameter it will crash.

Regarding enabling debugging, if you mean adding:
```cpp
Serial.setDebugOutput(true);
```
after:
```cpp
Serial.begin(115200);
```
I tried it but the Serial output is exactly the same.
(I 've never used this before so please bear with me if I'm doing something wrong)

mouridis, 2018-11-22:
Did any of you by any chance tried to flash my sketch in your dev boards and confirm the issue? 

Please feedback if you did because at this point I start questioning my sanity. This error makes no sense...  maybe I'm doing something profoundly wrong in my setup.

If I could get a "you're not the only one seeing this" it would help me decide the direction where I should search more for the solution. Thanks!

mouridis, 2018-11-22:
OK - important new piece of info:
If I move the construction of the http object outside of the function (meaning changing its scope to global), the function can accept the String parameter with no Guru Meditation Error.

I just move the following line:
```cpp
HTTPClient http;
```
from inside the getPayloadFromServer() function to the top of the sketch just above setup().

This is kind of a reasonable workaround but it makes the http object global, occupying memory all the time.

To sum up, for this example sketch:
- httpclient 1.1 works fine all the time no matter if the object is constructed inside the function, or outside of it and no matter if there are String parameters to the function or not. Of course, it has other issues that httpclient 1.2 solves.
- httpclient 1.2 works fine if the object is constructed outside the function, no matter if the function has a String parameter or not.
- httpclient 1.2 works fine if the object is constructed inside the function and the function has no String parameter.
- httpclient 1.2 causes a Guru Meditation Error if the object is constructed inside a function which has a String parameter

I think it's best to keep this issue open for two reasons:
1. It's still an issue if you don't want your httpclient objects to be global.
2. In my main app which is kind of huge, httpclient 1.2 causes the same error when the httpclient object is constructed inside a function, even without passing a String to that function. It seems the String parameter is only one way to easily reproduce the issue but it's not the only one. I temporarily solved the issue in my app by using the workaround of this comment.  

I hope all this helps to find the reason.

Jeroen88, 2018-11-22:
@mouridis I could reproduce your error. An uninitialized pointer in the HTTPClient seemed to be the problem. I created PR #2097 to solve it. The first sketch you provided runs fine with this fix.

mouridis, 2018-11-22:
@Jeroen88 You are the man!

I confirm this not only fixes the sketch above but also my main app issue. Thanks!

I guess your PR will be merged soon so I'm closing this.

Jeroen88, 2018-11-22:
@mouridis that is good news :). Maybe it is better to reopen the issue and keep it open until the PR is merged?

mouridis, 2018-11-23:
I hear you...

mouridis, 2018-11-27:
Addressed in commit 0640964879ffd420530503aa64524a6c1c8076c6

closed 2018-11-27 by mouridis

csteamengine, 2021-03-02:
I apologize for commenting on this thread so long after a fix was implemented, but I can't seem to get past this issue. I have the most recent versions of HTTPClient and have confirmed the `nullptr` update @Jeroen88 made is present in my library, but I am still having a `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`. Any help you could provide would be greatly appreciated!

It seems that the issue is happening in the `login()` function, which redirects after logging in. I have made sure to follow the redirect, but when I do it crashes. 

*P.S. I apologize for the state of my code, it is still in the very early stages!*

Hardware: 
- ESP32 Dev Module
- Macbook Pro 2015

Error Info
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
I (33) wifi:wifi driver task: 3ffb57ec, prio:23, stack:3584, core=0
I (167) wifi:wifi firmware version: 9ebfd5b
I (168) wifi:config NVS flash: enabled
I (168) wifi:config nano formating: disabled
I (168) wifi:Init data frame dynamic rx buffer num: 32
I (173) wifi:Init management frame dynamic rx buffer num: 32
I (178) wifi:Init management short buffer num: 32
I (182) wifi:Init dynamic tx buffer num: 32
I (187) wifi:Init static rx buffer size: 1600
I (191) wifi:Init static rx buffer num: 16
I (194) wifi:Init dynamic rx buffer num: 32
I (288) wifi:mode : sta (fc:f5:c4:2f:a2:a4)
I (315) wifi:new:<4,0>, old:<1,0>, ap:<255,255>, sta:<4,0>, prof:1
I (316) wifi:state: init -> auth (b0)
I (338) wifi:state: auth -> assoc (0)
I (351) wifi:state: assoc -> run (10)
I (434) wifi:connected with Kirk To Enterprise, aid = 4, channel 4, BW20, bssid = 28:80:88:46:88:84
I (434) wifi:security type: 3, phy: bgn, rssi: -48
I (520) wifi:pm start, type: 1

I (633) wifi:AP's beacon interval = 204800 us, DTIM period = 2
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x4016024b  PS      : 0x00060a30  A0      : 0x800d1bb5  A1      : 0x3ffb1d00  
A2      : 0x3ffcb99f  A3      : 0x400856e0  A4      : 0x00000000  A5      : 0x3ffc5d70  
A6      : 0x3ffc5d84  A7      : 0x0000002e  A8      : 0xfccbb43f  A9      : 0xfccbb43f  
A10     : 0xfccbb43e  A11     : 0x00000010  A12     : 0x3ffba948  A13     : 0x00000001  
A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x00000009  EXCCAUSE: 0x00000009  
EXCVADDR: 0x3ffcb9a3  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x4016024b:0x3ffb1d00 0x400d1bb2:0x3ffb1d20 0x400d1cc5:0x3ffb1d40 0x400d1e45:0x3ffb1d70 0x400d1762:0x3ffb1dc0 0x400d3d46:0x3ffb1df0 0x400d3dba:0x3ffb1e20 0x400d3f4b:0x3ffb1e70 0x400d3f6f:0x3ffb1e90 0x400d14e6:0x3ffb1eb0 0x400d161b:0x3ffb1f80 0x400d65aa:0x3ffb1fb0 0x400897a2:0x3ffb1fd0
```

Sketch:
```C#
#include <stdint.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <Wire.h>
#include <HTTPClient.h>
#include <esp_log.h>

#define SERIAL_BAUDRATE 115200
#define WIFI_SSID "My WIFI Network"
#define WIFI_PASS "My Wifi Password"
#define WARZONE_USERNAME "My Call of Duy Account"
#define WARZONE_PASS "COD Password"

HTTPClient http;
String XSRF;
const char* getXSRFURL = "https://profile.callofduty.com/cod/login";
const char* postLoginURL = "https://profile.callofduty.com/do_login?new_SiteId=cod";
const char* getProfileURL = "https://www.callofduty.com/api/papi-client/crm/cod/v1/identities";
String warzoneUsername = WARZONE_USERNAME;
String warzonePass = WARZONE_PASS;

void setup() {
  esp_log_level_set("*", ESP_LOG_DEBUG);
  Serial.begin(115200); 
  begin();
  updateXSRF();
  login(); //This function call is where the ESP32 crashes and reboots.
}

void loop() {
}

void begin(){
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
}

void updateXSRF(){
  if(WiFi.status()== WL_CONNECTED){
    http.begin(getXSRFURL);
    
    int httpResponseCode = -1;
    int timeout = 0;

    while(httpResponseCode < 0 && timeout < 10){
      httpResponseCode = http.GET();
      delay(50);
      timeout++;
    }
    
    if (httpResponseCode>0) {
      String payload = http.getString();
      XSRF = parseXSRF(payload);
      
    }else{
      XSRF = "";
    }
    // Free resources
    http.end();
  }
}

String getXSRF(){
  return XSRF;
}

String parseXSRF(String payload){
  int index = 0;
  String token = "";
  if(payload.indexOf("_csrf") > 0){
    index = payload.indexOf("_csrf");
    while(payload.charAt(index) != '='){
      index++;
    }
    index += 2;
    while(payload.charAt(index) != '"'){
      token += payload.charAt(index);
      index++;
    }
  }
  return token;
}

void login(){
  if(WiFi.status()== WL_CONNECTED){
    http.begin(postLoginURL);

    String httpRequestData = getLoginRequestData();
    
    http.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");
    http.addHeader("Accept-Encoding", "gzip, deflate, br");
    http.addHeader("Connection", "keep-alive");
    http.addHeader("Content-Length", String(httpRequestData.length()));
    http.addHeader("Cookie", "XSRF-TOKEN=" + getXSRF() + ";new_SiteId=cod;tfa_enrollment_seen=true;");

    const char * headerKeys[] = {"Location"};

    http.collectHeaders(headerKeys, 1);
              
    int httpResponseCode = http.POST(httpRequestData);

    if(httpResponseCode == 302){
      //TODO
    }

    // Free resources
    http.end();
  }
}

String getLoginRequestData(){
  String output = "username=";
  output += warzoneUsername;
  output += "&password=" + warzonePass;
  output += "&remember_me=true";
  output += "&_csrf="+getXSRF();
  return output;
}
```

me-no-dev, 2021-03-02:
please decode the backtrace with the ESP Exception Decoder :)

csteamengine, 2021-03-02:
> please decode the backtrace with the ESP Exception Decoder :)

Thanks for the tool! I have followed your instructions, but I'm able to determine what the issue is from the extra information provided. Here is the output from ESP Exception Decoder:
```
PC: 0x400d1769: WiFiClient::connect(char const*, unsigned short, int) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/WiFi/src/WiFiClient.cpp line 281
EXCVADDR: 0x00000048

Decoding stack results
0x400d1769: WiFiClient::connect(char const*, unsigned short, int) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/WiFi/src/WiFiClient.cpp line 281
0x400d3d5a: HTTPClient::connect() at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/HTTPClient/src/HTTPClient.cpp line 1111
0x400d3dce: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/HTTPClient/src/HTTPClient.cpp line 576
0x400d3f5f: HTTPClient::POST(unsigned char*, unsigned int) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/HTTPClient/src/HTTPClient.cpp line 504
0x400d3f83: HTTPClient::POST(String) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/libraries/HTTPClient/src/HTTPClient.cpp line 509
0x400d14ee: login() at /Users/gregorysteenhagen/Documents/Arduino/testHTTP/testHTTP.ino line 131
0x400d1623: setup() at /Users/gregorysteenhagen/Documents/Arduino/testHTTP/testHTTP.ino line 28
0x400d65be: loopTask(void*) at /Users/gregorysteenhagen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5/cores/esp32/main.cpp line 32
0x400897a2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
Does that mean anything to you?

Jeroen88, 2021-03-02:
I am not sure if it will solve your problem, but an important issue is that you are trying to connect to a secure server (http**s**) with a WiFiClient. This will _never_  work! You should use a WiFiClientSecure (see [this](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino) example). Using https requires you to pass a root certificate to the WiFiClientSecure and to set the time (like in the example). Further, you should use the new API, so one of the following:

```
    bool begin(WiFiClient &client, String url);
    bool begin(WiFiClient &client, String host, uint16_t port, String uri = "/", bool https = false);
```
instead of the ```::begin()``` method you use now. Please try that, maybe it will solve your problem, or at least it will bring you a step further! You could also try to declare a ```HTTPClient other;``` local to the ```login()```-function, because I am not sure if reusing it will work properly.

Next, if you ```http.setFollowRedirects(HTTPC_FORCE_FOLLOW_REDIRECTS);``` I expect you will not get a HTTP Code 302 since the client already follows redirects for you. You should either clear this flag and redirect yourself by using the _location_ header you requested or have the HTTPClient handle it for you (and get a HTTP code 200 OK or so).

Finally, leave out
```
http.addHeader("Accept-Encoding", "gzip, deflate, br");
```
because HTTP client **can not** handle zipped or inflated data.

csteamengine, 2021-03-04:
Thank you for the help, but I'm not sure if this is an issue with the Call of Duty servers, or something else, but I only get the connection refused message with the HTTPClientSecure, even after updating the certificate to the DigiCert one.
```
Waiting 10s before the next round...
[HTTPS] begin...
[HTTPS] GET...
[HTTPS] GET... failed, error: connection refused
```

I have tried this same GET request with Python on a Raspberry Pi and it seems to be working just fine, but it doesn't require any certificate configuration, so that is what I'm thinking I did wrong.

Is there any specific process that I could be getting wrong when capturing the certificate?

Thanks in advance!

Jeroen88, 2021-03-04:
Did you set the ESP time? Certificates have a validity period that is checked using system time.

The right certificate shoud be
```
-----BEGIN CERTIFICATE-----
MIIElDCCA3ygAwIBAgIQAf2j627KdciIQ4tyS8+8kTANBgkqhkiG9w0BAQsFADBh
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMSAwHgYDVQQDExdEaWdpQ2VydCBHbG9iYWwgUm9vdCBD
QTAeFw0xMzAzMDgxMjAwMDBaFw0yMzAzMDgxMjAwMDBaME0xCzAJBgNVBAYTAlVT
MRUwEwYDVQQKEwxEaWdpQ2VydCBJbmMxJzAlBgNVBAMTHkRpZ2lDZXJ0IFNIQTIg
U2VjdXJlIFNlcnZlciBDQTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB
ANyuWJBNwcQwFZA1W248ghX1LFy949v/cUP6ZCWA1O4Yok3wZtAKc24RmDYXZK83
nf36QYSvx6+M/hpzTc8zl5CilodTgyu5pnVILR1WN3vaMTIa16yrBvSqXUu3R0bd
KpPDkC55gIDvEwRqFDu1m5K+wgdlTvza/P96rtxcflUxDOg5B6TXvi/TC2rSsd9f
/ld0Uzs1gN2ujkSYs58O09rg1/RrKatEp0tYhG2SS4HD2nOLEpdIkARFdRrdNzGX
kujNVA075ME/OV4uuPNcfhCOhkEAjUVmR7ChZc6gqikJTvOX6+guqw9ypzAO+sf0
/RR3w6RbKFfCs/mC/bdFWJsCAwEAAaOCAVowggFWMBIGA1UdEwEB/wQIMAYBAf8C
AQAwDgYDVR0PAQH/BAQDAgGGMDQGCCsGAQUFBwEBBCgwJjAkBggrBgEFBQcwAYYY
aHR0cDovL29jc3AuZGlnaWNlcnQuY29tMHsGA1UdHwR0MHIwN6A1oDOGMWh0dHA6
Ly9jcmwzLmRpZ2ljZXJ0LmNvbS9EaWdpQ2VydEdsb2JhbFJvb3RDQS5jcmwwN6A1
oDOGMWh0dHA6Ly9jcmw0LmRpZ2ljZXJ0LmNvbS9EaWdpQ2VydEdsb2JhbFJvb3RD
QS5jcmwwPQYDVR0gBDYwNDAyBgRVHSAAMCowKAYIKwYBBQUHAgEWHGh0dHBzOi8v
d3d3LmRpZ2ljZXJ0LmNvbS9DUFMwHQYDVR0OBBYEFA+AYRyCMWHVLyjnjUY4tCzh
xtniMB8GA1UdIwQYMBaAFAPeUDVW0Uy7ZvCj4hsbw5eyPdFVMA0GCSqGSIb3DQEB
CwUAA4IBAQAjPt9L0jFCpbZ+QlwaRMxp0Wi0XUvgBCFsS+JtzLHgl4+mUwnNqipl
5TlPHoOlblyYoiQm5vuh7ZPHLgLGTUq/sELfeNqzqPlt/yGFUzZgTHbO7Djc1lGA
8MXW5dRNJ2Srm8c+cftIl7gzbckTB+6WohsYFfZcTEDts8Ls/3HB40f/1LkAtDdC
2iDJ6m6K7hQGrn2iWZiIqBtvLfTyyRRfJs8sjX7tN8Cp1Tm5gr8ZDOo0rwAhaPit
c+LJMto4JQtV05od8GiG7S5BNO98pVAdvzr508EIDObtHopYJeS4d60tbvVS3bR0
j6tJLp07kzQoH3jOlOrHvdPJbRzeXDLz
-----END CERTIFICATE-----
```

You could try to call ```WiFiClientSecure::setInsecure()```, then certificate verification is skipped. Not secure, but if you can connect with this call you know the certificate is indeed the problem.

csteamengine, 2021-03-04:
Thanks so much for the help, that certificate worked with the COD server. If I used this in my code, how often would I need to reflash the ESP32 with an updated certificate? On that note, how did you obtain this certificate, and is it not recommended to use the `setInsecure()` function?

Thanks again for the help!

Jeroen88, 2021-03-05:
> If I used this in my code, how often would I need to reflash the ESP32 with an updated certificate?

I do not understand your question, you need to reflash the ESP after every change, and after that you can start it over and over again.

> how did you obtain this certificate

That's easy, click on the padlock in your browser and click some options until you find the .pem file.

> ... and is it not recommended to use the setInsecure() function?

The whole idea of https / TLS is to encrypt the exchange of information and to identify at least one of the communicating parties (mostly the client identifies the server). Using this function the server identify is not checked, and could be spoofed.

This issue part of github is intended however to track issues in the libraries for the ESP32. Your questions are general user questions. Please read the examples with the libraries, find information on the internet or use e.g. gitter to get help. If any _library_ issues retain please report them here.