espressif/arduino-esp32 #2094: HTTPUpdate not working. Update.h:44:10: note: candidate expects 2 arguments, 4 provided

Zeni241, 2018-11-22:
1.	I installed arduino-esp32 using library manager in Arduino IDE.
2.	I copy and paste code from here:
        

 [arduino-esp32/libraries/HTTPUpdate/examples/httpUpdate/httpUpdate.ino](url)  In Arduino IDE and compiled. I get following error:

       

        **_ESP32HTTPUpdate:12:24: error: HTTPUpdate.h: No such file or directory_**

3.	I downloaded “arduino-esp32” zip file from github and unzipped two missing libraries, “HTTPUpdate” and “FFat” in my libraries folder _C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries_

4.	Again compiled the sketch and get following error: 

__> Multiple libraries were found for "WiFi.h"
> bool begin(size_t size=UPDATE_SIZE_UNKNOWN, int command = U_FLASH);
> Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
> Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
>_ **_C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update\src/Update.h:44:10: note:   candidate expects 2 arguments, 4 provided
> exit status 1
> Error compiling for board ESP32 Dev Module.__**

I am using esp32 DevKit V1. I don't know what I am doing wrong. I am using httpUpdate in esp8266 successfully.

Jeroen88, 2018-11-22:
What function call is causing the error? If I look in the ESP32 repository at line 44:10 in Update.h I see no function but a #define. 
Please add the exact sketch you are compiling. I suspect that you are calling the ESP32 library using deprecated ESP8266 functions

Zeni241, 2018-11-22:
First of all thanks for your quick response.
Here is complete code I am using:

```
#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

const char *ssid = "Flink";  //ENTER YOUR WIFI SETTINGS
const char *password = "razamiral";
String version="v2"; 

void setup() {

  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
       Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
 
  if ((WiFi.status() == WL_CONNECTED)) {

    WiFiClient client;
    
     //  t_httpUpdate_return ret = httpUpdate.update(client, "http://gintee.com/updater");
         t_httpUpdate_return ret = httpUpdate.update(client, "gintee.com",80,"/updater",version); // This I use successfully with esp8266
    
    // Or:
    //t_httpUpdate_return ret = httpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}
```

And here is the messages in bottom window of Arduino IDE:

```

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.cpp: In member function 'HTTPUpdateResult HTTPUpdate::update(WiFiClient&, const String&, const String&)':

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.cpp:49:27: error: no matching function for call to 'HTTPClient::begin(WiFiClient&, const String&)'

     http.begin(client, url);

                           ^
[... 148 lines cut ...]

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.cpp: In member function 'bool HTTPUpdate::runUpdate(Stream&, uint32_t, String, int)':

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.cpp:319:52: error: no matching function for call to 'UpdateClass::begin(uint32_t&, int&, int&, uint8_t&)'

     if(!Update.begin(size, command, _ledPin, _ledOn)) {

                                                    ^

In file included from C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.h:34:0,

                 from C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\HTTPUpdate\src\HTTPUpdate.cpp:26:

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update\src/Update.h:44:10: note: candidate: bool UpdateClass::begin(size_t, int)

     bool begin(size_t size=UPDATE_SIZE_UNKNOWN, int command = U_FLASH);

          ^

C:\Users\xxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update\src/Update.h:44:10: note:   candidate expects 2 arguments, 4 provided

exit status 1
Error compiling for board ESP32 Dev Module.

```

Zeni241, 2018-11-22:
Thanks again. I have downloaded arduino-esp32-master.zip file and added only two missing libraries,  HTTPUpdate” and “FFat” in my libraries folder C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries
OK now I replace the whole library folder from this downloaded zip file and try again.

Jeroen88, 2018-11-22:
Your welcome!

mouridis, 2018-11-22:
The errors you get about AzureIoT and BLE libraries are because you did not "recursively clone submodules" while cloning the repo with git. You can follow the official instructions for cloning this repo to fix this.

Zeni241, 2018-11-22:
Thanks mouridis, seems  like I can no more avoid learning GitHub , (which I avoided my whole life with utmost passion :) :) ).

Jeroen88, 2018-11-22:
I tried to be ignorant for git too, but once you see the light... ;)

closed 2018-11-23 by Zeni241

Jeroen88, 2018-11-27:
@Zeni241 I created PR #2116 implementing the request headers you mention above.

TLS1000, 2018-12-04:
Hi, I also get the message during compiling: error: HTTPUpdate.h: No such file or directory_**

While it's in the "C:\Users\myname\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPUpdate\src" folder.

Quick and dirty is to copy paste the HTTPUpdate.h file to the sketch folder. But why are other libraries of the ESP32 found, and this one not?

TLS1000, 2018-12-04:
update: before I was updating, or trying to update the ESP-Arduino core using GIT. But recently switched to the Arduino IDE for updating the core. Most of the libraries are part the the stable release, you need to set the  additional board manager url to the 'developer release' for libraries like httpupdate. developer url: 'https://dl.espressif.com/dl/package_esp32_dev_index.json'. So basically I can forget about the GIT-stuff, which I don't mind to be honest.