espressif/arduino-esp32 #2099: Webserver re-instantiation

bot1131357, 2018-11-23:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

g01d10x, 2019-05-25:
Did you ever get any resolution to this? I'm having a similar problem.

bot1131357, 2019-05-27:
I was going to implement a fix, but I never did get around to doing it. It seems like the WebServer classs does not have a proper destructor. 

A quick workaround would be to have the handler functions execute a function pointer which can be modified to access different functions during runtime. Something like this:

```
void serveMainPage()
{/*serve main page*/}

void serveAnotherPage()
{/*serve another page*/}

void (*function_ptr)(void) = &serveMainPage;

void changeFunctionPtr()
{
  function_ptr = &serveAnotherPage;
}
```
Hope this helps.

bot1131357, 2019-08-03:
Good bot. I think this should be closed given the lack of interest.

closed 2019-08-03 by bot1131357

atanisoft, 2019-08-03:
@bot1131357 ESPAsyncWebServer can do exactly what you're wanting to do without reinit or changing the server config once started. I'd suggest looking at that instead of the WebServer.h code.

bot1131357, 2019-08-10:
@atanisoft Noted. Thanks!