espressif/arduino-esp32 #2103: esp_partition_read function can only read 10 to 20 bytes

akshar001, 2018-11-24:
I am trying to do some ota work in my project and needed to read the partition where ota has currently written the data.
i am using arduino 1.8.5 in mac os.
when i try to use function esp_partition_read at the last parameter as length of the data to be read when i put of more than 50 bytes esp32 restarts itself but it works fine for 10-20 bytes length.
I actually need 1024 bytes to read at same time.
'
 uint32_t ret = esp_partition_read(update, 0,
                                         read_data,20);
'

This is an decoded guru meditation error code.

Decoding 7 results
0x4008ec72: vPortYieldFromInt at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portasm.S line 595
0x4008eb7b: _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portasm.S line 206
0x400e3b18: uartBegin at /Users/akshar/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c line 368
0x4008eafc: _frxt_int_enter at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portasm.S line 119
0x400e3b18: uartBegin at /Users/akshar/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c line 368

lbernstone, 2018-11-26:
The size needs to match the size assigned to your read_data buffer, which you do not show in your code.  Also, the return value for esp_partition_read is an esp_err_t, not uint32_t.

lbernstone, 2018-11-26:
Looking at the definition (https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/spi_flash.html#_CPPv218esp_partition_readPK15esp_partition_t6size_tPv6size_t), the buffer should be a void pointer.  Try making it explicit:
`esp_err_t ret = esp_partition_read(update, 0, (void*) read_data, 1024);`
Please include the error along with the backtrace.

akshar001, 2018-11-27:
@lbernstone i have already tried passing void pointer casting but it works like same, and it works very fine when size is 10 using this same code that i have poasted i can read whole partition when size is 10, i have tried it all it's esp32 side problem that's why i have posted the issue!

lbernstone, 2018-11-28:
If you post a minimal sketch, I will take a look and see if I can help.

akshar001, 2018-12-03:
I found the problem,
Though it's not en error of any kind but it's an error,
The actual program in which i was running the function was crashing,
But after that i tried minimal sketch and everything runs fine.
There were 3-4 freertos task running and ram would be very low though this shouldn't be done but it crashes at that time.

closed 2018-12-03 by akshar001