espressif/arduino-esp32 #2104: Best way to use I2C reliably?

rleinfellner, 2018-11-24:
Running on an M5Stack there seem to be many comments about I2C not working reliably in a Write/Read scenario on the ESP32 with Wire.h.
I am getting all sorts of odd errors myself, including heap corruption issues on multiple reads.
Having gone through the issues and proposed fixes I am unsure what the optimal course of action is to get reliable reads (in my case from an ADS1115).

I am using the code below, and it works 80% of the time, but thats not reliable enough, it also crashes the ESP32 with heap corruption randomly;

I get the following errors, with no obvious pattern;
Read of (2) bytes read 0 bytes Failed Err=7 lastError=2, text=ACK
Read of (2) bytes read 0 bytes Failed Err=7 lastError=5, text=BUSY
Read of (2) bytes read 0 bytes Failed Err=7 lastError=6, text=MEMORY

After a number of these it ends up with;

E (45079) boot: Assert failed in multi_heap_realloc, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c:236 (head != NULL)

Code used to read register (NB, register write works 100%, so its a almost certainly a read issue)

```
bool WireReadRegisterWordLE(uint8_t vAddress, uint8_t vRegister, uint16_t* vWord) {
	uint8_t	tSize = sizeof(uint16_t);
	uint8_t tCount = 0;
	Wire.beginTransmission(vAddress);
	Wire.write(vRegister);
	uint8_t err = Wire.endTransmission(false);
	if (err == I2C_ERROR_CONTINUE) {
		tCount = Wire.requestFrom(vAddress, tSize);
		if (Wire.lastError() == 0) {
			if (Wire.available() == 2) {
				*vWord = Wire.read();
				*vWord <<= 8;
				*vWord |= Wire.read();
				return	true;
			}
		}
	}
	Serial.printf("Error: Read of (%d) bytes got %d bytes Err=%d lastError=%d, text=%s\n", tSize, tCount,err, Wire.lastError(),
		Wire.getErrorText(Wire.lastError()));
	return	false;
}
```

So I need some advice on the best course of action please?

1.) Use the Wire.h fix by @stickbreaker, if so is there a doc which outlines the entire process for which files to replace & also any possible knock-on effect please?
2.) Dump wire.h and use ESP32's own functions, before investing the time to do this it would be good to know if the issues are also found below the hood, ie. there is no point doing this as it won't fix it.
3.) Use another method to read I2C based register variables on devices.
4.) Give up on the ESP32 until it becomes more reliable/stable?

stickbreaker, 2018-11-24:
This error pattern you are seeing is being generated by your sensor, or another device on the bus.

The ACK error is telling you the sensor did not respond.

BUSY is telling you the i2c peripheral saw activity on the Bus it did not generate.

MEMORY is telling you the internal `Wire()` 128 byte tx or rx buffer is overflowing.

Looking at your code example, Why are you using ReSTART coding?

The ADS1115 doesn't require it?

Post your i2c code.

I have never seen the i2c library/hal cause an Heap overflow.

Chuck.

stickbreaker, 2018-11-24:
Additionally, understand all ReSTART transactions are Buffered(not executed) until a transaction with a STOP is encountered. At which point all of the Buffered transactions are executed in order.

Chuck.

rleinfellner, 2018-11-26:
Many thanks for your comments, with your help managed to track it down to 2 issues.
1.) Memory corruption, caused by a 2nd task trying to set a I2C DAC which sometimes just happen to hit the first task trying to read the ADC. Was under the impression (probably wrongly) that I2C handed resource contention between tasks internally. When I removed the DAC write, the memory panics went away.
2.) Abandoned batching the write (Restart) followed by a write, and it works ok now. 

Guess I spent too much time programming PICs where I have more control over the I2C bus as its not trying to multitask.

All good now

closed 2018-11-26 by rleinfellner