espressif/arduino-esp32 #2110: is ESP32 really able to connect to Mobile Phone's Hotspot ?

anandpanchal, 2018-11-27:
Board:							 ESP32 Dev Module
Core Installation/update date:			2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 7

Hello Everyone,

i am trying to connect the esp32 to mobile phone's hotspot. but esp is not connecting to the hotspot and rebooting everytime.
does the esp has capability to connect to the mobile phone's hotspot ?
Thanks

mouridis, 2018-11-27:
Yes, I use it with my phone's hotspot very often with no issue whatsoever... Do you mean that you can connect to any other access point but have trouble connecting specifically to your phone's hotspot?

anandpanchal, 2018-11-27:
thanks @mouridis  for the reply.
i have trouble connecting to my phone's hotspot and i went for the poc in a organization. there i faced the same problem, my esp did not connect to the company's wifi router. i am surprised why this is happening. may i know the exact problem/reason behind this behavoiur

mouridis, 2018-11-27:
Sorry, your problem is still unclear to me. Let me rephrase: using your current code have you ever successfully connected to any wifi access point?

anandpanchal, 2018-11-27:
yes, i am able to connect to wifi router using my current code. but not able to connect to the mobile phone's hotspot. it reboot every time when i try to connect to the mobile phone's hotspot

mouridis, 2018-11-27:
In your deleted comment, I thought you mentioned that you had also trouble connecting to your company network.

Anyway... I would suggest to offer more information if you would like to get some help. For example, the code you are using, the error message you are getting (use the ESP32 Exception Stack Trace Decoder and paste the decoded error results here) and details about your mobile hotspot (phone OS, phone OS version, wifi protocol - is it 802.11b/g/n, is it 5GHz or 2.4GHz? security type - is it WPA/WPA? is it open?, is it hidden SSID? do you know which WiFi channel it uses?).

anandpanchal, 2018-11-27:
wifi scan result

scan done
17 networks found
1: honor (-53) 
2: PYRIOS (-55)*
3: office1110 (-69)*
4: RBM  (-69)*
5: Transon (-75)*
6: travoGrapher (-75)*
7: Keystone (-76)*
8: Iprosonic (-79)*
9: Suruchi (-80)*
10: Keystone-Tripleplay2.4GHz (-84)*
11: Vishal1102 (-85)*
12: WirelessNet (-87)*
13: DIRECT-GMSHAMSHULmsBZ (-87)*
14: Office@1110 (-88)*
15: DASAN_GONT128eb2 (-89)*
16: Keystone-Tripleplay2.4GHz (-91)*
17: HP-Print-5C-LaserJet Pro MFP (-93) 
error logs when trying to connect to the mobile phone's hotspot

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Connecting to honor
Monitoring interrupts: 
95 Mode IR
95 relay switch on at power up
4175 Exit detected
4175 Relay off after timeout
5661 Exit detected
10707 WiFi status Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060430  A0      : 0x800d5e84  A1      : 0x3ffb1e00  
A2      : 0xbf6bbae1  A3      : 0xbf6bbadd  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000001  A9      : 0x3ffb1dfc  
A10     : 0x3ffba4bc  A11     : 0x00000000  A12     : 0x3ffb21f0  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbf6bbae1  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffff2  

Backtrace: 0x400014dc:0x3ffb1e00 0x400d5e81:0x3ffb1e10 0x400d5ee1:0x3ffb1e30 0x400d2043:0x3ffb1e50 0x4013e29e:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Connecting to honor
Monitoring interrupts: 
143 Mode IR
143 relay switch on at power up
2184 Exit detected
2184 WiFi status Disconnected
4224 Exit detected
4224 Relay off after timeout
6264 Exit detected
8304 Exit detected
10344 Exit detected
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x4015b0b4  PS      : 0x00060034  A0      : 0x80080f2e  A1      : 0x3ffc0bb0  
A2      : 0x00002b88  A3      : 0x000005dc  A4      : 0x3ffcccd4  A5      : 0x000000ff  
A6      : 0x00000000  A7      : 0x3ffcccd4  A8      : 0x80081282  A9      : 0x3ffc0b90  
A10     : 0x00aa0d88  A11     : 0x00000000  A12     : 0x000003e8  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00060c23  SAR     : 0x00000014  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40082683  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4015b0b4

Backtrace: 0x4015b0b4:0x3ffc0bb0 0x40080f2b:0x3ffc0bd0 0x40080fbd:0x3ffc0bf0 0x4008176d:0x3ffc0c10 0x40082680:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

phone OS : Oreo 8.0.0
security : WPA2 PSK
wifi protocol

marius240, 2019-03-07:
a bit late, but make sure your mobile hotspot isnt 5ghz esp32 only supports 2.4ghz

closed 2019-08-15 by stale[bot]

hardmax, 2022-07-03:
HI,

I have the same issue with Android HostSpot (Xiaomi with andorid 12). 
Does anyone have a solution to the problem?. 

I use the latest version of Arduino IDE and ESP32 for arduino, If I connetc to router all its fine, only the problem is with HotSpot. Some Androids works some other not works.

Regards

bbahavati, 2022-09-09:
We all wrongly think that it is the issue with Phone or SIM or board or some other thing.  The main issue here is Greedy of Corporates.  If they allow you to DIY, they can't sell $1 items as big things and sell for $100.  So, they formed a Dacoit gangs and corrupt your networks and create issues like some inherent connection issues. Really it is man made by Corporates.

anu-maheshwari, 2023-05-16:
Hello, I am facing the same issue did, in my case, I using the esp IDF framework, and in the same way I am able to connect to a WIFI router but not to my phone's hotspot.
And I tried all combinations of my WIFI hotspot it still doesn't work.
And I am getting this error 

 disconnected code 201: WIFI_REASON_NO_AP_FOUND

hardmax, 2023-05-16:
Hi.You need to know that some routers add an space in his ssid. Some after a some other before of name.Please check if you have this kind of router. When we use a PC we select ssid and don't see that space in ssid name.Another is that esp32 is not compatible with new password and new wifi technologies..review that please.Regards

ciranjiiv, 2024-01-06:
HI, im using esp32 and trying to connect to my phone samsung a54 5g and it says like this 

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[22] Connecting to CiranjiivA545G

BUT not connecting in my phone , i see in phone no connection to my hotspot .. how to solve this i need help ASAP..

hardmax, 2024-01-06:
Hi,

Review Android Version. I see that andorid versión lower 10 has problems. Only connect the firts time a not other time, I dont know why.

If you put credential manualy, you has to review, some Android versions add an space to SSID (at end),

Regards

Max

Enviado el: sábado, 6 de enero de 2024 08:44
Para: espressif/arduino-esp32 ***@***.***>
CC: Max Sanchez Torres ***@***.***>; Comment ***@***.***>

ciranjiiv, 2024-01-06:
Hi. 
THIS MY CODE COMBINED WITH WIFI CLIENT CODE  .
THIS CODE IS FOR ERGONOMIC CHAIR
WE ARE USING ESP32 to connect to blynk app

code:
#include <WiFi.h>
#include <BlynkSimpleEsp32.h>

#define TRIG_PIN_1 22
#define ECHO_PIN_1 23
#define TRIG_PIN_2 25
#define ECHO_PIN_2 26
#define TRIG_PIN_3 27
#define ECHO_PIN_3 14
#define BUZZER_PIN_1 21
#define BUZZER_PIN_2 19
#define TOUCH_BUZZER_PIN 5
#define TOUCH_SENSOR_PIN 32
#define TOGGLE_SWITCH_PIN 12
#define BLYNK_PRINT Serial
#define BLYNK_TEMPLATE_ID "TMPL6jk2SHLPI"
#define BLYNK_TEMPLATE_NAME "IDP PROJECT 1"

char auth[] = "iTTpBosLaIeBrYFML8GKfMr5TyWCYI-O";
char ssid[] = "Ciranjiiv A54 5G";
char pass[] = "ciranjiiv";

WidgetTerminal terminal(V1);

bool systemOn = false;
bool objectDetected1 = false;
bool objectDetected2 = false;
bool objectDetected3 = false;
bool touchSensorPressed = false;

void setup() {
  Serial.begin(115200);
  
  // Connect to WiFi
  connectToWiFi();
  Blynk.begin(auth, ssid, pass);

  // Add a delay to allow sensors to stabilize
  delay(500);

  pinMode(TRIG_PIN_1, OUTPUT);
  pinMode(ECHO_PIN_1, INPUT);
  pinMode(TRIG_PIN_2, OUTPUT);
  pinMode(ECHO_PIN_2, INPUT);
  pinMode(TRIG_PIN_3, OUTPUT);
  pinMode(ECHO_PIN_3, INPUT);
  pinMode(BUZZER_PIN_1, OUTPUT);
  pinMode(BUZZER_PIN_2, OUTPUT);
  pinMode(TOUCH_BUZZER_PIN, OUTPUT);
  pinMode(TOUCH_SENSOR_PIN, INPUT);
  pinMode(TOGGLE_SWITCH_PIN, INPUT_PULLUP);
}

void loop() {
  Blynk.run();
  
  bool toggleSwitchState = digitalRead(TOGGLE_SWITCH_PIN);

  if (toggleSwitchState == HIGH) {
    if (!systemOn) {
      Serial.println("System is ON, You're ready to go");
      systemOn = true;
    }

    float distance1 = getDistance(TRIG_PIN_1, ECHO_PIN_1);
    checkAndTriggerBuzzer(distance1, 10, objectDetected1, BUZZER_PIN_1);

    float distance2 = getDistance(TRIG_PIN_2, ECHO_PIN_2);
    checkAndTriggerBuzzer(distance2, 10, objectDetected2, BUZZER_PIN_1);

    float distance3 = getDistance(TRIG_PIN_3, ECHO_PIN_3);
    checkAndTriggerBuzzer(distance3, 10, objectDetected3, BUZZER_PIN_2);

    bool touchSensorState = digitalRead(TOUCH_SENSOR_PIN);
    checkAndTriggerTouchBuzzer(touchSensorState, touchSensorPressed, TOUCH_BUZZER_PIN);
  } else {
    if (systemOn) {
      Serial.println("System is OFF, Thank You!!!");
      systemOn = false;
      turnOffBuzzers();
      resetDetectionFlags();
    }
  }

  delay(100);
  sendDataToBlynk();
  delay(100);
}

void connectToWiFi() {
  Serial.println("Connecting to Wi-Fi...");
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting...");
  }

  Serial.println("Connected to Wi-Fi");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

float getDistance(int triggerPin, int echoPin) {
  digitalWrite(triggerPin, LOW);
  delayMicroseconds(2);
  digitalWrite(triggerPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(triggerPin, LOW);
  long duration = pulseIn(echoPin, HIGH);
  float distance = duration * 0.034 / 2;
  return distance;
}

void triggerBuzzer(int buzzerPin, bool beep) {
  if (beep) {
    tone(buzzerPin, 1000);
  } else {
    noTone(buzzerPin);
  }
}

void sendDataToBlynk() {
  Blynk.virtualWrite(V2, "Sensor 1: " + String(objectDetected1) +
                          ", Sensor 2: " + String(objectDetected2) +
                          ", Sensor 3: " + String(objectDetected3) +
                          ", Touch Sensor: " + String(touchSensorPressed));
}

void checkAndTriggerBuzzer(float distance, float threshold, bool &objectDetected, int buzzerPin) {
  if (distance < threshold && !objectDetected) {
    Serial.println("Movement detected, You're not in the correct position");
    objectDetected = true;
    triggerBuzzer(buzzerPin, true);
  } else if (distance >= threshold && objectDetected) {
    Serial.println("You're back in the correct position");
    objectDetected = false;
    triggerBuzzer(buzzerPin, false);
  }
}

void checkAndTriggerTouchBuzzer(bool touchSensorState, bool &touchSensorPressed, int touchBuzzerPin) {
  if (touchSensorState == HIGH && !touchSensorPressed) {
    Serial.println("Touch sensor detected, You're in the correct position");
    touchSensorPressed = true;
    triggerBuzzer(touchBuzzerPin, false);
  } else if (touchSensorState == LOW && touchSensorPressed) {
    Serial.println("Touch sensor not detected, You're not in the correct position");
    touchSensorPressed = false;
    triggerBuzzer(touchBuzzerPin, true);
  }
}

void turnOffBuzzers() {
  noTone(BUZZER_PIN_1);
  noTone(BUZZER_PIN_2);
  noTone(TOUCH_BUZZER_PIN);
}

void resetDetectionFlags() {
  objectDetected1 = false;
  objectDetected2 = false;
  objectDetected3 = false;
  touchSensorPressed = false;
}

BUT WHEN UPLOADED THE CHAIR SENSORS ARE NOT FUNCTIONING AS THOUGH THE CODE FOR CHAIR NOT UPLOADED,
THE BLYNK APP CANNOT DETECT THE ESP32 AFTER connecting , as in app it says offline,

PLS HELP, 
I DONT KNOW IF CODE IS WRONG OR WHY?

I NEED HELP USING BLYNK APP.. CAN HELP?

THANK YOU

ciranjiiv, 2024-01-06:
my android version is 14

hardmax, 2024-01-08:
HI

ESP32 has no connection to 5G only workd in 2.4ghz

Regards

Max

Enviado el: sábado, 6 de enero de 2024 10:23

HankB, 2025-03-10:
Good morning,
Just adding a "me too." I was trying to use a Pixel 6a (Android 15) WiFi hotspot for testing and the ESP32 is unable to associate with the phone.  I have another phone that can connect to the hotspot and the ESP32 is able to connect to other (Archer and Deco) APs so it appears to be some incompatibility between the WiFi stacks on the EXP32 vs. Android.

me-no-dev, 2025-03-10:
There are two possibilities:
- Network uses WEP encryption (should be upgraded to WPA/WPA2)
- AP is 5GHz (ESP32 has only 2.4GHz radio)

Can you run the WiFi Scan example and check if it can see the AP and what is the security?

HankB, 2025-03-10:
Hi @me-no-dev,
On a Linux host, `iwlist scanning shows the following for the 2.4G channel

```text
          Cell 06 - Address: E0:3F:49:70:B6:5B
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=70/70  Signal level=-40 dBm  
                    Encryption key:on
                    ESSID:"hidden"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 1288ms ago
                    IE: Unknown: 000668696464656E
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030106
                    IE: Unknown: 050402030000
                    IE: Unknown: 2A0100
                    IE: Unknown: 2F0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 32040C121860
                    IE: Unknown: 2D1AFC191BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1606080400000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F0101
                    IE: Unknown: DD090010180200F02C0000
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: 3D1606080000000000000000000000000000000000000000
```

If you point me to the scan example, I'll give it a shot. I don't see it at <https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#examples>. (I'm using PIO with Arduino libraries via VS Code, but I'm OK with the esp-idf libraries too.)

me-no-dev, 2025-03-10:
@HankB the example is here: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino

If the Linux logs is for the same AP, then it looks OK to be connectable. 2.4GHz WPA2_PSK. I will also suggest to enable verbose core debug level (in the board menu). It will explain at least what is the error when you are trying to connect

HankB, 2025-03-10:
Thanks, got it. And I need to add that the output of `iwlist scaning` above is from the test AP, not the Andriod phone hotspot. The scan program produces:

```text
Scan start
Scan done
5 networks found
Nr | SSID                             | RSSI | CH | Encryption
 1 | drone                            |  -34 |  2 | WPA2                   <<< Android hotspot
 2 | Nacho24                          |  -50 |  1 | WPA2                   <<< Asus AP
 3 | TMOBILE-2AE1                     |  -91 |  1 | WPA2+WPA3
 4 | Dredze                           |  -94 |  1 | WPA2
 5 | Verizon_9XT7LV                   |  -96 | 11 | WPA2
```

And the detail for `drone` (the Android hotspot)

```text
          Cell 07 - Address: 22:C1:9F:33:82:72
                    Channel:2
                    Frequency:2.417 GHz (Channel 2)
                    Quality=70/70  Signal level=-28 dBm  
                    Encryption key:on
                    ESSID:"drone"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 8528ms ago
                    IE: Unknown: 000564726F6E65
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030102
                    IE: Unknown: 050401020000
                    IE: Unknown: 0706555320010D80
                    IE: Unknown: 200100
                    IE: Unknown: 23021400
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2D1A2D001BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1602000000000000000000000000000000000000000000
                    IE: Unknown: 7F080400080000000040
                    IE: Unknown: FF1C230108001A0080202002000D009E000C0000F5FFF5FF391CC7711C07
                    IE: Unknown: FF072404000000FCFF
                    IE: Unknown: FF0E260003A4FF27A4FF4243FF6232FF
                    IE: Unknown: FF0427000000
                    IE: Unknown: DD1E00904C0408BF0C3270810FFAFF0000FAFF0020C0050002000000C302006C
                    IE: Unknown: DD0A0010180200001C000000
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
```

I'll see what I can do to collect some debug data.

HankB, 2025-03-10:
@me-no-dev with `build_flags = -DCORE_DEBUG_LEVEL=5` in `platform.ini` The ESP emits the following when trying to connect to `drone`.

```text
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    36][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    46][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    57][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.
[    89][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   167][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   165][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
[   184][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   446][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   715][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   984][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  1253][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2522][E][WiFiUdp.cpp:185] endPacket(): could not send data: 118
loop() WiFi lost connection
[  3551][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  3820][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  4089][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  4358][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  4627][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
```

`loop() WiFi lost connection` is generated by the application and this just continues. A succesful connection emits:

```text
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[     7][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    37][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    46][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    57][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.
[    90][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   167][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   165][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
[   184][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[   243][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Channel: 1, Auth: WPA2_PSK
[   255][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   270][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got New IP:192.168.20.201
[   278][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   286][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.20.201, MASK: 255.255.255.0, GW: 192.168.20.1
[   450][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   451][V][WiFiGeneric.cpp:363] _arduino_event_cb(): STA Disconnected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Reason: 8
[   472][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[   479][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[   545][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Channel: 1, Auth: WPA2_PSK
[   558][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[   576][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got Same IP:192.168.20.201
[   584][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[   591][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.20.201, MASK: 255.255.255.0, GW: 192.168.20.1
[   745][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   746][V][WiFiGeneric.cpp:363] _arduino_event_cb(): STA Disconnected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Reason: 8
[   767][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[   774][W][WiFiGeneric.cpp:1062] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[   854][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Channel: 1, Auth: WPA2_PSK
[... 1 lines cut ...]
[... cut ...]
[  1458][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Nacho24, BSSID: 10:c3:7b:55:05:e8, Channel: 1, Auth: WPA2_PSK
[  1471][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  1495][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got Same IP:192.168.20.201
[  1503][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  1510][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.20.201, MASK: 255.255.255.0, GW: 192.168.20.1
                        time() returns 1741644908
```

The last line is from the application.

me-no-dev, 2025-03-11:
@HankB please post a snippet of your WiFi connect code. The static IP messages are very strange and unexpected. There is no CONNECTED or DISCONNECTED event either.

me-no-dev, 2025-03-11:
@HankB also if you are using PlatformIO, please switch to pioarduino, because the version supported by PlatformIO is no longer supported on our end (both ESP-IDF and Arduino)

HankB, 2025-03-13:
@me-no-dev 

> post a snippet of your WiFi connect code. 

<https://github.com/HankB/ESP32-Arduino-DS18B20_MQTT>

me-no-dev, 2025-03-13:
@HankB you are calling just `WiFi.begin();` without giving it any SSID or PASS to connect to

HankB, 2025-03-13:
Lines 9-19 in `src/app_wifi.cpp` explain this.

```text
/******************************************
    Associate with a WiFi AP. The ESP32 stores the most recently used
    credentials and in that case the call to WiFi.begin(); should work.

    If there is a need to associate with a different AP or this ESP
    never used WiFi, the correct call is

    const char* ssid = "REPLACE_WITH_YOUR_SSID";
    const char* password = "REPLACE_WITH_YOUR_PASSWORD";
    WiFi.begin(ssid, password);
******************************************/
```

This behavior is not well described in Espressif's API description at <https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html#sta-configuration> but does work.

me-no-dev, 2025-03-13:
I suggest you first test it with giving it credentials. There are many things at play, especially if you are switching core versions.