espressif/arduino-esp32 #2114: SD Mount failed

scottw-finao, 2018-11-27:
What you are trying to do.
Trying to connect an SD card or card module to a Wemos D1 mini ESP-32 clone.

Describe your system( Hardware, computer, O/S, core version, environment)
Tried in both platformio and the arduino ide. Linux based development platform with clion.
Dev board is a clone of the Wemos D1 mini ESP32
SD adapter I tried is a generic (no branding) "MicroSD card adapter" with level shifter. (5v vcc, spi bus)
I also have a microsd card (SD size) adapter that I have direct wired which I also tried and got the same result.
Using default MOSI/MISO/SCLK/SS pins. On direct adapter, I wired it exactly per the image on the SD library page documentation.

What is failing:

setup init code to 'begin' the SD card returns error:

```
Starting SD card reader...
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed!
```

relevant code to initialize the SD card (called in setup() {} ):

```
bool initSDCard() {
    Serial.println("Starting SD card reader...");
    // enable pullup on MOSI pin
    pinMode(MOSI,INPUT_PULLUP);
    if(!SD.begin()){
        Serial.println("Card Mount Failed!");
        return false;
    } else {
        return true;
    }
}
```

Note I added the pinMode per suggestion in a similar issue where someone was having a 'first time only' error included the fail error. This happens for me every time. I have tried adding a 1K pull up resistor to the SD MISO. (as suggested in the other issue. The suggestion that issue said to do the code pullup on pin 23 which is MOSI on the esp32. But I have tried it with and without both.)
I have also tried multiple SD cards with multiple sizes and formats but always get the same result. (2Gb FAT, 4Gb FAT32, 8GB [not sure on format but card has been used on arduino/atmel based projects with same format], 32Gb [not sure on format]

treii28, 2018-11-28:
As an update, I dug around in my toy chest and finally found the smaller profile SD card adapter. (I actually ordered another like it when I couldn't find it prior to starting this project as I would prefer to use the smaller size)
I also re-formatted one of the SD cards from scratch to be 100% certain it was set at fat16 and managed to get it to function.  However, I did and tried the reformatting first and it still did not work on either the other adapter or the direct-connection to the sdcard. So I would still be curious to know of any possible things to try to get the other one working as well as that type of device has one of the spring in/out SD slots where these little ones do not. (on some future projects I may wish to have the other style adapter people are more used to that let you know by the 'click' when they are properly inserted and where no part of the card is sticking out in a way that could be broken)

atsteich, 2018-12-09:
@scottw-finao maybe this can help you: https://github.com/espressif/arduino-esp32/issues/2171

closed 2019-08-15 by stale[bot]

he3117, 2020-05-11:
for anybody else that has the same problem with the micro SD module you should connect the VCC pin of micro SD card module to pin Vin of ESP32 not 3.3 V out.

lbernstone, 2020-05-12:
SD cards run at 3V3.  Do not apply 5V unless your SD module has a power regulator on board to reduce to 3V3.

pubudu123, 2021-08-19:
I have same issue. I tried 5volt to vcc . Perfectly worked !

aimeiz, 2022-02-13:
Same problem.
SD card wo0rks with Arduino with AVR processor, also with STM32 processors but doesn't work with ESP32.
Card Mount Failed.
Problem with SPI communication?

aimeiz, 2022-02-13:
Finally I got it working.
My SD modules have voltage converters for signals. Designed to use with 5V Arduinos. I Made my own connector for SD Card just by soldering pins directly to SD to micro converter. Then connected directly to proper ESP32 ports, GND and 3.3V Supply.
It started working immediately. 
I was confused, since one of micro SD socket bulit into display worked with STM32 - 3.3V supply and logic, but not working with ESP32. Also 3V supply and logic. Maybe levels were on the edge. OK for STM32 but not for ESP32.
Now I know.

EdBrisebois, 2022-02-15:
I had a lot of grief getting it to run. 
Turned out the jumper lengths were too long.
I used about 5cm and works fine now.
3.3v supply, direct ~3v logic, and no level shifting.