espressif/arduino-esp32 #2118: Windows fails to download tooolchain via get.py

DouglasWWolf, 2018-11-27:
Using Windows 10, trying to install the ESP32 toolchain.

Running get.py results in the error below.     Is the ESP32 toolchain download broken??

-- Doug

D:\Program Files (x86)\Arduino\hardware\espressif\arduino-esp32-master\tools>python get.py >x.txt
Traceback (most recent call last):
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\contrib\pyopenssl.py", line 453, in wrap_socket
    cnx.do_handshake()
  File "D:\Pythons\Python36-32\lib\site-packages\OpenSSL\SSL.py", line 1907, in do_handshake
    self._raise_ssl_error(self._ssl, result)
  File "D:\Pythons\Python36-32\lib\site-packages\OpenSSL\SSL.py", line 1639, in _raise_ssl_error
    _raise_current_error()
  File "D:\Pythons\Python36-32\lib\site-packages\OpenSSL\_util.py", line 54, in exception_from_error_queue
    raise exception_type(errors)
OpenSSL.SSL.Error: [('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\connection.py", line 344, in connect
    ssl_context=context)
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\util\ssl_.py", line 344, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\contrib\pyopenssl.py", line 459, in wrap_socket
    raise ssl.SSLError('bad handshake: %r' % e)
ssl.SSLError: ("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\Pythons\Python36-32\lib\site-packages\requests\adapters.py", line 449, in send
    timeout=timeout
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "D:\Pythons\Python36-32\lib\site-packages\urllib3\util\retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /igrr/mkspiffs/releases/download/0.2.3/mkspiffs-0.2.3-arduino-esp32-win32.zip (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "get.py", line 148, in <module>
    get_tool(tool)
  File "get.py", line 98, in get_tool
    r = requests.get(url)
  File "D:\Pythons\Python36-32\lib\site-packages\requests\api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "D:\Pythons\Python36-32\lib\site-packages\requests\api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "D:\Pythons\Python36-32\lib\site-packages\requests\sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "D:\Pythons\Python36-32\lib\site-packages\requests\sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "D:\Pythons\Python36-32\lib\site-packages\requests\adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /igrr/mkspiffs/releases/download/0.2.3/mkspiffs-0.2.3-arduino-esp32-win32.zip (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),))

D:\Program Files (x86)\Arduino\hardware\espressif\arduino-esp32-master\tools>

bertmelis, 2018-11-27:
why don't you use get.exe?

DouglasWWolf, 2018-11-27:
How do I use get.exe?   I was following directions I found here:

http://exploreembedded.com/wiki/Arduino_Setup_for_ESP32

DouglasWWolf, 2018-11-27:
Running get.exe results in the following error:   Is there a way to resolve this?   I never encountered these issues when I downloaded the ESP-8266 toolchain previously.

System: Windows, Info: Windows-10-10.0.17134
Platform: i686-mingw32
Tool xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip already downloaded
Extracting xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Tool esptool-2.5.0-windows.zip already downloaded
Extracting esptool-2.5.0-windows.zip
Downloading mkspiffs-0.2.3-arduino-esp32-win32.zip
Traceback (most recent call last):
  File "get.py", line 139, in <module>
    sys_name = 'Windows'
  File "get.py", line 89, in get_tool
    print('Downloading ' + archive_name)
  File "site-packages\requests\api.py", line 70, in get
  File "site-packages\requests\api.py", line 56, in request
  File "site-packages\requests\sessions.py", line 488, in request
  File "site-packages\requests\sessions.py", line 609, in send
  File "site-packages\requests\adapters.py", line 497, in send
requests.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:590)
Failed to execute script get

bertmelis, 2018-11-27:
you don't know how you doubleclick?

You could also just use the instructions in the readme of this repo.

Please close this issue after you find out.

DouglasWWolf, 2018-11-27:
I'll start over using the instructions in the README in the repository.  Thanks for the tip.   I'll close this if I manage to get the install to work.

bertmelis, 2018-11-27:
(deleted my comment)

DouglasWWolf, 2018-11-27:
Thanks for the tip.  I'll check out PlatformIO.

I don't have my ESP32 board yet (it should arrive in the mail later today), but when I fire up the Arduino IDE, it appears to support ESP32.

Thanks *much* for pointing me in the right direction.

--- Doug

closed 2018-11-27 by DouglasWWolf