espressif/arduino-esp32 #2126: What is the version of esp-idf that Release 1.0.0 depends on?

AdrienFelon, 2018-11-29:
I have a crash in esp-idf (very likely my fault...) with the following stack trace:

```
0x400fa3ee: handle_dhcp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/apps/dhcpserver.c line 1035
0x40102371: udp_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/udp.c line 417
0x40104f1d: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/ipv4/ip4.c line 705
0x4010aec6: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/netif/ethernet.c line 176
0x400f97ed: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 474
```

and I would like to look at the source of esp-idf. I'm using release 1.0.0 of arduino-esp32. Would anyone know what version/tag of esp-idf I should at be looking at?

Thanks!

chegewara, 2018-11-29:
Those errors are not from esp-idf. It's from 3rd party library lwip.

AdrienFelon, 2018-11-29:
If I am not mistaken, the sources of this 3rd party library are in the esp-idf repo:

https://github.com/espressif/esp-idf/tree/master/components/lwip/apps/dhcpserver

AdrienFelon, 2018-11-29:
Got it: crash happens in 3rd party lib. Thanks for clarifying. I'm still hoping to get clues of what is wrong by looking at proper version of source code of dhcpserver.c. I doubt that the master branch of esp-idf is what I should be looking at since line 1035 does not look like a line that would throw.

```c
        case DHCPS_STATE_NAK://4
#if DHCPS_DEBUG   // line 1035
            DHCPS_LOG("dhcps: handle_dhcp-> DHCPD_STATE_NAK\n");
#endif
```

lbernstone, 2018-11-29:
If you read the thread referenced [here](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md#compilation-errors), it explains how to use git to find the version of esp-idf used.

AdrienFelon, 2018-11-30:
Thanks. That is really useful. FYI, I also found an API call esp_get_idf_version()  that returned "v3.2-dev-39-gaaf12390" in my case. The actual format of this string is unclear to me, but it contains the esp-idf commit hash (aaf1239) I got with the method using git in arduino-esp32 (method from thread referenced previous).

closed 2018-11-30 by AdrienFelon