espressif/arduino-esp32 #2127: WiFi reconnection issue with reason: 6 - NOT_AUTHED

DimaRU, 2018-11-29:
### Hardware:

Hardware:
Board:	DOIT ESP32 DEVKIT V1  
IDE name:	Platform.io with VSCode  
Platform: Espressif32 1.5.0  
Flash Frequency:	40MHz  
Upload Speed:	115200  

### Description:

I am having issues with reconnecting to a WiFi network if it is disconnected. (For example when the WiFi AP rebooted).

### Debug messages:

```
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4 :STA_CONNECTED
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 172.16.1.68, MASK: 255.255.255.0, GW: 172.16.1.1
[WiFi-event] event: 7 :STA_GOT_IP
[1232] Connected to WiFi
[1233] IP: 172.16.1.68
[1233]
    ___  __          __
   / _ )/ /_ _____  / /__
  / _  / / // / _ \/  '_/
 /____/_/\_, /_//_/_/\_\
        /___/ v0.5.4 on ESP32

[1238] Connecting to userv.bdm:8080
[1262] Ready (ping: 4ms).
Blynk connected
White: 585->563
Brigthness[1] 585->563 942 t=100
White: 563->527
duty[1] = 942
Brigthness[1] 563->527 809 t=100
duty[1] = 809
White: 527->516
Brigthness[1] 527->516 771 t=100
duty[1] = 771
White: 516->509
Brigthness[1] 516->509 748 t=100
White: 509->505
duty[1] = 748
Brigthness[1] 509->505 734 t=100
duty[1] = 734
Preferences written
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 6 - STA_AUTHMODE_CHANGE
[WiFi-event] event: 6 :STA_AUTHMODE_CHANGE
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[D][WiFiClient.cpp:452] connected(): Disconnected: RES: -1, ERR: 113
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0 :WIFI_READY
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0 :WIFI_READY
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0 :WIFI_READY
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[143863] Connecting to userv.bdm:8080
[E][WiFiClient.cpp:213] connect(): lwip_connect_r: 118
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0 :WIFI_READY
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[WiFi-event] event: 0 :WIFI_READY
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[WiFi-event] event: 2 :STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
[WiFi-event] event: 4 :STA_CONNECTED
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 172.16.1.68, MASK: 255.255.255.0, GW: 172.16.1.1
[WiFi-event] event: 7 :STA_GOT_IP
[148864] Connecting to userv.bdm:8080
[148880] Ready (ping: 5ms).
Blynk connected
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 6 - NOT_AUTHED
[WiFi-event] event: 5 :STA_DISCONNECTED
[D][WiFiClient.cpp:452] connected(): Disconnected: RES: -1, ERR: 113
[164362] Connecting to userv.bdm:8080
[E][WiFiClient.cpp:213] connect(): lwip_connect_r: 118
```

### Temporary workaround:
```cpp
void setup()
{
  // Debug console
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  .
  .
  .
 }

static void WiFiEvent(WiFiEvent_t event, WiFiEventInfo_t info)
{
  if (event == SYSTEM_EVENT_STA_DISCONNECTED) {
    if (info.disconnected.reason == 6) {
      Serial.println("NOT_AUTHED reconnect");
      WiFi.reconnect();
    }
  }
}
```

CircuitSetup, 2019-07-24:
Thanks for this - did you ever figure out why this was happening?

DimaRU, 2019-07-25:
The problem still exists. I did not try to understand, in my opinion, it should be done by developers.

Board:	DOIT ESP32 DEVKIT V1
IDE name:	Platform.io with VSCode
Platform: Espressif32 1.9.0
Blynk: v0.6.1
AP: Securifi Almond +

[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 172.16.1.68, MASK: 255.255.255.0, GW: 172.16.1.1
																<--- AP rebooted
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 6 - NOT_AUTHED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 8 - STA_LOST_IP
.
.
.
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 172.16.1.68, MASK: 255.255.255.0, GW: 172.16.1.1
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 6 - NOT_AUTHED
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 8 - STA_LOST_IP

CircuitSetup, 2019-07-25:
If you're getting NOT_AUTHED that means that there's something in your router that is preventing it from giving your ESP an IP. Maybe it's disconnecting before the handshake is complete even. Right before that you're getting NO_AP_FOUND, so the ESP isn't even seeing the AP, until it does for a bit.

DimaRU, 2019-07-25:
Thanks for the comment, John. 
I'll clarify the problem. 
My ESP establishes the connection then gets IP address and drop the connection immediately after that. The problem is related to authentication on Data Link layer. 
According to the documentation on ESP32, Reason: 6 - NOT_AUTHED described as:
Class-2 frame received from a non-authenticated STA.
For the ESP32 Station, this reason is reported when it is received from the AP.

closed 2019-10-07 by stale[bot]

bmedici, 2025-03-01:
Sorry but I'm still having this error, disconnecting just after connection, with error 6. Any idea ?