espressif/arduino-esp32 #2131: WiFi.h WiFiClient instance cause ESP32 to freeze randomly

YordanYanakiev, 2018-11-30:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows  8.1

### Description:
I am using the example provided with the ESP32 SDK about WiFiServer to send constantly data to a client.
Anyway in a few hundreds of lops the loop() just freeze for indefinitely long, doing nothing.
I have put a delay(100);, but still the issue is reprocuting on all ESP32 boards I have.
Every 100-500 loops it freeze for at least 2 seconds up to 11 seconds.

### Sketch:  
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>

WiFiServer * wifiServer;

WiFiClient * clientX;

unsigned long rBuffer[ 65 ];

void setup() 
{
       Serial.begin( 115200 );
       wifiServer = new WiFiServer( 20800 );
	
	WiFi.onEvent( WiFiEvent );
	
	WiFi.mode( WIFI_AP );
	
	WiFi.softAP( "lVision" );
	
	wifiServer->begin();
	serverX = IPAddress( 192, 168, 100, 2 );
	clientX = new WiFiClient();
}

void loop()
 {
        rBuffer[ 64 ] = (unsigned long )'d'; // this is data

	if( clientX->connect( serverX, 20800 ) )
	{
		if( clientX->connected()  )
		{
			Serial.println( millis() );
			clientX->write( ( const uint8_t* )rBuffer, sizeof( rBuffer ) );

			clientX->stop();
			clientX->flush();
		}
	}
}

void WiFiEvent( WiFiEvent_t event )
{
	switch( event )
	{
		case SYSTEM_EVENT_AP_START:
		{
			IPAddress ipA = IPAddress( 192, 168, 100, 1 );
			IPAddress ipS = IPAddress( 255, 255, 255, 0 );
			WiFi.softAPConfig( ipA, ipA, ipS );
		}
		break;
		
		default:
		break;
	}
}

```

### Debug Messages:
```
```

asetyde, 2018-11-30:
Also I 've report some esp32 freeze when I check Wifi status or some network operations, udp reply or mysql query

YordanYanakiev, 2018-11-30:
I don't check neither the wifi status, nor i do network operations, nor database stuff. 
I got nothing attached to the board.
Actually I do nothing into the mail loop. Just sending some random data, and yet - it stuck.

asetyde, 2018-11-30:
I also use OTA , random data is sent sometimes

YordanYanakiev, 2018-12-01:
?

lbernstone, 2018-12-01:
me-no-dev has done some work on the WiFi queue since the 1.0.0 release.  Can you test with the [pre-release version](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) and see if it is better?  Also, don't close the session if it is not necessary.  There's a fair amount of overhead establishing a TCP session.

YordanYanakiev, 2018-12-02:
All done! It just stucking once per reboot seems like so far, and it's current performance is just great! Thanks dev team!

copercini, 2018-12-02:
Closed as solved =)

closed 2018-12-02 by copercini