espressif/arduino-esp32 #2137: Barely installing this board causes sketches for unrelated board to fail to compile

teo1978, 2018-12-02:
I installed the ESP32 cores on the Arduino IDE via the Board Manager.

Then I opened a sketch that uses another board, the ESP8266 Thing Dev from SparkFun, which is one of the board in a separate package: https://github.com/esp8266/Arduino.

This sketch used to complie just fine. The old board is still selected, I am trying to compile the sketch (which I haven't touched) for the same ESP8266 as before. That is, I am **not even using** the ESP32 core I just installed.

Yet I get compile errors like this:

```
Error compiling for board SparkFun ESP8266 Thing Dev.
fork/exec /home/teo/.arduino15/packages/esp32/tools/esptool/2.3.1/esptool: no such file or directory
Error compiling for board SparkFun ESP8266 Thing Dev.
```

Basically, merely installing the ESP32 board backage has messed with the preexisting ESP8266 package that was previously installed and worked fine before. Obviously I should be able to have both installed ans working.

teo1978, 2018-12-02:
Also, when removing the ESP32 board, a NullPointerException showed up (though they seem to have been removed succesfully):

[image]

lbernstone, 2018-12-02:
I understand your frustration, but the esp32 board package installs nothing outside of the .arduino15/packages/esp32 directory.  Any interference with another board's core files is entirely an arduino IDE issue.  I regularly use esp8266 and esp32 to compile without issues.
Good luck

teo1978, 2018-12-03:
Please have a loot at the response from the Arduino team:
[https://github.com/arduino/Arduino/issues/8258#issuecomment-443641224](https://github.com/arduino/Arduino/issues/8258#issuecomment-443641224)

From what I understand, there is a fundamental design flaw in the Arduino IDE, but also an error on your side, or at least a trivial workaround that you could apply to easily avoid the issue completely.

lbernstone, 2018-12-03:
Ok, I'll take a look.

closed 2019-08-15 by stale[bot]