espressif/arduino-esp32 #2139: WiFi connection drops [WiFiClient.cpp:443] connected(): RES: -1

Aietes, 2018-12-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		latest
IDE name:						Platform.io with Visual Studio Code
Flash Frequency:					?40Mhz?
PSRAM enabled:                                       no
Upload Speed:					921600
Computer OS:                                           Mac OSX

### Description:
I'm trying to connect and stay connected to WiFi. Code that used to run flawlessly now fails with error [WiFiClient.cpp:443] connected(): RES: -1. I'm not sure what is causing it, but I'm using the PubSubClient library to send and receive MQTT message.

I'm connecting to WiFi in the standard way:

```cpp
bool myWiFi::connect(String *ssid, String *pass)
{
  Serial.print("Connecting to WiFi");

  WiFi.mode(WIFI_STA);
  while (WiFi.status() == 255)
    ; //wait till WiFi is available

  if (!WiFi.reconnect() )
  {
    WiFi.begin(ssid->c_str(), pass->c_str());
  }

  unsigned long millis_rto = millis();
  unsigned long millis_cto = millis();
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    Serial.print(".");
    if (millis() - millis_rto >= RETRY_TIMEOUT)
    {
      //reset timer
      millis_rto = millis();
      //start new try
      Serial.print("\nConnecting to WiFi");
      WiFi.begin(ssid->c_str(), pass->c_str());
    }
    if (millis() - millis_cto >= CONN_TIMEOUT)
    {
      Serial.println("\nWiFi connection timed out, open AP.");
      getConfig(ssid, pass);
      return false;
    }
    delay(500);
  }

  Serial.print("\nConnected to WiFi with IP ");
  Serial.println(WiFi.localIP());
  return true;
}
```

The connection works, the error message appears after successful connection and receiving an IP. 
I know because I'm tracing the WiFi events to find out what's going on with WiFi.onEvent(WiFiEvent). I'm getting the expected sequence of events:

```
Connecting to WiFi
WiFi Event: 0
WiFi Event: 2
WiFi Event: 5
Connecting to WiFi.
WiFi Event: 4
WiFi Event: 7
Connected to WiFi with IP 192.168.178.57
```

But then the following error occurs:

```
[E][WiFiClient.cpp:443] connected(): RES: -1
```
Despite this error message, a connection to the WiFi exists, as the time is successfully updated and an MQTT package send. When I call the mqtt.loop() funtion though, this error appears in every tick.

Again, this code worked for months now, this error message is new after updating the libraries to the latest version. I'm looking for help on how to find the cause for this error, I'm not sure where to start.

Aietes, 2018-12-02:
I'm realizing that this must be connected to the PubSubClient making a connection to an MQTT server via WiFiClient, so probably this error has little to do with the actual connection to the AP. I'll reference this issue in PubSubClient.

luc-github, 2018-12-02:
check this : https://github.com/espressif/arduino-esp32/pull/2140

closed 2018-12-03 by me-no-dev

macedolfm, 2019-01-28:
Did anyone manage to fix the WiFiClientSecure library as this is a VERY important feature for any reasonable secure IoT application? For what I have tested the Client is still very unstable, gets disconnected during publish for no reason, may get stuck connecting for minutes, crashes the ESP32 sending long (over 1024 bytes), get stuck if MQTT server stops, get stuck if WiFi stops and all sort of unexpeceted problems in a very random fashion.

I´ve been trying everything for the last couple days and letting the MCU run overnight to collect error data and just can´t figure out what is wrong (I am NOT a inner guts expert), tried with "arduino-esp32-master" version 1.0.0 & 1.0.1, even tried [this guy tips](http://www.iotsharing.com/2017/08/how-to-use-esp32-mqtts-with-mqtts-mosquitto-broker-tls-ssl.html?m=1) and a ["this partial release"](https://github.com/espressif/arduino-esp32/tree/a38ffe58fe1d73a56295edb0c6051a46932e0175) of WiFiClientSecure but still, lots of trouble.

I managed to addapt the own library example to show case the many problems on a MQTT permanent connection sketch:

```
#include <WiFi.h> 
#include <WiFiClientSecure.h>

#define  ARDUINOJSON_USE_LONG_LONG  1
#define  ARDUINOJSON_USE_DOUBLE     1
#include <ArduinoJson.h>

#define MQTT_MAX_PACKET_SIZE 4096
#include <PubSubClient.h>

#include <SerialDebug.h>

// Lets Encrypt root CA-CRT

const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n" \
  "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
  "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
  "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
  "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
  "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
  "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
  "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
  "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
  "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
  "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
  "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
  "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
  "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
  "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
  "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
  "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
  "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
  "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
  "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
  "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
  "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
  "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
  "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
  "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
  "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
  "-----END CERTIFICATE-----";

const char* foo_4000 = "0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567 [... cut ...]
(E p:^14284 mqttConnect C1) MQTT_CONNECT_OK
(E p:^40006 mqttConnect C1) MQTT_CONNECTING
(E p:^25991 mqttConnect C1) MQTT_CONNECT_OK
(E p:^122003 mqttConnect C1) MQTT_CONNECTING
(E p:^11105 mqttConnect C1) MQTT_CONNECT_OK
(E p:^90011 mqttConnect C1) MQTT_CONNECTING
(E p:^14638 mqttConnect C1) MQTT_CONNECT_OK
```

And just running it for a few min utes you can see that the MQTT has to reconnect over & over to get the job done.

Crying for HELP here, as I am in a tight corner here.

Cheers evryone !

macedolfm, 2019-01-28:
OK, to be fare, the WiFiClient also did NOT perform perfectly, more or less it is as realiable as the WiFiClientSecure, using the above example and replacing:

#include <WiFiClientSecure.h> => #include <WiFiClient.h> 
and
WiFiClientSecure wifi_client; => WiFiClient wifi_client;

I still get lots of errors and some success in publishing to Mosquitto, so I gotta be doing something VERY wrong:

```
(E p:^10001 mqttConnect C1) WIFI_CONNECTING
(E p:^6119 mqttConnect C1) WIFI_CONNECT_OK
(E p:^3882 mqttConnect C1) MQTT_CONNECTING
(E p:^16150 mqttConnect C1) MQTT_CONNECT_OK
(E p:^0004 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^10000 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^6039 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3964 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^11989 loop C1) MQTT_PUBLISHING
(E p:^10000 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^4285 mqttConnect C1) MQTT_CONNECT_OK
(E p:^5718 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^12993 loop C1) MQTT_PUBLISHING
(E p:^4154 loop C1) MQTT_PUB_OK
(E p:^5843 mqttConnect C1) MQTT_CONNECTING
(E p:^5828 mqttConnect C1) MQTT_CONNECT_OK
(E p:^4175 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9997 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0001 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0002 mqttConnect C1) MQTT_CONNECTING
(E p:^3991 mqttConnect C1) MQTT_CONNECT_OK
(E p:^6012 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^10001 loop C1) MQTT_PUB_ERR
(E p:^0001 mqttConnect C1) MQTT_CONNECTING
(E p:^18265 mqttConnect C1) MQTT_CONNECT_ERR
(E p:^0000 mqttConnect C1) MQTT_CONNECTING
(E p:^6718 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3284 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9997 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
(E p:^9998 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^0002 loop C1) MQTT_PUB_OK
(E p:^9999 loop C1) MQTT_PUBLISHING
(E p:^7008 loop C1) MQTT_PUB_OK
(E p:^2991 mqttConnect C1) MQTT_CONNECTING
(E p:^6022 mqttConnect C1) MQTT_CONNECT_OK
(E p:^3981 loop C1) MQTT_PUBLISHING
(E p:^0003 loop C1) MQTT_PUB_OK
```