espressif/arduino-esp32 #2143: compare OTA1 with OTA2 partition, then decide to update or not?

TLS1000, 2018-12-03:
Hi, I got the S3 update program working. But problem is that the ESP32 remains in an endless loop updating it's firmware and restarting.

I see some examples on the web that use MQTT command to control the update, but that adds again complexity.

Could the ESP periodically (and after reboot) download the latest version and compare it with the active OTA? for example compare MD5 calculation of both OTA's? 

 or any better idea's?

Jeroen88, 2018-12-03:
@TLS1000 See PR #2116, that was merged recently

TLS1000, 2018-12-04:
ah, great.
Noob question: Why is the library 'httpupdate' (and some other new libraries) not added to my local library folder when I do the ESP32-core update?
- via GIT GUI I do 'fetch from origin'
- then I execute the 'get.exe' script from the 'tools' folder
- my library folder only contains 17 items, the repository contains 25 already

TLS1000, 2018-12-04:
Solved above using the Arduino IDE boards manager to update.

closed 2018-12-04 by TLS1000