espressif/arduino-esp32 #2144: I am having the same problem. Stable esp code now very seldomly connects to Arris XB6 modem since it was installed.

gaurav0190, 2018-12-03:
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START

_Originally posted by @panther1 in https://github.com/espressif/arduino-esp32/issues/1829#issuecomment-417990979_

lbernstone, 2018-12-28:
1) Add `WiFi.disconnect()` at the beginning of setup()
2) Run python esptool.py erase_flash from the command line. This will wipe all code from the device (including the nvs partition).
3) Follow the instructions for posting an issue at [ISSUE_TEMPLATE](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md)

arizzi, 2018-12-29:
I solved (independently of any master fix) it was a router problem solved rebooting it (while most of other devices were connecting to it with no problems I noticed one laptop was also not able to connect anymore hence I suspected the router)

thehellmaker, 2019-03-01:
Guys, Thanks for keeping it open. It looks like I am facing same issue with some access points

Where does this happen?
1. Happens on my phone Hotspot (One Plus 6, OxygenOS Version: 9.0.4, Build Number: A6000_22_190123)
2. It doesn't happen on my Wifi router though. It happens only on specific access points

Here is the test scenario and how to reproduct
1.  Connect to Access Point
2. Reset the module right after it connects to the access point (Not I am using WiFi.disconnect(true) before Wifi.begin())
3. When it comes back up it goes into infinite AUTH_EXPIRE loop
No amount of resetting the module helps the only thing that helps is closing doing the hotspot and bringing it back up. Its also shown in logs below that it connected only after NO_AP_FOUND event where I turn my hotspot off.

Here are the logs:
```
[I][Test_Wifi.cpp:134] connectToStoredAP(): Connecting to stored credentials ***********
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 4

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - **STA_GOT_IP**
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.43.40, MASK: 255.255.255.0, GW: 192.168.43.233
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 7

[D][Test_Wifi.cpp:109] WiFiEvent(): WiFi connected. IP address: 192.168.43.40
�x␌␎␃���[I][Test_Wifi.cpp:134] connectToStoredAP(): Connecting to stored credentials ***********
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 5

[D][Test_Wifi.cpp:113] WiFiEvent(): WiFi lost connection
[I][Test_Wifi.cpp:134] connectToStoredAP(): Connecting to stored credentials ***********
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 5

[D][Test_Wifi.cpp:113] WiFiEvent(): WiFi lost connection
[I][Test_Wifi.cpp:134] connectToStoredAP(): Connecting to stored credentials ***********
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[... 37 lines cut ...]

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 5

[D][Test_Wifi.cpp:113] WiFiEvent(): WiFi lost connection
[I][Test_Wifi.cpp:134] connectToStoredAP(): Connecting to stored credentials ***********
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 2

[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 0

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 4

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.43.40, MASK: 255.255.255.0, GW: 192.168.43.180
[D][Test_Wifi.cpp:106] WiFiEvent(): [WiFi-event] event: 7

[D][Test_Wifi.cpp:109] WiFiEvent(): WiFi connected. IP address: 192.168.43.40
```

SviFi, 2019-04-26:
**I have exactly the same issue as @thehellmaker describes.** It does not connect to some hotspots, with infinite auth expire loop. And exactly the same thing, if I restart the hotspot - all of the sudden it connects. Actually, it happened to completely different wifi networks similar way. I need a solution ASAP. 

@me-no-dev did you say you made some fixed that could be downloaded? Is it a new version of WiFi.h module? Where is it and how to update it? I am willing to test and report if it works. Thank you!

sansillusion, 2019-04-27:
I had the same symptoms at a friend's place. Turns out his router default
DHCP settings permitted only 10 devices at the same time.

>

asetyde, 2019-04-28:
I've this problems after some time, lease time i think is problem, you can test with static address

kuoyaoming93, 2019-05-03:
I have the same problem. I can scan the ssid, but cant connect. 
I have to get closer to the AP to connect...

SviFi, 2019-05-03:
I had a similar issue. **What helped:** I updated the Arduino's ESP32 in board manager to version 1.0.2 and it fixed the WiFi module behavior! It appears like there was a bug in the previous one. The second thing that helped in my case was a good quality USB cable. It first sounds crazy that USB cable can affect something like that, but indeed it does! In bad quality USB cables, the voltage drop is very high due to high resistance. This can cause the voltage to be too low for ESP32 and its power-hungry WiFi module to function. In such case, it would generate "Brownout detector triggered" warnings in the serial monitor and WiFi may be unable to connect. Replacing it with good quality USB cable immediately resolves this. It is not only about the length (even though typically the shorter the better), but primarily about quality, I had even 3-meter quality cables behave better than 0.5-meter bad quality one. I hope this infomration helps!

Iron24fit, 2019-05-09:
Still this same problem of AUTH_EXPIRE loop . I don't find a reliable way to face it :(.

asetyde, 2019-05-09:
This issue is showed from many users also in some our devices not all , 5% on all.  But I can't find any solution or bug alert on ESP , @me-no-dev

me-no-dev, 2019-05-11:
there are some updates in the 3.2 branch of IDF. I will merge them and rerelease the libs.

kuoyaoming93, 2019-05-13:
My issue: 
- Same code with 2 different ESP32. The older hardware version can scan and connect to wifi. 
The newest version can scan, but it connects only short tange (around 2 meters from AP).

Solution: ESP32 Arduino 1.0.3 Pre-Release
Thank you very much @me-no-dev

me-no-dev, 2019-05-13:
Hey :D it fixed something. Could you all please test 1.0.3-rc1 from the development package manager link?

kuoyaoming93, 2019-05-25:
I test with 2 different ESP32:
- The older hardware version works fine as always.
- The new one improve the range in 1.0.3-rc1. But only in some APs. I think the problem is in the handshake of connection. Because I can scan the APs from long distance.

asetyde, 2019-06-03:
Yes also us seems some connection problems it resolves (sometimes with reboot), but we ve noticed about range is few with some AP, it's strange thing

kuoyaoming93, 2019-06-24:
@me-no-dev There is a new release in ESP-IDF (v3.0.8), could you please build for this version? I will test it and report if this new version fix this issue or not.

Thank you,
YM Kuo

lbernstone, 2019-06-24:
arduino-esp32 is currently on esp-idf 3.2.  When either 3.2.1 or 3.3 is released, the idf will be updated.

a-c-sreedhar-reddy, 2019-06-26:
hey @lbernstone  3.2.2 is released.

kuoyaoming93, 2019-08-17:
I compiled the IDF v3.2.2 by myself and also the **ESP-IDF Pre-release v3.3-rc**. 
But this bug in  **wifi's range** stil exists.

bsayiner, 2019-09-22:
Is there any update about this fix?

nandra, 2019-10-08:
I experienced same. Have 2 esp32 modules and sometimes after flash one or other cannot connect ot wifi (I have WiFi.disconnect(true) and beginning of setup()). Is there some other workaround which can improve this @me-no-dev ? Thanks.

angusgrantBC, 2019-10-16:
I'm getting the same problem on Netgear CG3100D-2 router and an older Apple AirPort Express.

I have tried flashing my current ESP32, bought a new device and got same problem straight away, etc.

Any help on this? It's really holding up a project I wanted to release end of last month.

Here is the end of the logs when trying to get ESP32 to connect to my wireless network

10:52:55.630 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
10:52:57.719 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:52:57.719 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
10:52:59.629 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:52:59.629 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
10:53:01.743 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:53:01.743 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
10:53:03.821 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:53:03.856 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
10:53:06.794 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:53:06.794 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
10:53:07.611 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
10:53:07.644 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 202 - AUTH_FAIL
10:53:26.574 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 17 - AP_STAIPASSIGNED

Is there any more debug info I can add?

angusgrantBC, 2019-10-23:
None of that stuff worked. I searched for so long across so many resources.

I ended up digging around in the ESP code base and happened upon this little nugget

ESP_ERROR_CHECK(nvs_flash_erase());
nvs_flash_init();

This seems to have cleared my "AUTH_EXPIRE" credentials and got my WiFi working again. One caution is that it clears everything in flash. So if you are using preferences library, etc, it all gets wiped.

michelpa, 2019-10-23:
Hello,

I'm sorry, i do not understand your last message. Can you explain a little further please?
Thanks

angusgrantBC, 2019-10-23:
I was experiencing the AUTH_EXPIRE problem when trying to re-connect to my WiFi. Nothing had changed, the ESP had been working for a month or so. All of a sudden, it refused to re-connect to the same WiFi. I don't really know what was causing the AUTH_EXPIRE exception, but perhaps an expired DHCP lease not being released, or something along those lines.

I tried erasing the device as per instructions, but that did not work. I tried calling disconnect on the WiFi connection but it was never connected in the first place so couldn't disconnect, the AutoConnect library has no way to wipe all memorised WiFi networks. Or the std WiFi library

I found the code snippet above which did exactly what I wanted: erased the flash memory where the WiFi details were saved by the AutoConnect library. Basically all the WiFi management libraries all save their config in flash.

I forgot to mention above that you will also the the below include:
#include "nvs_flash.h"

michelpa, 2019-10-23:
Thank you for your response.

Unfortunately that did not work for me:

[WiFi-event] event: 5
Disconnected from WiFi access point
WiFi lost connection. Reason: 2

Too sad!

angusgrantBC, 2019-10-23:
Maybe try a

ESP.restart();

after the flash erase statement

bsayiner, 2019-10-24:
Hello, I can't erase flash because other configuration settings exist.
Do you have any workaround?

angusgrantBC, 2019-10-24:
No, I dont. :(

I spent so long trying to find out a way wipe WiFi details from flash, directly read details from flash, etc.

The only solution I found was to wipe the entire flash area.

Hopefully someone else has a suggestion. There must be a way, I don't know what it is though.

lbernstone, 2019-10-24:
`WiFi.disconnect(false,true)` will delete the saved WiFi connection info.

angusgrantBC, 2019-10-25:
Does that work if you aren't connected to the WiFi network?

I didn't try with the true, false parameters, but WiFi.disconnect(); did nothing to help with my problem

thehellmaker, 2019-10-25:
> `WiFi.disconnect(false,true)` will delete the saved WiFi connection info.

Hi.
This doesn't work. 
But I just updated esp32 platform version to 1.11.0 .Now it doesn't go into infinite loops. It does AUTH_EXPIRE for sometime and then AUTH_FAIL s and then connects. I had this issue until 1.9.0

angusgrantBC, 2019-10-28:
I've only started ESP32 development, so forgive me if this is a simple question.

I can only see 1.0.4  as the latest version for the ESP32 development board option in Arduino IDE, and can see 1.0.4 as the latest release on the espressif GitHub page
https://github.com/espressif/arduino-esp32/releases

Where are you getting version 1.11.0 from?

I assume you are not using the Arduino IDE, and there is some other platform you are using?

lbernstone, 2019-10-28:
[Platform.io](https://github.com/platformio/platform-espressif32) v1.11.0

bsayiner, 2019-11-20:
Does this issue be related to the ADC2 channel? Because Wifi drivers use ADC2 pins

thehellmaker, 2019-12-04:
Update: the issue with hotspot was resolved with 1.11.0 platform io version. But I still have issues with with tp link TL WA 850 RE and goes into infinite AUTH_EXPIRE loop

Tonhom-Tools-Maker, 2019-12-23:
I have this problem too.

I founded that. this problem occurred.  when I use "Serial Monitor" of Arduino IDE 1.8.5.  on. Ubuntu. 16.04 LTS

My (temporary) solution is - - - - > use PuTTY instead Serial Monitor. of. Arduino IDE

Tonhom-Tools-Maker, 2019-12-23:
[image]

lele89m, 2020-01-02:
yep! that work, using putty no problem for me

digitib, 2020-02-09:
Hi Arduino 1.8.10 on Macos Catalina 64
WeMos D1 R32 with ESPRESSIF ESP32-WROOM-32

Same debug as @gaurav0190 with WiFiClientEvents.

Looking forward to understanding, patching or correcting this issue.
Should you have any question or suggestion, please feel free.
Thanks.
TdL

closed 2020-04-25 by stale[bot]

sohtamei, 2020-07-13:
I faced this issue with CH340+ESP32 board (Wemos D1 R32), when USB-UART connection is OFF (RTS=H, DTR=H).
I/F level of CH340 is 5V. When RTS=H & DTR=H, IO0 is 4.1V, EN is 4.1V.
https://github.com/sohtamei/docs/blob/master/images/WenosD1R32_.jpg

When IO0 is 4.1V, the connection error which was reported on this thread occurs. I don't know why IO0=4.1V causes connection error.

lbernstone, 2020-07-13:
[image]

Applying 4.1V power to your esp32 can permanently damage the device.

sohtamei, 2020-07-13:
Hi lbernstone,
Thanks !

When using "Wenos D1 R32" and "ESPDUINO32", there is a following workaround:
Add 330 ohm to 3V3-RST and 3V3-IO0.
[image]

dtmix, 2020-09-27:
Hi, sohtamei and everybody - I stumbled upon the above solution after being at near wit's end on how to solve my ESP32 (its a Wemos D1 R32 dev board) WIFI issue. I am inexperienced in electronics, and would like to ask 2 questions:

1. With the resistors used above, can I still make use of the 3V3 pin?
2. Is there any other solution (e.g. firmware) that has resolved this todate?
(I am trying to deploy my solution to a pilot environment and would prefer to avoid having resistors stick out like this, if possible to avoid)

Thanks so much!

sohtamei, 2020-10-01:
Hi dtmix,
> can I still make use of the 3V3 pin?
Yes, you can use 3V3 pin but you have to keep the register during wifi connection. To do it, you have to connect both the register and your device to one 3V3 pin.

> (I am trying to deploy my solution to a pilot environment and would prefer to avoid having resistors stick out like this, if possible to avoid)
I also want it strongly, but I could not..

Could you try following solution ?

[image]
[image]

dtmix, 2020-10-01:
Dear sohtamei,

THANK YOU so much for the prompt response!! Yes, I see your point! I ended up with similar idea of bending the resistor to the side eventually (but its not pretty) :-)   I do wish there can be a firmware upgrade for this issue, but I guess not ..  Nvm, will conceal this inside the box for final display.

Just to share info also, I think offloading sensor power requests to a breadboard with independent power supply can also make D1 R32 more stable ...

Again thanks and have a good day!!

SilverWx, 2020-11-25:
> Hi dtmix,
>
>
> Yes, you can use 3V3 pin but you have to keep the register during wifi connection. To do it, you have to connect both the register and your device to one 3V3 pin.
>
>
> I also want it [...]

It is a problem present on all wemos R32, because of the electronics design or only with some manufacturer ?

sohtamei, 2020-11-25:
It is a problem present on most of wemos R32 and compatible (clone) board of it, because of the PCB design.

It is very easy to fix this issue, because CH340 supports both 5V I/O and 3.3V I/O. So I hope the PCB has been updated, but I have not found another version of PCB, yet.

SilverWx, 2020-11-26:
> It is a problem present on most of wemos R32 and compatible (clone) board of it, because of the PCB design.
> 
> It is very easy to fix this issue, because CH340 supports both 5V I/O and 3.3V I/O. So I hope the PCB has been updated, but I have not found another version of PCB, yet.

Since IO0 seems to be pull-up by default, it should be possible to connect 3V3 without resistor, isn't it ? (I'm a noob so sorry if I say something horrible xD)

sohtamei, 2020-11-28:
To set ESP32 to flash write mode, writing tool such as Arduino IDE sets RTS=H & DTR=L, then IO0=L. To support the flash write mode, the 330 ohm register is necessary.
[image]

sohtamei, 2020-11-28:
Another version of schematics (same circuit ?)
[image]

danielmweb, 2020-12-29:
I was facing the same issue, which is that my DOIT ESP32 DEVKIT V1 connects to my home network, but it doesn't connect to my phone's mobile hotspot. And after searching for days for a good solution for my problem, the only one I found that worked for me was:

1. Turn off mobile hotspot

2. Run the code that attempts to make the connection

  WiFi.begin(ssid, password);
  Serial.print("Connecting...");
  while(WiFi.status() != WL_CONNECTED){
    delay(500);
    Serial.print(".");
  }

3. Turn on the mobile hotspot when the loop has already started

4. Repeat the past 3 steps whenever you want to connect again.

This was the solution for my case, so I hope it helps someone else as well.

LemarinelNet, 2021-08-31:
I can confirm that the 330 ohm resistor on IO0-GND does the trick !
I also faced the issue : just after flashing, the Wifi did connect correctly (due to CH340 pulling IO0 ?), but after a power cycle, it did not connected anymore. 
Adding the resistor make it work all the time !

blacknax, 2021-11-22:
Yes, it worked for me also! Thank you very much!
I searched a solution for days and suspected my Raspi hostapd to causing the issue because my D1R32 did not connect to it but connected to a different wlan successfully. But after adding the 330 ohm resisitor it immediately worked with Raspberry hostapd.

sohtamei, 2021-11-23:
This issue has been found in M5Atom/M5StickC/TimerCAM/unitV, and we fixed it with FW update of CH552.

https://sohtamei.github.io/docs/esp32AndUsbUartWith5V_IO_Issue2.html
[image]

uboatsalamander, 2022-03-05:
Designers of the D1 R32board did really terrible job interfacing the CH340(used in 5V I/O mode) to the 3.3V only ESP32.
There is another issue that the TXD output of the CH340 goes to RX input of the ESP32 just via current limiting 470R resistor. I measured the voltages on both ends and it is 5V, probably the diode towards VCC which I would expect to be on the ESP32 side is already killed.
1k pulldown after the 470R would have solved this easily and bring the voltage to comfortable level for the ESP... 
Not sure if this also have any impact like the 4.1V on the other pins.

beachmiles, 2022-08-11:
I just got my LOLIN esp32-c3 mini WIFI working by digging this piece of errata from their site. So ANNOYING!!!
```
//You have to add this code after running WiFi.softAP or WiFi.begin for the Wifi to work.
WiFi.setTxPower(WIFI_POWER_8_5dBm);             //https://www.wemos.cc/en/latest/tutorials/c3/get_started_with_arduino_c3.html#wifi
```

rajveer01, 2022-09-25:
I do not know the cause of the problem, but when I removed all the connections to my espe32 except the USB, It worked for me.

Tuckie, 2022-09-30:
> This issue has been found in M5Atom/M5StickC/TimerCAM/unitV, and we fixed it with FW update of CH552.
> 
> https://sohtamei.github.io/docs/esp32AndUsbUartWith5V_IO_Issue2.html [image]

I just wanted to verify that this fix seems to have worked for me! the Atom Lite was unusable with wifi and the BLE tracker enabled. Made a custom cable and flashed the CH552 with the provided bin and now I'm only getting 2% packet loss.

Xylopyrographer, 2022-10-04:
@Tuckie : Wondering, was it only required to reflash the CH552 or did you also have to do the hardware mod? Also, would you know when your Atom Lite was manufactured? I have a bunch of ATOM MATRIX units & trying to figure if this reflash is required for those. Many thanks.

[3 more replies]