espressif/arduino-esp32 #2145: Latency in receiving notifications from BLE server

Umar-Yaqoob, 2018-12-03:
### Hardware:
Board:							      DOIT ESP32 DEVKIT V1
Core Installation/update date:			3/dec/2018
IDE name:							Arduino IDE
Flash Frequency:					       80Mhz
PSRAM enabled:                                             no
Upload Speed:					       921600
Computer OS:                                                 Windows 10

### Description:
Hello!
I am trying to develop an application that will program a BLE server through OTA using ESP32 serving as BLE client. The main role of the application is to write the .hex file of the program into Server's flash memory. This is done by writing into the characteristics of the server. After writing a block of code into the server the server will then notify the client with the next address in which next block will be written and so on until whole file is sent.
The issue is for 5 to 6 seconds every thing seems perfect server responds to the write event and notify the client with the next address in about   54 to 64 milliseconds. But after this somehow this process slows down and now i am receiving notifications after two to three seconds. I do not want this latency in my application and don't know why this appears after 5-6 seconds. Please help if you can understand and solve this problem. Thanks in advance.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "BLEDevice.h"
#include "FS.h"
#include "SD.h"
#include "SPI.h"
//#include "BLEScan.h"

void configureForNotify(void);
byte readByte();
byte convertCharToHex(char ch);
void getMetaData(byte* data);
void sendMetaData(uint8_t* data);
bool fetchNextData(uint8_t* data);
void sendFirmware(void);
void vSendFirmware(void* pvParameters);

#define LED       2
#define BUTTON    4

#define SERVICE_UUID               "f000ffc0-0451-4000-b000-000000000000"
#define CHARACTERISTIC_UUID_1      "f000ffc1-0451-4000-b000-000000000000"
#define CHARACTERISTIC_UUID_2      "f000ffc2-0451-4000-b000-000000000000"
#define CHARACTERISTIC_UUID_3      "f000ffc3-0451-4000-b000-000000000000"
#define CHARACTERISTIC_UUID_4      "f000ffc4-0451-4000-b000-000000000000"

enum Characteristics {IDENTIFY, BLOCK, COUNT, STATUSC};

static BLEAddress *pServerAddress;
static BLERemoteCharacteristic* pRemoteCharacteristic[4];
static BLERemoteDescriptor*     pRemoteDescriptor[4];

static boolean doConnect = false;
static boolean connected = false;

bool ledStatus = LOW;
bool buttonStatus = false;
bool lastStatus;
String led_on = "LED_on";
String led_off = "LED_off";
uint8_t nextData[18];

File myFile;

SemaphoreHandle_t xSemaphore;
TaskHandle_t firmwareTaskHandle;
bool sema = false;
long timen;
/****************************************************
   Notify Callbacks
*****************************************************/
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
[... 434 lines cut ...]
[D][BLERemoteCharacteristic.cpp:579] writeValue(): << writeValue
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1485] dumpGattClientEvent(): [conn_id: 0, remote_bda: 0a:d0:ad:0a:d0:ad, handle: 15 0x0f, value_len: 2, is_notify: 1]
[D][BLERemoteCharacteristic.cpp:174] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: f000ffc2-0451-4000-b000-000000000000, handle: 15 0xf, props: broadcast: 0, read: 0, write_nr: 1, write: 1, notify: 1, indicate: 0, auth: 0
Time taken: 1958
[D][BLERemoteCharacteristic.cpp:551] writeValue(): >> writeValue(), length: 18
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1623] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 15 0x0f, offset: 0]
[D][BLERemoteCharacteristic.cpp:579] writeValue(): << writeValue
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1485] dumpGattClientEvent(): [conn_id: 0, remote_bda: 0a:d0:ad:0a:d0:ad, handle: 15 0x0f, value_len: 2, is_notify: 1]
[D][BLERemoteCharacteristic.cpp:174] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: f000ffc2-0451-4000-b000-000000000000, handle: 15 0xf, props: broadcast: 0, read: 0, write_nr: 1, write: 1, notify: 1, indicate: 0, auth: 0
Time taken: 2960
```
```
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 16, bd_addr: 0a:d0:ad:0a:d0:ad, min_int: 80, max_int: 800, latency: 0, conn_int: 12, timeout: 600]
[D][BLEClient.cpp:402] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 0, bd_addr: 0a:d0:ad:0a:d0:ad, min_int: 0, max_int: 0, latency: 0, conn_int: 800, timeout: 1000]
[D][BLEClient.cpp:402] handleGAPEvent(): BLEClient ... handling GAP event!
```

chegewara, 2018-12-03:
Could you turn on timestamp in arduino console to see what is blocking for so long?
I am suspecting that `xSemaphorTake` is waiting to be released by notification. Also you have very big mx connection interval (1000 ms) `min_int: 80, max_int: 800, latency: 0, conn_int: 12, timeout: 600`. I would suggest to decrease it and see result.

Umar-Yaqoob, 2018-12-04:
@chegewara  Thanks for reply.

1. Semaphore is there for synchronization. I have to send next chunk of data when i have received the next address where the data would be written. And i think that semaphore is not causing any problem. After 5 to 6 seconds a GAP event occurs  as you can see in the console data, after this the notifications slow down dramatically and i don't know why it happens after exact 5-6 seconds. I suspect that i have to change some default settings in the BLE library, but after spending a lot of time i can't find anything there too.  

2. Please guide me how to do this because i am new to esp32 programming and BLE 

> Also you have very big mx connection interval (1000 ms) min_int: 80, max_int: 800, latency: 0, conn_int: 12, timeout: 600. I would suggest to decrease it and see result.

Umar-Yaqoob, 2018-12-04:
I am  not able to find any function in BLE library that will change the connection interval.
Moreover, it seems that connection interval automatically changes after 5-6 seconds (Increase to 2 to 3 seconds).

Umar-Yaqoob, 2018-12-04:
> I am not able to find any function in BLE library that will change the connection interval.
> Moreover, it seems that connection interval automatically changes after 5-6 seconds (Increase to 2 to 3 seconds).

In ``` esp_gap_ble_api.h``` the function ```esp_ble_gap_set_prefer_conn_params``` is defined as follows

`
esp_err_t esp_ble_gap_set_prefer_conn_params(esp_bd_addr_t bd_addr,
                                                                 uint16_t min_conn_int, uint16_t max_conn_int,
                                                                 uint16_t slave_latency, uint16_t supervision_tout);
`

But don't know where it is defined and how it is called

chegewara, 2018-12-04:
Best way to call it is from onConnect() callback. Similar function is already implemented but for BLEServer. I will add this one to BLEClient too, but it may take few days.

https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLEServer.cpp#L415

Umar-Yaqoob, 2018-12-04:
Now I have cloned the arduino_ESP32 library from GIT GUI. Previously i downloaded from Arduino board manager. Now the same application crashes right after connecting to our service. I have used BLE_Client example as template (updated by @chegewara).
And I have to mention that this example also crashes after connecting to the service. The console message is given below.

regards,  
```
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: OAD Target Image A, Address: 0a:d0:ad:0a:d0:ad, serviceUUID: 0000ffc0-0000-1000-8000-00805f9b34fb, txPower: 0
Forming a connection to 0a:d0:ad:0a:d0:ad
 - Created client
 - Connected to server
 - Found our service

Stack smashing protect failure!

abort() was called at PC 0x4014d54b on core 1

Backtrace: 0x4008e1c4:0x3ffca3c0 0x4008e3f5:0x3ffca3e0 0x4014d54b:0x3ffca400 0x400d509a:0x3ffca420 0x400d50c2:0x3ffca4e0 0x400d1889:0x3ffca540 0x400d199e:0x3ffca590 0x4018dabc:0x3ffca5e0 0x40092ded:0x3ffca600

Rebooting...
```

chegewara, 2018-12-04:
Could you use arduino tool to decode backtrace? All is described in readme.md arduino-esp32 repository. 
Thanks.

Umar-Yaqoob, 2018-12-04:
[image]
This what i got using BackTrace tool.

chegewara, 2018-12-04:
I dont know what to say, i tested it on esp-idf without stack smashing. When i turned on stack smashing i got issue only with wif + ble app (esp-idf issue reported).

Maybe try to change debug log level to info or warn, please.

closed 2019-08-15 by stale[bot]

oless, 2021-02-07:
Split the bluetooth sending part of the system to another thread that is configured to 10% of system resources. This way you will get a smaller latency numbers. Once the message is ready to be sent out this thread will sent it.

Never done this on ESPs but done that on arduinos. 

Hope this helps.