espressif/arduino-esp32 #2146: Username to authenticate

maximiliankapra, 2018-12-03:
### Hardware:
Board:							?ESP32 Dev Module? 
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            Mac OSX

### Description:
I want to know if it is possible to authenticate with a username and a password. 
It is not  really a Captive Portal. When I try connect to the wifi with my phone there is a field username in addition to the password field. 
Is it possible to send also a username at the connection? I did not found anything of that in the WiFi.h and .cpp file.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```// include library, include base class, make path known
#include <GxEPD.h>

//Display Class
#include <GxGDEW042T2/GxGDEW042T2.h>      // 4.2" b/w

#include GxEPD_BitmapExamples

// Font for Display
#include <Fonts/FreeMonoBold9pt7b.h>

#include <GxIO/GxIO_SPI/GxIO_SPI.h>
#include <GxIO/GxIO.h>

//Pins on board
GxIO_Class io(SPI, /*CS=5*/ SS, /*DC=*/ 17, /*RST=*/ 16); // arbitrary selection of 17, 16
GxEPD_Class display(io, /*RST=*/ 16, /*BUSY=*/ 4); // arbitrary selection of (16), 4

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>

#define USE_SERIAL Serial

WiFiMulti wifiMulti;

void setup() {
    USE_SERIAL.begin(115200);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    display.init(115200); //enable diagnostic output on Serial

    display.eraseDisplay(true);

    wifiMulti.addAP(„SSID“, „pwd“);
}

void show(const GFXfont* f, String payload)
{
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(f);
  display.setCursor(0, 0);
  display.println();
  display.println(payload);
  display.update();
}
    
void loop() {
    // wait for WiFi connection
    if((wifiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTPS] begin...\n");
        // configure traged server and url
        http.begin("https://Domain.domain"); //HTTPS

        USE_SERIAL.print("[HTTPS] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();
        
        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTPS] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                show(&FreeMonoBold9pt7b, payload);
                USE_SERIAL.println(payload);
            }
        } else {
          Serial.println(http.errorToString(httpCode));
            USE_SERIAL.printf("[HTTPS] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-12-03:
https://en.wikipedia.org/wiki/Wi-Fi_Protected_Access#WPA_terminology

mkapra, 2018-12-03:
?

atanisoft, 2018-12-03:
If you are meaning on the http connection to the server you need to call this method: https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.h#L154

mkapra, 2018-12-03:
No I wrote „when connecting with the wifi“. So the Router is not only asking for a password, he‘s asking also for a username.

mkapra, 2018-12-04:
So is it possible to use WPA Enterprise?

closed 2019-06-22 by mkapra