espressif/arduino-esp32 #2150: ESP32 BLE Exception on Init [Arduino]

wezleysherman, 2018-12-04:
Hello,

I'm currently trying to run the sample BLE arduino sketch and every time I run it, my ESP32 seems to crash on BLE init.

I have it connected to a 1amp 3.3v regulator and am using the sample code for BLE provided by the Arduino library.

When I look at the stack trace,  it seems to be something with power control? But that doesn't make sense given that my power supply is pretty high current.

`C: 0x40199480: ram_tx_pwr_backoff at phy_chip_v7_cal.c line 1382
EXCVADDR: 0x00000000

Decoding stack results
0x40199480: ram_tx_pwr_backoff at phy_chip_v7_cal.c line 1382
0x401996f2: tx_pwctrl_init_cal at phy_chip_v7_cal.c line 2090
0x40199823: tx_pwctrl_init at phy_chip_v7_cal.c line 2158
0x40197620: bb_init at phy_chip_v7.c line 2418
0x40197c46: register_chipv7_phy at phy_chip_v7.c line 3087
0x4014a54d: esp_phy_rf_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 124
0x4014a7fc: esp_phy_load_cal_and_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/phy_init.c line 563
0x400da1b3: esp_bt_controller_enable at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bt.c line 926
0x400d70c2: btStart at /home/wezley/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/esp32-hal-bt.c line 44
0x400d4075: BLEDevice::init(std::__cxx11::basic_string  , std::allocator  >) at /home/wezley/.arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/BLE/src/BLEDevice.cpp line 337
0x400d1445: setup() at /home/wezley/Desktop/BLE_uart/BLE_uart.ino line 128
0x401893ea: loopTask(void*) at /home/wezley/.arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp line 15
`

I am running the latest version of esp-idf (literally installed it a week ago).

What am I missing?

Thank you!

chegewara, 2018-12-04:
Could you try with different USB cable?

wezleysherman, 2018-12-04:
Yeah I’ve tried with a couple of different power supplies and am getting the same result.

wezleysherman, 2018-12-04:
Quick update (sorry for the double post): It seems to be my 3.3V regulators? When I connect it straight to my battery it works perfectly. When I run my battery through the regulator it stops working and throws this exception. Why? My regulator is giving a lot of current.

stickbreaker, 2018-12-04:
@wezleysherman you will have to measure the voltage at the ESP32 module.  Somewhere you are loosing power.  The WiFi init has been measured to use 500mA, I don't have an numbers on combined BT&WiFi.

What voltage of battery are you directly connecting?  The ESP32 module is spec'd at a max of 3.6V.  If you have a fully charged LiPo you are pushing 4.2V?  It could be poor pcb design, (too small of traces, Bad solder joints).  You will have to measure the actual voltage at the input to the module.

If you suspect pcb traces, just run some 18~20ga wires directly from the regulator to the module, and battery to regulator.

Chuck.

stickbreaker, 2018-12-05:
BLE and Wifi use the same electronics or transmit and receive, the data interpretation is what is different.
Can you post a schematic of your power circuit? what is the part number for your regulator.  If it is a linear regulator LM1117 (800mA) or similar  your battery voltage is too low.  The LM1117-3.3 is specified with:
* Vin (voltage in) 4.75 < Vin < 10V
* Iout( current Out) 0<= Iout <= 800mA

It needs at least 4.75V on it's input before it can produce 3.3V out.

Chuck.

wezleysherman, 2018-12-05:
Thank you @stickbreaker! 

The regulator I'm using is the LD1117 -- so dropout is an issue with my battery.

I've figured it out. The power sources I was using through my 5V rail weren't able to provide the necessary 500Ma (unless ran in parallel). However; connecting just my 3.7V battery (charged at 3.8V) straight to the ESP module is doing the trick.

My only concern is how will the module react when my battery is fully charged and above 4V? Should it still run fine? Will I need to find another way to drop the voltage a bit?

stickbreaker, 2018-12-05:
The ESP32 is rated a 3.6V MAX anything over that is beyond its rated input, It is designed to operate a 3.3V. It will DIE maybe not immediately but SOON. Here is a list of LDO regulators that would work [TexasInstruments](http://www.ti.com/power-management/linear-regulators-ldo/products.html?utm_source=google&utm_medium=cpc&utm_term=%252Bldo&utm_campaign=app_ldo_us_n_m_ldo&utm_content=_k_EAIaIQobChMImZ_-jpmJ3wIV05N-Ch1TBg3yEAAYASABEgKa0_D_BwE_k_&gclid=EAIaIQobChMImZ_-jpmJ3wIV05N-Ch1TBg3yEAAYASABEgKa0_D_BwE#p238min=0;3.6&p238max=3.6;12&p451max=0.8;1.2000000000000002&p182=Fixed%20Output&p1154=3.3;3.3)

Chuck.

wezleysherman, 2018-12-05:
Looks like I'll move to using the TLV1117V. 

Thank you so much for all your help @stickbreaker !

closed 2018-12-05 by wezleysherman