espressif/arduino-esp32 #2153: D32 Pro defines TF_CS, SD card library uses SS

mtfurlan, 2018-12-04:
If you run the SD_Test example: https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino
on a lolin d32 pro https://github.com/espressif/arduino-esp32/blob/master/variants/d32_pro/pins_arduino.h
using the SD card slot, it will fail with `Card Mount Failed`, because the SD library uses `SS` as the default slave/chip select, while the d32 has `TF_CS` defined for the slave select.

In the example if you set the CS, it runs as expected.
```diff
-    if(!SD.begin()){
+    if(!SD.begin(4)){
```

I think the actual solution is to also define `SS` for the D32 pro?

me-no-dev, 2018-12-06:
totally correct!

me-no-dev, 2018-12-06:
fixed :)

closed 2018-12-06 by me-no-dev