espressif/arduino-esp32 #2154: WiFiClientSecure make issue

Goldenstarc, 2018-12-05:
### Hardware:
Board:							node32
Core Installation/update date:			latest
IDE name:							IDF-with Arduino
Flash Frequency:					40
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                          MacOS

### Description:
error in make command !
USING LATEST LIBRARIES

`CXX build/arduino-esp32/libraries/WiFiClientSecure/src//ssl_client.o
/Users/hamid/Documents/Projects/ESP-IDF/IDF-Arduino/components/arduino-esp32/libraries/WiFiClientSecure/src/ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, const char*, const char*, const char*, const char*, const char*)':
/Users/hamid/Documents/Projects/ESP-IDF/IDF-Arduino/components/arduino-esp32/libraries/WiFiClientSecure/src/ssl_client.cpp:144:67: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
                  (const unsigned char *)pskIdent, strlen(pskIdent));
                                                                   ^
make[1]: *** [libraries/WiFiClientSecure/src//ssl_client.o] Error 1
make: *** [component-arduino-esp32-build] Error 2`

atanisoft, 2018-12-05:
you need to enable MBED_TLS_PSK options in your sdkconfig, please refer to the arduino sdkconfg for an example, there are four or five lines you need to enable.

Goldenstarc, 2018-12-06:
ok dude, I'm surrender ! :)
I just get a new version (TODAY) and it fixed
I have a simple function to update my esp repos in bash profile. it may help  :)

`function esp_update()
{
currentDir=$PWD && \
cd ~/esp && \
cd arduino-esp32 && \
git pull origin master && \
git submodule update --init --recursive && \
cd .. && \
cd esp-idf && \
git pull origin master && \
pip install esptool --upgrade && \
cd $currentDir
}`

henryjw, 2019-01-13:
For anyone else that runs into this issue, another solution is to disable the WifiClientSecure component altogether. This can be done by following the steps below:

1) Run `make menuconfig` from the project's directory
2) Select "Arduino Configuration"
3) Select "Include only specific Arduino libraries"
4) Deselect Enable WifiClientSecure" and any other libraries you don't need

Goldenstarc, 2019-01-14:
Thanks henry
But the problem was that the WifiClientSecure was FORCE selected (i think it was gray, not white)

henryjw, 2019-01-14:
@Goldenstarc Whoops. Forgot to mention that there are other libraries that are dependent on WifiClientSecure, so those have to be disabled first. So the order to disable them in is HttpClient, AzureIOT, WifiClientSecure. I'll update my original comment to reflect this.

sfowlr, 2019-01-14:
Should this issue be re-opened? Seems to me like the Kconfig file for Arduino should prevent you from using WiFiClientSecure if you don't have its dependencies enabled. I'll see if I can put together a PR to fix this.

qt1, 2019-01-16:
Same issue after last update ..

tevino, 2019-01-16:
Same issue on latest master

nevercast, 2019-01-18:
This seemed to work for me
`make menuconfig`
```
Component Config -> mbedTLS -> TLS Key Exchange Methods -> 
  [*] Enable pre-shared-key ciphersuits
    [*] Enable PSK based ciphersuite modes
```

Good luck.

tevino, 2019-01-18:
@nevercast it works, thank you.

Goldenstarc, 2019-01-19:
> Should this issue be re-opened? Seems to me like the Kconfig file for Arduino should prevent you from using WiFiClientSecure if you don't have its dependencies enabled. I'll see if I can put together a PR to fix this.
 
you mean this part of Kconfig.projbuild?

`config ARDUINO_SELECTIVE_WiFiClientSecure
    bool "Enable WiFiClientSecure"
    depends on ARDUINO_SELECTIVE_COMPILATION
    select ARDUINO_SELECTIVE_WiFi
    default y`

sfowlr, 2019-01-21:
Yes, a proper kconfig should really have all of an option's dependencies
listed, either with require or select. So that option really should also
select MBEDTLS_PSK_MODES as well as the configuration symbols required for
the PSK modes that have been implemented. Otherwise this codebase is just
going to keep getting broken every time people try to update without
starting from scratch or totally replacing their sdkconfig.

Goldenstarc, 2019-01-25:
And as a result, Arduino development team must work on this ?!

tmeysson, 2019-01-29:
Ran into this just now, with a fresh install. Adding the required options in menuconfig as @nevercast stated fixed it.

nevercast, 2019-01-30:
@Goldenstarc Can you please open this issue so that the Arduino team may review it again.

closed 2019-08-15 by stale[bot]