espressif/arduino-esp32 #2158: ESP32-WROOM battery measuring / monitoring

Qaldak, 2018-12-05:
### Hardware:
Board:							doit ESP32 devkit v1 (36 Pins) (ESP32-WROOM)
Core Installation/update date:			?11/jul/2017?
IDE name:						Arduino IDE
Flash Frequency:					80MHz
Upload Speed:						921600
Computer OS:                                         Windows 10

### Description:
Hi folks

I use doit esp32-wroom (doit esp32 devkitv1) as temperature sensor in my tortoise enclosure.
I use this microcontroller with an 3.7v 18650 battery (4200 mAh). And it works. I connected the battery to the VIN pin.

Now I would like to measure the battery power level and send a MQTT message to log this. Because I would like to know when the battery is empty and send a message to change the battery. Another use case is how the battery will be reloaded (by solar panel)… but this comes later.

My problem is: I can't find how I can measure the power correctly.

I found a method with "extern "C" int rom_phy_get_vdd33();", but I don’t know what kind of data these are:
>     5471.00
>     5474.00
>     5477.00
>     5477.00
>     5480.00
>     5473.00
>     5474.00

Do I have to calculate something? Is there a better/standard way to calculate battery level? 
These values seems to have no relevance to my 18650 (3.7v, 4200mAh)...

I don't have used any resistors from 18650 to VIN pin. I found a lot posts about use resistors for analog pins, but nothing about VIN pin.

That's my first "battery"-project, and I didn't find anything to help me.

### Sketch:
```
extern "C" int rom_phy_get_vdd33();
...
...
void setup() {
...
  /* battery power measuring */
  int internalBatReading;
  if (WiFi.status() == 255) {
    btStart();
    internalBatReading = rom_phy_get_vdd33();
    Serial.println(internalBatReading);
    btStop();
  }
...
}
```
[image]

lbernstone, 2018-12-06:
Weird- an undocumented function doesn't do what you think it should? :smile:
Run a voltage divider from the battery to an ADC pin, set to be 1.1V at your max voltage and read it with standard ADC functions.

Qaldak, 2018-12-06:
> Weird- an undocumented function doesn't do what you think it should? 😄
You're absolutly right. ;)

Qaldak, 2018-12-06:
> Weird- an undocumented function doesn't do what you think it should? 😄

You're absolutely right.;-)

closed 2018-12-19 by Qaldak

lbernstone, 2019-07-01:
Calculate the needed resistors with a [voltage divider calculator](http://www.ohmslawcalculator.com/voltage-divider-calculator).
There's a code example for calibrated ADC readings at https://github.com/espressif/arduino-esp32/issues/1804.

mixaobo, 2019-07-02:
> Calculate the needed resistors with a [voltage divider calculator](http://www.ohmslawcalculator.com/voltage-divider-calculator). Try to get your typical values in the middle of the range.
> There's a code example for calibrated ADC readings at #1804.

For 18650  which i just buy has 3.6V, so when max charge it's is 4.2.
for 4.2v i need 27Kohm and 100Kohm -->then it would be 3.3v at the measure point
but if i got tthe new one from market it will be 3.6v then i need 10 and 82 ohm--> 3.3v at the measure point.
so if i design a board with 27k and 100kohm then put a new battery the voltage about 2.9v
so which resistors should i use, i dont get the point

mixaobo, 2019-07-02:
> Weird- an undocumented function doesn't do what you think it should? 😄
> Run a voltage divider from the battery to an ADC pin, set to be 1.1V at your max voltage and read it with standard ADC functions.

sorry, do you mean you divider to set 1.1v at measure point or to set 3.3 at measurue point and use  adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_0db) to set at 1.1v