espressif/arduino-esp32 #2159: abort() was called at PC 0x4008355e on core 0 - sketch worked perfectly on older release

mitchsf, 2018-12-05:
Here is the error:

abort() was called at PC 0x4008355e on core 0

Backtrace: 0x4009027c:0x3ffe3bd0 0x4009047f:0x3ffe3bf0 0x4008355e:0x3ffe3c10 

0x400836f0:0x3ffe3c40 0x4007892f:0x3ffe3c60 0x40078995:0x3ffe3c90 0x400789a0:0x3ffe3cc0 0x40078b33:0x3ffe3ce0 0x40080335:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

Exception decoder results:

0x4009027c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x4009047f: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x4008355e: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 381

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		installed today, 12/4/18
IDE name:					        Arduino IDE 1.8.7
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:				        115200
Computer OS:                                          Windows 10

This is a very large sketch that was working using a previous release, about six months ago. I'm not sure where to start looking. Thanks in advance for any help on this.

atanisoft, 2018-12-06:
you can start by decoding the stack above using this tool: https://github.com/me-no-dev/EspExceptionDecoder

me-no-dev, 2018-12-06:
Maybe the sketch does not fit? It is possible that updates increase the size of the binary and as you said, it's a pretty large sketch :) Try changing the partition scheme

Dserog, 2019-01-21:
@mitchsf did you ever solve this issue? I am having the same problem. Sketches not containing BLE work fine but even if I run the BLE examples I get this error.

luanmm, 2019-05-20:
I had the exact same error when trying to setup OTA without setting up wi-fi connection first. It seems the Arduino SDK is not so good at checking for states (or, at least, giving some clues of what is going on).

atanisoft, 2019-05-21:
This kinda looks like attempting to declare an array that is too large to fit into memory and this is causing cpu abort prior to running.

Do you have any large buffers allocated outside of methods or possibly in setup() or loop()?

luanmm, 2019-05-21:
@atanisoft in my case, just changing the wi-fi startup to before the OTA startup solved the problem, without changing any considerable logic or different heap allocation. So I end up thinking that maybe there is more cases when this error may occur.

But I can't give a lot of technical details because the project I'm working on has grown up a lot (and maybe it shouldn't be using Arduino framework at all). It seems ESP-IDF would fit better in large projects.

atanisoft, 2019-05-21:
> But I can't give a lot of technical details because the project I'm working on has grown up a lot (and maybe it shouldn't be using Arduino framework at all). It seems ESP-IDF would fit better in large projects.

This would more come down to preference. Everything you can do in ESP-IDF can be done in the Arduino side with the same APIs.

> just changing the wi-fi startup to before the OTA startup solved the problem

These two should have zero dependency on each other. It would likely come down to which API is being used though.

In the case of @mitchsf original failure that looks like an over allocation of memory in the global scope. This can be reproduced by using a static array allocation over about 40-50kb (maybe higher) as there may not be a large enough memory block by the time the startup code gets to that object and thus it aborts.

luanmm, 2019-05-21:
> These two should have zero dependency on each other. It would likely come down to which API is being used though.

Interesting. I thought it was the case here.

If I have some time to investigate here, I post updates in this thread. Currently, I'm not with time for do some tests, so all my comments are nothing but the best guesses I have in the moment (without any internal framework knowledgements).

lbernstone, 2019-05-21:
I would expect there is a dependency of OTA on WiFi (or netif really).  On begin(), it attempts to open a listening socket.  If there is no interface, it can't open a socket.  It would be nice if it didn't panic the kernel, but it is something that the programmer should be thinking about.

closed 2019-08-14 by stale[bot]