espressif/arduino-esp32 #2162: BLE Client and WiFi Client simultaneously

nb-ceffa, 2018-12-06:
### Description:
Hi, I'm trying to use BLE Client and WiFi Client simultaneously but reading value from BLE device when WiFi is connected seems to be impossible. :( I have created simple demo attached below. Reading works fine when WiFi part is removed. Any idea what is wrong? 
Thank you in advance! 

### Sketch:
```cpp

#include <Arduino.h>
#include <WiFi.h>
#include "BLEDevice.h"

BLEClient* bleClient;
BLEAddress bleAddress = BLEAddress("<address>");

WiFiClient wifiClient;

void doBleStuff() {
  bleClient->connect(bleAddress);
  Serial.println("Reading...");
  std::string value = bleClient->getValue(BLEUUID("00001800-0000-1000-8000-00805f9b34fb"), BLEUUID("00002a00-0000-1000-8000-00805f9b34fb"));
  Serial.print("Value: ");
  Serial.println(value.c_str());
  bleClient->disconnect();
}

void setup() {
  Serial.begin(115200);
  BLEDevice::init("");
  bleClient = BLEDevice::createClient();
  
  wifiClient = WiFiClient();
  WiFi.begin("<SSID>", "<PASSWD>");

  delay(5000); 

  if (WiFi.waitForConnectResult() == WL_CONNECTED) { 
      Serial.print("WiFi connected. IP address: ");
      Serial.println(WiFi.localIP());
  }
}

void loop() {
  delay(10000); 

  doBleStuff();
}
```

### Debug Messages:
```
[D][BLEDevice.cpp:54] createClient(): >> createClient
[D][BLEDevice.cpp:60] createClient(): << createClient
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.123.69, MASK: 255.255.255.0, GW: 192.168.123.1
WiFi connected. IP address: 192.168.123.69
[D][BLEClient.cpp:91] connect(): >> connect(<MAC_ADDRESS>)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_CONN_CANCEL, conn_id: 0, remote_bda: <MAC_ADDRESS>]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: <MAC_ADDRESS>, mtu: 0]
[D][BLEClient.cpp:123] connect(): << connect(), rc=0
Reading...
[D][BLEClient.cpp:386] getValue(): >> getValue: serviceUUID: 00001800-0000-1000-8000-00805f9b34fb, characteristicUUID: 00002a00-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001800-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:358] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
```

### Build log:
```
Processing esp32dev__ (platform: espressif32; board: esp32dev; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <ESP32 BLE Arduino> 0.4.16
Compiling .pioenvs\esp32dev__\src\main.cpp.o
Linking .pioenvs\esp32dev__\firmware.elf
Retrieving maximum program size .pioenvs\esp32dev__\firmware.elf
Checking size .pioenvs\esp32dev__\firmware.elf
Building .pioenvs\esp32dev__\firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  24.0% (used 78688 bytes from 327680 bytes)
PROGRAM: [=======   ]  74.4% (used 1462654 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
```

chegewara, 2018-12-07:
Did you try to have 2 separate tasks?

nb-ceffa, 2018-12-07:
Version with tasks:
### Code:
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include "BLEDevice.h"

BLEClient* bleClient;
BLEAddress bleAddress = BLEAddress("<address>");

WiFiClient wifiClient;

void doBleStuff(void * parameter) {
  delay(6000);
  BLEDevice::init("");
  bleClient = BLEDevice::createClient();

  while(true) {
      Serial.println("Connecting (BLE)...");
      bleClient->connect(bleAddress);
      Serial.println("Reading...");
      std::string value = bleClient->getValue(BLEUUID("00001800-0000-1000-8000-00805f9b34fb"), BLEUUID("00002a00-0000-1000-8000-00805f9b34fb"));
      Serial.print("Value: ");
      Serial.println(value.c_str());
      bleClient->disconnect();

    Serial.println("Waiting (BLE)...");
    delay(10000);
  }

  delete bleClient;
  vTaskDelete(NULL);
}

void doWifiStuff(void * parameter) {
  wifiClient = WiFiClient();
  while(true) {  
    Serial.println("Connecting (WiFI)...");
    WiFi.begin("<SSID>", "<PASSWD>");

    delay(5000); 

    if (WiFi.waitForConnectResult() == WL_CONNECTED) { 
        Serial.print("WiFi connected. IP address: ");
        Serial.println(WiFi.localIP());
    }

    Serial.println("Waiting (WiFi)...");
    delay(500000);
    Serial.println("Disconnecting (WiFI)...");
    WiFi.disconnect();
  }

  vTaskDelete(NULL);
}

void setup() {
  Serial.begin(115200);
  xTaskCreate(doWifiStuff, "WifiTask", 10000, NULL, 1, NULL);
  xTaskCreate(doBleStuff, "BleTask", 10000, NULL, 1, NULL);
}

void loop() {
  delay(10000); 
}
```

### Log:
```
Connecting (WiFI)...
[... 46 lines cut ...]
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_TERMINATE_LOCAL_HOST, conn_id: 0, remote_bda: <MAC_ADDRESS>]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_UNREG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_UNREG_EVT
Connecting (BLE)...
[D][BLEClient.cpp:91] connect(): >> connect(<MAC_ADDRESS>)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_DISCONNECT_EVT
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_FAIL_ESTABLISH, conn_id: 0, remote_bda: <MAC_ADDRESS>]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: 7d:d8:be:22:96:21, mtu: 0]
[D][BLEClient.cpp:123] connect(): << connect(), rc=0
Reading...
[D][BLEClient.cpp:386] getValue(): >> getValue: serviceUUID: 00001800-0000-1000-8000-00805f9b34fb, characteristicUUID: 00002a00-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001800-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:358] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
```

chegewara, 2018-12-08:
One more thing, if its possible try first obtain IP address and then start with ble. This is how i am doing my stuff.

nb-ceffa, 2018-12-08:
That's what i did in the the example with 6sec delay. Isn't it?

```
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.123.69, MASK: 255.255.255.0, GW: 192.168.123.1
WiFi connected. IP address: 192.168.123.69
Waiting (WiFi)...
[D][BLEDevice.cpp:54] createClient(): >> createClient
[D][BLEDevice.cpp:60] createClient(): << createClient
Connecting (BLE)...
```

but it is still stopped on:
```
Reading...
[D][BLEClient.cpp:386] getValue(): >> getValue: serviceUUID: 00001800-0000-1000-8000-00805f9b34fb, characteristicUUID: 00002a00-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:322] getService(): >> getService: uuid: 00001800-0000-1000-8000-00805f9b34fb
[D][BLEClient.cpp:358] getServices(): >> getServices
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
```

chegewara, 2018-12-08:
@nb-ceffa Please try this:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L108

Sorry for bad support from my side, i am very occupied lately.

nb-ceffa, 2018-12-09:
@chegewara I'm glad for any support from your side ;)

the behavior is still the same :( and it sometimes throws an exception:
```
0x40094ec8: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x400950cb: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x401a9a67: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401a9aae: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x401973a7: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
```

chegewara, 2018-12-09:
Could you try with different service UUID, it should work for all services, but i think its something wrong in here.

nb-ceffa, 2018-12-09:
still the same :( 

Should I use https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp instead of https://github.com/nkolban/ESP32_BLE_Arduino/blob/11a0aa645326aed0cb681e86060fe28d4623bc68/src/BLEClient.cpp ?

chegewara, 2018-12-09:
This should be version with some improvements, but is not prepared PR yet:
https://gist.github.com/chegewara/d3ed0e3365407f6d3abcffc486b2c462

nb-ceffa, 2019-02-25:
Hi @chegewara sorry for a long pause. I have updated to the newest lib (platformio "Espressif 32" 1.7.0): 
Dependency Graph
|-- <WiFi> 1.0
|-- <ESP32 BLE Arduino> 1.0.1

and current stacktrace is:
```
0x40091c18: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40091e49: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4014d50f: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400973fd: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 214
0x40084582: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 269
0x4008b0f1: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400e5371: btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x400e1456: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x400940a9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

last few lines of log:
```
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 24, UUID: 00002a28-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 24 0x24, uuid: 00002a28-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a28-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:215] retrieveCharacteristics(): << getCharacteristics()
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: 00002a24-0000-1000-8000-00805f9b34fb, handle: 20 0x14
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3fff2a30), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3fff2a30), owner: readValue
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3fff2a30), owner: readValue for readValue
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
CORRUPT HEAP: Bad head at 0x3fff4e3c. Expected 0xabba1234 got 0x3fff5b90
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4014d50f on core 0
```

nb-ceffa, 2019-02-25:
Second crash looks like this:

Stacktrace: 
```
0x40091c18: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40091e49: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4014d50f: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4008457b: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008b0f1: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400e5371: btc_gattc_cb_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x400e1456: btc_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/btc/core/btc_task.c line 110
0x400940a9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

last few lines of log:
```
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 24, UUID: 00002a28-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 24 0x24, uuid: 00002a28-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a28-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:215] retrieveCharacteristics(): << getCharacteristics()
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: 00002a24-0000-1000-8000-00805f9b34fb, handle: 20 0x14
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3fff3970), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3fff3970), owner: readValue
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3fff3970), owner: readValue for readValue
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4014d50f on core 0
```

fortifyIT, 2019-04-03:
any update on dis issue? facing the same...

elettro1, 2019-05-21:
this error seems to be generated by BLERemoteCharacteristic::readValue() 

try this my fix:
https://github.com/espressif/arduino-esp32/pull/2800

kenken64, 2019-06-22:
after applying #2800 I am facing new issue 

`
14:16:35.444 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
14:16:35.444 -> Debug exception reason: Stack canary watchpoint triggered (btuT) 
14:16:35.479 -> Core 0 register dump:
14:16:35.479 -> PC      : 0x4000c472  PS      : 0x00060536  A0      : 0x8011bede  A1      : 0x3ffe11b0  
14:16:35.479 -> A2      : 0x3ffe11c4  A3      : 0x00000000  A4      : 0x00000268  A5      : 0x3ffe13f4  
14:16:35.479 -> A6      : 0x3ffee8bd  A7      : 0x00000026  A8      : 0x8013b2fa  A9      : 0x3ffe1c40  
14:16:35.479 -> A10     : 0x3ffed930  A11     : 0x3ffef184  A12     : 0x3ffe1eda  A13     : 0x00000000  
14:16:35.514 -> A14     : 0x00000004  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
14:16:35.514 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000002  
14:16:35.514 -> 
14:16:35.514 -> Backtrace: 0x4000c472:0x3ffe11b0 0x4011bedb:0x3ffe11c0 0x4013b26d:0x3ffe1450 0x4011b7aa:0x3ffe16e0 0x4013a0c5:0x3ffe1700 0x4013b2f7:0x3ffe1990 0x4011b7aa:0x3ffe1c20 0x4013a0c5:0x3ffe1c40 0x4013b2f7:0x3ffe1ed0 0x4011b7aa:0x3ffe2160 0x4013a0c5:0x3ffe2180 0x4013ae9a:0x3ffe2410 0x4011b7aa:0x3ffe2430 0x4013b10e:0x3ffe2450 0x4011b7aa:0x3ffe2470 0x4013adcd:0x3ffe2490 0x4013ade9:0x3ffe24b0 0x4011b7aa:0x3ffe24d0 0x4011b8e9:0x3ffe24f0 0x4011e90a:0x3ffe2510 0x400f9466:0x3ffe2530 0x4008e189:0x3ffe2560
14:16:35.548 -> 
14:16:35.548 -> Rebooting...
14:16:35.548 -> ets Jun  8 2016 00:22:57
14:16:35.548 -> 
14:16:35.548 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:16:35.582 -> configsip: 0, SPIWP:0xee
14:16:35.582 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:16:35.582 -> mode:DIO, clock div:1
14:16:35.582 -> load:0x3fff0018,len:4
14:16:35.582 -> load:0x3fff001c,len:1100
14:16:35.582 -> load:0x40078000,len:9232
14:16:35.582 -> load:0x40080400,len:6400
14:16:35.582 -> entry 0x400806a8
`

`
0x4011bedb: bta_gattc_enqueue at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_utils.c line 460
0x4013b26d: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1143
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c5: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1515
0x4013b2f7: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1166
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c5: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1515
0x4013b2f7: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1166
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c5: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1515
0x4013ae9a: bta_gattc_cfg_mtu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 934
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013b10e: bta_gattc_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1047
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013adcd: bta_gattc_reset_discover_st at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 839
0x4013ade9: bta_gattc_disc_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 858
0x4011b7aa: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4011b8e9: bta_gattc_hdl_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 397
0x4011e90a: bta_sys_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 496
0x400f9466: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 233
0x4008e189: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
`

kenken64, 2019-07-25:
Paste your stack trace here12:24:51.387 -> Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
12:24:51.387 -> Debug exception reason: Stack canary watchpoint triggered (btuT) 
12:24:51.387 -> Core 0 register dump:
12:24:51.387 -> PC      : 0x4000c472  PS      : 0x00060336  A0      : 0x8011bef6  A1      : 0x3ffe1410  
12:24:51.422 -> A2      : 0x3ffe1424  A3      : 0x00000000  A4      : 0x00000268  A5      : 0x3ffe1654  
12:24:51.422 -> A6      : 0x3ffe4369  A7      : 0x00000026  A8      : 0x8013b2f6  A9      : 0x3ffe1ea0  
12:24:51.422 -> A10     : 0x3ffedbb8  A11     : 0x3ffde060  A12     : 0x3ffe213a  A13     : 0x00000000  
12:24:51.422 -> A14     : 0x00000004  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000001  
12:24:51.422 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000002  
12:24:51.457 -> 
12:24:51.457 -> Backtrace: 0x4000c472:0x3ffe1410 0x4011bef3:0x3ffe1420 0x4013b269:0x3ffe16b0 0x4011b7c2:0x3ffe1940 0x4013a0c1:0x3ffe1960 0x4013b2f3:0x3ffe1bf0 0x4011b7c2:0x3ffe1e80 0x4013a0c1:0x3ffe1ea0 0x4013b2f3:0x3ffe2130 0x4011b7c2:0x3ffe23c0 0x4013a0c1:0x3ffe23e0 0x4013ae96:0x3ffe2670 0x4011b7c2:0x3ffe2690 0x4013b10a:0x3ffe26b0 0x4011b7c2:0x3ffe26d0 0x4013adc9:0x3ffe26f0 0x4013ade5:0x3ffe2710 0x4011b7c2:0x3ffe2730 0x4011b901:0x3ffe2750 0x4011e922:0x3ffe2770 0x400f94ba:0x3ffe2790 0x4008e461:0x3ffe27c0
12:24:51.492 -> 
12:24:51.492 -> Rebooting...
12:24:51.492 -> ets Jun  8 2016 00:22:57
12:24:51.492 -> 
12:24:51.492 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:24:51.492 -> configsip: 0, SPIWP:0xee
12:24:51.492 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:24:51.492 -> mode:DIO, clock div:1
12:24:51.492 -> load:0x3fff0018,len:4
12:24:51.526 -> load:0x3fff001c,len:1100
12:24:51.526 -> load:0x40078000,len:9232
12:24:51.526 -> load:0x40080400,len:6400
12:24:51.526 -> entry 0x400806a8

PC: 0x4000c472
EXCVADDR: 0x00000000

Decoding stack results
0x4011bef3: bta_gattc_enqueue at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_utils.c line 461
0x4013b269: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1146
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c1: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013b2f3: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c1: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013b2f3: bta_gattc_write at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1170
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013a0c1: bta_gattc_pop_command_to_send at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1519
0x4013ae96: bta_gattc_cfg_mtu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 936
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013b10a: bta_gattc_disc_cmpl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 1049
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4013adc9: bta_gattc_reset_discover_st at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 841
0x4013ade5: bta_gattc_disc_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 860
0x4011b7c2: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x4011b901: bta_gattc_hdl_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 397
0x4011e922: bta_sys_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 496
0x400f94ba: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 233
0x4008e461: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

chegewara, 2019-07-25:
`Debug exception reason: Stack canary watchpoint triggered (btuT)`
This is stack overflowed in esp-idf bluetooth core library. What debug level did you setup?

nb-ceffa, 2019-09-23:
It still does not work :(
Current version:
```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), tool-mkspiffs 2.230.0 (2.30), framework-arduinoespressif32 2.10003.190916 (1.0.3), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
|-- <ESP32 BLE Arduino> 1.0.1
```

stacktrace:
```
PC: 0x400d52fe: BLERemoteService::getCharacteristic(BLEUUID) at C:\Users\Kefurt\.platformio\packages\framework-arduinoespressif32\libraries\BLE\src\BLERemoteService.cpp line 142
EXCVADDR: 0x00000034

Decoding stack results
0x400d52fe: BLERemoteService::getCharacteristic(BLEUUID) at C:\Users\Kefurt\.platformio\packages\framework-arduinoespressif32\libraries\BLE\src\BLERemoteService.cpp line 142
0x400d2ecb: BLEClient::getValue[abi:cxx11](BLEUUID, BLEUUID) at C:\Users\Kefurt\.platformio\packages\framework-arduinoespressif32\libraries\BLE\src\BLEClient.cpp line 443
0x400d18ae: doBleStuff(void*) at src\main.cpp line 19
0x4008d83d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

```

wakwak-koba, 2019-09-24:
Try my repository, which fixes many bugs in BLEClient
https://github.com/wakwak-koba/arduino-esp32

You can replace it completely, or you can copy only the minimum required files.
- libraries/BLE/src/BLEAdvertisedDevice.cpp
- libraries/BLE/src/BLEClient.cpp
- libraries/BLE/src/BLEClient.h
- libraries/BLE/src/BLEDevice.cpp
- libraries/BLE/src/BLEDevice.h
- libraries/BLE/src/BLERemoteCharacteristic.cpp
- libraries/BLE/src/BLERemoteCharacteristic.h
- libraries/BLE/src/BLERemoteService.cpp
- libraries/BLE/src/BLEScan.cpp
- libraries/BLE/src/BLEScan.h

me-no-dev, 2019-09-24:
if you have fixes, why not contribute them here?

nb-ceffa, 2019-09-24:
@wakwak-koba I have copied your version of BLE but with the same result.

BUT! I finally have found one of two issues!!! I will prepare pull request.

The second issue is still present:
```
CORRUPT HEAP: Bad head at 0x3fff6344. Expected 0xabba1234 got 0x3fff6758
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free

0x40091f34: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x40092165: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4014a64b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40091ba9: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 214
0x40084b66: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008aee9: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400dc98d: osi_free_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/allocator.c line 176
0x400dd422: list_free_node at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/list.c line 245
0x400dd4e4: list_clear at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/osi/list.c line 191
0x40119a9f: bta_gattc_clcb_dealloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_utils.c line 312
0x40138a7e: bta_gattc_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_act.c line 807
0x4011967a: bta_gattc_sm_execute at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 288
0x401197b9: bta_gattc_hdl_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/gatt/bta_gattc_main.c line 397
0x4011c7da: bta_sys_event at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 496
0x400f702e: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 233
0x4008de81: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
any idea?

wakwak-koba, 2019-09-24:
I also sometimes disconnected Bluetooth while using Bluetooth and WiFi at the same time.

WiFi and Bluetooth share antennas, and when used at the same time, they seem to interfere and become unstable.
There was such exchange in the past.
https://esp32.com/viewtopic.php?t=6707

mrdc, 2019-10-17:
@nb-ceffa 
Your sketch works fine for me - no crashes.

nb-ceffa, 2019-10-17:
@mrdc hmm that's really interesting! Could you please share the compiled version for MAC: 7D:D8:BE:22:96:21. I would like to try it on my HW.

mrdc, 2019-10-20:
@nb-ceffa 
> Could you please share the compiled version for MAC: 7D:D8:BE:22:96:21

No problem. BTW MAC is not in compiled binary or I'm missing something?

mrdc, 2019-10-21:
> hmm that's really interesting!

Unfortunately, after some tests with WiFI+BLE+HTTPS my ESP32 reboots after this error:
`esp32 freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!`

stickbreaker, 2019-10-21:
@mdr the ESP FreeRTOS implementation of a counting semaphore results in a memory buffer of MAXCOUNT integers.  so if you specify a CountingSempahore of 1000 max, it allocates a QUEUE of 1000 integer members.

Base on the error, you are probably running out of RAM.

Chuck.

mdr, 2019-10-21:
@stickbreaker 640kb ought to be enough for anybody, surely?

stickbreaker, 2019-10-21:
I wonder how many kids get that joke?  I remember trying to figure out DRAM refresh rates for a 16MHz 80386 CPU and 4GB DRAM.  This was when the largest DRAM (16pin dip) was 64kbit.

It was only theoretical, because to get 4GB you would have to have 524,288 chips at $10 each that is a bit much for a desktop PC!

I really question the decision to use a QUEUE to simulate a CountingSemaphore.  It seems to me to be a lazy approach.

Chuck.

mrdc, 2019-10-22:
@stickbreaker 

Is BT classic disabled by default during compilation? I've checked BLEDevice.cpp for `esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT)`

```
#ifndef CLASSIC_BT_ENABLED
		esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);  
#endif 
```

mrdc, 2019-10-22:
Have tried code from @chegewara (https://github.com/espressif/arduino-esp32/issues/2401#issuecomment-458777266) - it works OK without Firebase libs, but with libs again reboot because of panic (this time panic is different compared to my code):
```
12:39:59.642 -> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
12:39:59.642 -> Core 0 register dump:
12:39:59.642 -> PC      : 0x4015d1d4  PS      : 0x00060330  A0      : 0x801572a1  A1      : 0x3ffe28f0  
12:39:59.676 -> A2      : 0x3ffe2938  A3      : 0x3fffffc0  A4      : 0x00000000  A5      : 0x00000004  
12:39:59.676 -> A6      : 0x3ffdf230  A7      : 0x3ffb79c8  A8      : 0x8015d1d2  A9      : 0x3ffe28e0  
12:39:59.676 -> A10     : 0x3ffe28f8  A11     : 0x3fffffc7  A12     : 0x00000006  A13     : 0x3ffe28fe  
12:39:59.676 -> A14     : 0x000000c1  A15     : 0x000000e9  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
12:39:59.710 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
12:39:59.710 -> 
12:39:59.710 -> Backtrace: 0x4015d1d4:0x3ffe28f0 0x4015729e:0x3ffe2930 0x4008e6a9:0x3ffe2960
```

mrdc, 2019-10-22:
Here is my debug log for BTE+WIFI+Firebase:

```
13:30:08.716 -> ets Jun  8 2016 00:22:57
13:30:08.716 -> 
13:30:08.716 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:30:08.750 -> configsip: 0, SPIWP:0xee
13:30:08.750 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:30:08.750 -> mode:DIO, clock div:1
13:30:08.750 -> load:0x3fff0018,len:4
13:30:08.750 -> load:0x3fff001c,len:1216
13:30:08.750 -> ho 0 tail 12 room 4
13:30:08.750 -> load:0x40078000,len:9720
13:30:08.750 -> ho 0 tail 12 room 4
13:30:08.750 -> load:0x40080400,len:6352
13:30:08.750 -> entry 0x400806b8
13:30:09.260 -> Starting BLE work!
13:30:09.396 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
13:30:09.396 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
13:30:09.906 -> .[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
13:30:10.348 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
13:30:10.382 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.49, MASK: 255.255.255.0, GW: 192.168.1.1
13:30:10.382 -> .
13:30:10.382 -> WiFi connected
13:30:10.382 -> IP address: 
13:30:10.416 -> 192.168.1.49
13:30:10.416 -> [W][HTTPClient.cpp:247] beginInternal(): unexpected protocol: https, expected http
13:30:10.416 -> [D][HTTPClient.cpp:276] beginInternal(): host: xxxxxxxxxxxxxxxxxxxxxxxxxxx.com port: 443 url: /xxxxxxxxx.json?auth=xxxxxxxxx
13:30:10.450 -> [I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
13:30:11.774 -> [D][HTTPClient.cpp:1025] connect():  connected to xxxxxxxxxxxxxxxxxxxxxxxxxxx.com:443
13:30:11.978 -> [D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
13:30:12.012 -> [D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 5
13:30:12.012 -> [D][HTTPClient.cpp:1295] writeToStreamDataBlock(): connection closed or file end (written: 5).
13:30:12.012 -> [D][HTTPClient.cpp:368] disconnect(): tcp keep open for reuse
13:30:12.012 -> 
13:30:12.733 -> Starting BLE work!
13:30:12.733 -> [D][BLEDevice.cpp:78] createServer(): >> createServer
13:30:12.733 -> [D][BLEServer.cpp:290] registerApp(): >> registerApp - 0
13:30:12.733 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegisterAppEvt (0x3ffdb918), owner: <N/A> for registerApp
13:30:12.733 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffdb918), owner: registerApp
13:30:12.733 -> [D][BLEDevice.cpp:104] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
13:30:12.733 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
13:30:12.733 -> [D][BLEServer.cpp:294] registerApp(): << registerApp
13:30:12.733 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
13:30:12.733 -> [D][BLEDevice.cpp:85] createServer(): << createServer
13:30:12.733 -> [W][ESP32BLE_WIFI_FIREBASE_NEW.ino:55] my_gatts_event_handler(): custom gatts event handler, event: 0
13:30:12.733 -> [D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
13:30:12.733 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: CreateEvt (0x3ffdb820), owner: <N/A> for createService
13:30:12.733 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: CreateEvt (0x3ffdb820), owner: createService
13:30:12.733 -> [D][BLEService.cpp:67] executeCreate(): >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
13:30:12.733 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: CreateEvt (0x3ffe2d84), owner: <N/A> for executeCreate
13:30:12.733 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: CreateEvt (0x3ffe2d84), owner: executeCreate
13:30:12.733 -> [D][BLEDevice.cpp:104] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
13:30:12.733 -> [D][BLEServer.cpp:153] handleGATTServerEvent(): >> handleGATTServerEvent: Unknown
13:30:12.733 -> [D][BLEService.cpp:201] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
13:30:12.733 -> [D][BLEService.cpp:207] setHandle(): << setHandle
13:30:12.767 -> [D][BLEService.cpp:83] executeCreate(): << executeCreate
13:30:12.767 -> [D][BLEServer.cpp:280] handleGATTServerEvent(): << handleGATTServerEvent
13:30:12.767 -> [D][BLEServer.cpp:92] createService(): << createService
13:30:12.767 -> [W][ESP32BLE_WIFI_FIREBASE_NEW.ino:55] my_gatts_event_handler(): custom gatts event handler, event: 7
13:30:12.767 -> [D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
13:30:12.767 -> [D][BLEService.cpp:232] addCharacteristic(): Adding characteristic: uuid=aeb5483e-36e1-4688-b7f5-ea07361b26a8 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
13:30:12.800 -> [D][BLEService.cpp:244] addCharacteristic(): << addCharacteristic()
13:30:12.800 -> [D][BLECharacteristic.cpp:559] setCallbacks(): >> setCallbacks: 0x3ffdb880
13:30:12.800 -> [D][BLECharacteristic.cpp:561] setCallbacks(): << setCallbacks
13:30:12.800 -> [D][BLECharacteristic.cpp:631] setValue(): >> setValue: length=22, data=4920616d204368617261637465726973746963202331, characteristic UUID=aeb5483e-36e1-4688-b7f5-ea07361b26a8
13:30:12.834 -> [D][BLECharacteristic.cpp:638] setValue(): << setValue
13:30:12.834 -> [D][BLEService.cpp:229] addCharacteristic(): >> addCharacteristic()
[... cut ...]
13:31:13.834 -> 
13:31:13.834 -> Backtrace: 0x40091f60:0x3ffe7290 0x40092191:0x3ffe72b0 0x4008df93:0x3ffe72d0 0x40159ca5:0x3ffe72f0 0x40159fed:0x3ffe7310 0x4017b194:0x3ffe7330 0x4017b8ac:0x3ffe7360 0x4017daea:0x3ffe7380 0x4017dbd3:0x3ffe73a0 0x401a9669:0x3ffe73d0 0x4016d75b:0x3ffe7430 0x4016d945:0x3ffe7450 0x4016db79:0x3ffe7490 0x401a9729:0x3ffe74b0 0x4016faed:0x3ffe7510 0x401702f1:0x3ffe7530 0x401703c7:0x3ffe7550 0x4017060e:0x3ffe7570 0x4008e6a9:0x3ffe75a0
13:31:13.866 -> 
13:31:13.866 -> Rebooting...
```

closed 2020-01-04 by stale[bot]