espressif/arduino-esp32 #2163: tools/espota.exe is detected as thread by Windows Defender

SuperIlu, 2018-12-07:
Windows Defender now detects espota.exe as thread, too.

Name: Trojan:Win32/Skeeyah.B!rfn
Thread: severe

limerickcabin, 2018-12-08:
I have the same issue. I cannot exclude it so it keeps getting deleted by Defender.

lrmoreno007, 2018-12-08:
It has also been detected, but after deleting everything, I downloaded it again from Git and it does not detect it anymore.

DerMaulwurf, 2018-12-09:
Same issue here. Defender even removes zip-archive downloaded from github.com/espressif/arduino-esp32 because of  espato.exe contained within.

stickbreaker, 2018-12-10:
I'll pile on, with Windows 7 pro, Virus definitions 1.283.285.0 10DEC2018 06:12

tralay520, 2018-12-10:
same problem, but so is the gen_esp32part.exe as well showing up as a trojan.

kthordarson, 2018-12-11:
On my machine it was also found in \AppData\Local\Temp\WCS9E13.tmp

welir, 2018-12-11:
I have same issue

gcormier, 2018-12-11:
Had the same issue. Delete it. No reason for an EXE to be here.

nEuDy, 2018-12-11:
Same issue here with extracting esp32-1.0.1-rc1 release

me-no-dev, 2018-12-12:
well... say thank you to Microsoft. They are doing this... and it's not like it's the first time... this file is nothing but the python script with the same name turned into binary with py2bin so you do not need to install python, modules and have tons of headaches. Someone needs to report this to MS so they can do their "research" and unflag it. The file has not changed recently at all...

me-no-dev, 2018-12-12:
@gcormier no reason until you decide that you want to do OTA, then you will come back to say that it does not work ;)

nEuDy, 2018-12-12:
I reported this to Microsoft. Let's wait for any further feedback and results.
Nevertheless, this is not only recognized as a virus with MS Windows Defender, also with other Definitions.
Please check the follow site, which virus scanner will detect it as malware/trojan:
https://www.virustotal.com/de/file/5e2ff3daf359cc549dd64ac03b1d349076baf9df09bc0078cd6e99d445597c61/analysis/1544657837/

DerMaulwurf, 2018-12-14:
@nEuDy Good news! Thank you for dealing with this issue!

Leonos, 2018-12-14:
Thanks a lot! One worry less...

buccaneer-jak, 2018-12-25:
On the 24th Dec 2018 Windows Defender (up to date) reported 2 Trojans, Trojan:Win32/Spursint.F!cl in gen_esp32part.exe and not sure if the second report is the same file but possible.  I may try removing previous definitions as mentioned above.
Even if a false positive this should be noted by distributors of these files.

me-no-dev, 2018-12-26:
@buccaneer-jak this has not happened in a while... I was expecting that after particular files are reported and cleared a few times, MS will do their thing and not reflag the same files that are generated by the same software on the same exact VMs... this makes no sense...

DerMaulwurf, 2018-12-26:
@buccaneer-jak I scanned the entire tree incl. gen_esp32part.exe with windows defender with up to date definitions. There was no thread found.
Maybe it's an individual problem on your machine?

buccaneer-jak, 2018-12-29:
Well I have noticed other people having this problem. I removed the virus definitions and updated them (29th Dec 2018) and got the same result.

closed 2019-08-15 by stale[bot]