espressif/arduino-esp32 #2168: If used, there is a problem with reconnection.

seongbin86, 2018-12-08:
bluetoothserial lib. is a problem with reconnection when using this.

Software serial information is being sent via Bluetooth serial.

source code is very simple.
-------------------------------------------------------------------------------------------------------------
#include "BluetoothSerial.h"
#include "HardwareSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

HardwareSerial SerialGPS(0);
BluetoothSerial SerialBT;

static const uint32_t Baud_Rate = 115200;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(Baud_Rate);
  SerialBT.begin("GNSS");
  SerialGPS.begin(Baud_Rate, SERIAL_8N1, 3, 1);
}

void loop() {  
   if (SerialBT.hasClient() > 0 && SerialGPS.available()) {
    byte receivedData = SerialGPS.read();
    SerialBT.write(receivedData);
    Serial.write(receivedData);
  }
  if (SerialBT.available()) {
    byte receivedData = SerialBT.read();
    SerialGPS.write(receivedData);
  }
}
-------------------------------------------------------------------------------------------------------------
It works well the first time it is connected to Bluetooth.
but, the second connection will be connected but no data will be received.

Please check.

seongbin86, 2018-12-09:
core version 1.0.0...

ankitsehgalaries, 2018-12-09:
What is the flash size of esp32?
I am getting this error

Sketch uses 1421706 bytes (108%) of program storage space. Maximum is 1310720 bytes.
Global variables use 78896 bytes (24%) of dynamic memory, leaving 248784 bytes for local variables. Maximum is 327680 bytes.

mouridis, 2018-12-09:
@ankitsehgalaries Have you thought of googling your error message before deciding to comment on a totally irrelevant issue? You would find a solution to your problem in the top 5 results and you wouldn't hijack another person's issue. If you don't find a solution, please create a new issue and don't post off topic comments to other issues. While you are at it, please use the issue report template.

TiSiAi, 2019-07-03:
Same problem
ESP32 Core 1  (loop) is locked avec re connexion
Core 0 is running
No crash

RealParity, 2019-08-21:
Same issue like #2650 

I have posted a minimal code example to reproduce the issue over there.

closed 2019-11-03 by stale[bot]