espressif/arduino-esp32 #2171: SD mount not working  reliable

atsteich, 2018-12-09:
### Hardware:
Board:							Adafruit HUZZAH32 Feather + SD/RTC Shield
Core Installation/update date:		05/Nov/2018
IDE name:						Platform.io
Computer OS:                                          Windows 10 and Ubuntu

### Description:
There are problems mounting micro SD cards after rebooting. 

The first time after inserting a micro SD card (e.g. Sundisk Ultra) and rebooting the chip, everything works fine. Leaving the card in the holder and rebooting again i get in ~50% of all attempts the error:
`[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)` 

After reinserting the card (or removing power supply of the sd card holder) everything works fine again, but only for the first reboot.

I found some kind of 'workaround'. If you call SD.begin(CS) in a loop until it returns true, u will be able to mount the card, sometimes after 2-3 attempts, sometimes after 10. 

I have tried with at least 3 different chips, multiple different SD card holder (some feather shields, an other micro SD holder, and an SD holder) and multiple different SD cards. -> same result for all of them. 

### Sketch:
```cpp
#include <Arduino.h>
#include <SD.h>

void setup(){
  Serial.begin(115200);
    delay(200);
    while(! SD.begin(33)){
    }
    Serial.println("SD OK");
}

void loop(){
    File file = SD.open("/testfile.txt", FILE_APPEND);
    file.println("printAnythingHerer");
    file.close();
}

```

### Debug Messages:
```ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2

load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
SD OK
```

imcort, 2018-12-10:
I have the same problem.

imcort, 2018-12-10:
I've tried to put the SD.begin(CS) into a loop and added some delays, but it doesn't have any effect.

atsteich, 2018-12-10:
@imcord Is it working after reinserting the card?

imcort, 2018-12-10:
@atsteich yes, I reinserted the card and reconnect the cable.
My SPI bus also connect a lcd screen and the screen works fine, but the sd card doesn't work even if I remove all of the wires and lcd, only plug the sd card.

imcort, 2018-12-10:
@atsteich Are you using a tf-to-sd adapter? I found that when I pinch the two sides of the adapter, then the program runs correctly.

atsteich, 2018-12-10:
I have been trying with both, tf-to-sd adapter, and with the sd/rtc feather wing from adafruit (https://www.adafruit.com/product/2922) same result... I'll try pinching the adapter later, but I do not think that this will solve the problem, since it always works after calling SD.begin(CS) multiple times. 
I think this could be some kind of buffer problem that happens when the esp is rebooting while SPI transmission is active... But to be honest, I have no idea about SPI protocoll, I will dig into as soon I find some time... I also will try adding a capacitor to the power supply of the sd card holder, maybe there are some power problems in my hardware... Although I don't think it is a hardware problem, since i have tried multiple different chips, and it works fine as soon as I reinsert the card.

atsteich, 2018-12-10:
Update: So far I can say that this is ONLY happening if the esp is rebooting while SPI transmission is active.  The "workaround (calling SD.begin(cs) until it returns true) has never failed until now, but sometimes it takes up to ~20 times. I have tried pulling cs HIGH manually before calling SD.begin(), didn't change anything. The only way to get it work every time (at least until now) is disconnecting power line to the sd card holder, or removing and reinserting the micro SD card.

luandro, 2021-01-02:
I'm having exactly this same problem.

Removing and inserting the card works somt times and pinching two sides of the adapter also helps some times. But it's getting harder and harder for it to work.

Has there been any updates on this? Is it really a hardware problem with the card adapter?

hicham-vv, 2021-03-15:
i have the same issue here

FlorianBogeschdorfer, 2021-03-30:
I had the same problems. Solved by using 5V instead of 3.3 for the VCC of the sd card adapter

acarpe13, 2021-05-12:
@FlorianBogeschdorfer Switching from 3.3v to 5v worked for me!

malbrook, 2021-07-02:
Same problem but this will be an embedded system so no chance to remove the SD card, pity there is no reset in the SD specification as I get a similar problem with W5500 SPI Ethernet chip which was sorted with by using the reset connection. As said above the SD card is an intelligent peripheral and resetting the ESP32 whislt the SD card is in use will result in an SD card left in the middle of a transaction. 
I did some research into the SD card command structure and I am now looking into the library to see how it is implemented.

acer1204, 2022-02-13:
i have some problem with always read sdcard file...
my sdcard have 100 files. 
if i "read txt"->"delete txr" ->"get next txt" do cycle always, 
about 2 min,the sd.begin will return false f_mount(0x01)[first] and f_mount(0x03)
then, the sd.begin alwayss return false...
but if i reinsert sdcard, it the sd.begin will return good,
but if i do "read txt"->"delete txr" ->"get next txt" long time, the sd.begin will read false.
does anyone has this problem?