espressif/arduino-esp32 #2182: Sending audio file node-red through Mqtt

hazee007, 2018-12-12:
Make your question, not a Statement, inclusive. Include all pertinent information:
hello guys am trying to send audio buffer to node red using the pubsubclient on esp32.
the problem is i am trying to save the audio buffer of 1kb into a char buffer since its the only way i can send data to node-red but i when i gave a char buff[1000], i get no data coming to node-red execpt i set the char buff[110] which is not enough for the voice sample i need.

components am using is esp32 with  SparkFun Electret Microphone Breakout(not worried about sound quality)

here the code below

`#include <PubSubClient.h>
#include <WiFi.h>
#include <WiFiClient.h>

const char* ssid = ""; 
const char* password = "";

const char* mqttServer = ""; 
const int mqttPort = 1883;

String str_payload;

WiFiClient espClient;
PubSubClient client(espClient);

void setup()
{
  Serial.begin(9600);
  WiFi.begin(ssid,password);
  while(WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
  client.setServer(mqttServer, mqttPort);
  while(!client.connected()) {
    Serial.println("Connecting to MQTT...");
    if(client.connect("ESP8266Client" )) {
      Serial.println("connected");
    }else{
      Serial.print("failed state ");
      Serial.print(client.state());
      delay(2000);
    }
   }
vibrate();
 my_analog();
}

void loop()
{
}

void vibrate(){
  pinMode(2, OUTPUT);
  digitalWrite(2,HIGH);
  delay(2000);
  digitalWrite(2,LOW);
   my_record();
}

void my_record()
{
  for(int y=0;y<11;y++)
  {
    for(int i=0;i<1000;i++)
    {
      str_payload += char(analogRead(35)/4);
//       Serial.println(str_payload.length());
    }
  Send();
  }
}

void my_analog()
{
  if(digitalRead(2)==LOW)
  {
    my_record();
    
  }
}

void Send()
{
  char buf[110];
  String (str_payload).toCharArray(buf,110);
  Serial.print("publishing");
  client.publish("test/esp8266",buf);
  str_payload="";
  Serial.println("finish");
}`

stickbreaker, 2018-12-13:
@hazee007 Which pubSubClient are you using? The first one Google lists has the ability to send arbitrarily large blocks.[knolleary/pubSubClient](https://github.com/knolleary/pubsubclient/blob/26ce89fa476da85399b736f885274d67676dacb8/src/PubSubClient.h#L155)

Chuck.

hazee007, 2018-12-13:
@stickbreaker yea when i give the char size of [1000]bytes, i compiles which means(i think) the library is fine but am not getting any output data at node-red even when it say's publishing on the serial monitor

stickbreaker, 2018-12-13:
What library are you using?yea, nea ichiban doesn't tell me what you are using.  

Just because code compiles, does not mean it works. 
```
uint8_t * deathByte=NILL;
*deathByte =1;
```
Will compile, but, when executed will cause immediate core dump and reboot.

Chuck.

hazee007, 2018-12-13:
alright am using pubsubclient(mqtt)  
[PubSubClient.zip](https://github.com/espressif/arduino-esp32/files/2674857/PubSubClient.zip)
version 3.1

stickbreaker, 2018-12-13:
@hazee007 you should have read the readme.md file.
>
## Limitations

 - It can only publish QoS 0 messages. It can subscribe at QoS 0 or QoS 1.
 - The maximum message size, including header, is **128 bytes** by default. This
   is configurable via `MQTT_MAX_PACKET_SIZE` in `PubSubClient.h`.
 - The keepalive interval is set to 15 seconds by default. This is configurable
   via `MQTT_KEEPALIVE` in `PubSubClient.h`.
 - The client uses MQTT 3.1.1 by default. It can be changed to use MQTT 3.1 by
   changing value of `MQTT_VERSION` in `PubSubClient.h`.

It does tell you explicitly that the maximum message size including header is 128 bytes. So, it is working as designed. 

Chuck.

hazee007, 2019-01-02:
@stickbreaker thank u, i changed the size but the quality of the sound is very bad compare to the previous pubsubclient i used which i was able to send the audio file as a string to mqtt. the previous pubsubclient i used is below
[pubsubclientesp8266.zip](https://github.com/espressif/arduino-esp32/files/2721390/pubsubclientesp8266.zip)

closed 2019-08-15 by stale[bot]