espressif/arduino-esp32 #2187: FreeRTOS

nitroge, 2018-12-13:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							M5stack
Core Installation/update date:			?11/jul/2017?
IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:					    115200
Computer OS:                                             Ubuntu

HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <FreeRTOS.h>

void vAsycTask(void *pvParameters)
{
    for (;;)
    {
        Serial.println("im ok");
        vTaskDelay(1000); //  modify to vTaskDelay(50); return to normal, but why
    }
}

void setup() {
    xTaskCreate(vAsycTask,
                "Task1",
                1024,
                NULL,
                tskIDLE_PRIORITY + 2,
                NULL);
    vTaskStartScheduler();
}

void loop() {
   Serial.println("main loop");
  delay(1000);
}
```

### Debug Messages:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
im ok
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x3ffb32b4  PS      : 0x00050133  A0      : 0x3ffb32b4  A1      : 0x3ffb1f00
A2      : 0x00000000  A3      : 0x3f4013f4  A4      : 0x3ffc2600  A5      : 0x800e558a
A6      : 0x3ffb1f70  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x800e558a
A10     : 0x3ffb1f70  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3f400ef9  SAR     : 0x00000000  EXCCAUSE: 0x00000014
EXCVADDR: 0x3ffb32b4  LBEG    : 0x00000001  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x3ffb32b4:0x3ffb1f00 0x3ffb32b1:0x3ffb1f40 0x400d09d3:0x3ffb1f70 0x400e5587:0x3ffb1fa0

lbernstone, 2018-12-13:
vTaskStartScheduler is a system function.  The scheduler is running (since the OS is)- you do not need to call it.

nitroge, 2018-12-13:
Thank you again.

closed 2018-12-13 by nitroge