espressif/arduino-esp32 #2188: Arduno-ESP32 v1.0.1rc5 's   "esp_http_client.h

TridentTD, 2018-12-14:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

Jeroen88, 2018-12-14:
In the example given here https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_http_client.html an extra check is done:

```
            if (!esp_http_client_is_chunked_response(evt->client)) {
                printf("%.*s", evt->data_len, (char*)evt->data);
            }

```

I can imagine that the response _is_ chunked an that in that case either evt->data_eln is negative and/or evt->data is undefined.

Jeroen88, 2018-12-14:
Then I can only advise you to do some basic debugging:
- decode the backtrace
- add a println before every statement
- check if client != NULL

climateguard, 2019-06-09:
Hi, Friend i passed the same proplem in the past. You need to add {} after  "esp_http_client_config_t config" . It look like: ```
esp_http_client_config_t ota_client_config {};
                ota_client_config.url = "yoururl";
                ota_client_config.cert_pem = yoursertificate;``
```

closed 2019-08-22 by stale[bot]