espressif/arduino-esp32 #2199: python3 get.py issue

LottesLiu, 2018-12-16:
### Hardware:
IDE name:							Arduino IDE
Computer OS:                                            Mac OSX
Description:
when finish clone the repository, use python3 get.py , issue occurred like blow:
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051)>

mouridis, 2018-12-16:
Accessing internet from within China, I had the same issue a couple of times in the past. The error was related to the VPN server I was using. Try using a different VPN server or totally disabling your VPN.

lbernstone, 2018-12-16:
If you are completely unable to get the get.py script to work, you can manually download the files: https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/

closed 2019-08-15 by stale[bot]