espressif/arduino-esp32 #2203: eTaskGetState() Always returns 1 (ready) even when task is active

euquiq, 2018-12-16:
Weird thing:
I declare a task handle:

`TaskHandle_t h_gsm_loop;`

Then on my loop() I start a task:

`xTaskCreate(gsm_taskloop, "gsm_taskloop", 2000, NULL, 0, &h_gsm_loop);`

The task itself outputs it's ongoing work over Serial Port, and I can see it is working.

But on the loop() each 500ms I query the task status with:

`Serial.println("L" + (String)eTaskGetState(&h_gsm_loop));`

And I still get 1 as result (task ready) instead of task running ...

Furthermore, I add the same printout inside the task itself, and from the actual task, running, I also get a 1 (task ready) value in return, instead of 0 (task running).

I think eTaskGetState() is just a dummy function ? Or most probably, I must be doing something lamely wrong.

Regards,

Enrique

chegewara, 2018-12-16:
Just put this line in your `gsm_taskloop`:
`Serial.println("L" + (String)eTaskGetState(&h_gsm_loop));`

euquiq, 2018-12-16:
@chegewara yes, as I state at my first post, such line IS there, and returns 1 (task READY) instead of 0, task Running ... which is the weird part

chegewara, 2018-12-16:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html?#_CPPv213eTaskGetState12TaskHandle_t

You should pass handle not handle reference to function.

euquiq, 2018-12-16:
@chegewara Yes I already seen that, for IDF. If I take out the "&" ... the code crashes with "GURU MEDITATION" when reaching the eTaskGetState(h_gsm_loop) ... So I would think it is different in Arduino, (or wrongly implemented, and thus, a bug, hence this issue may be a true one)

me-no-dev, 2018-12-17:
I think that you should post this in IDF's issue tracker :) better chance to get a proper response/fix for this.

closed 2019-08-15 by stale[bot]

alexvazquez, 2022-05-22:
@euquiq Hi, having same issue. Did you find out?

playmiel, 2023-04-05:
I think this problem will never be fixed

chegewara, 2023-04-06:
Im pretty sure it is working good. Here is quick test output:
```
06:44:35.166 > Task1 0
06:44:35.270 > Task2 0
06:44:36.167 > Task1 0
```
- `Task1` is some task running on core 0
- `Task2` is inside `loop()`

Like i said long time ago, you have to pass properly task handle, by value, not by reference.
This is decalaration:
```
eTaskState eTaskGetState( TaskHandle_t xTask ) PRIVILEGED_FUNCTION;
```
and this is code:
```
Serial.println("Task2 " + (String)eTaskGetState(&h_gsm_loop)); <------------ WRONG
Serial.println("Task2 " + (String)eTaskGetState(h_gsm_loop));  <------------ GOOD
```

playmiel, 2023-04-07:
thanks I thought it was a bug now it works