espressif/arduino-esp32 #2205: Guru Meditation Error LoadProhibited - Code Worked Last Week

romankc, 2018-12-17:
I am seeing the standard "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled." error.

Under normal circumstances, I would conclude that it is somehow my fault and that there is a bug in my code. **However, in this case, the code that I am uploading worked perfectly fine last week.**

I tried uploading to multiple different ESP32 modules but all of them have the same problem.
I did not manually update any libraries or tools before this problem started.

ESP32 modules which are running the same code (compiled 1+ weeks ago) are working without any problems.

The crash occurs while using the painlessMesh library whenever the device attempts to establish a connection to another device.

Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Auto-detected Flash size: 4MB

[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
{ "nodes": [{ "nodeId":3298747729}]}
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:386] _eventCallback(): STA IP: 10.37.17.2, MASK: 255.255.255.0, GW: 10.37.17.1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d47c0  PS      : 0x00060130  A0      : 0x800d4fa0  A1      : 0x3ffce5b0  
A2      : 0x3ffc561c  A3      : 0x3ffce61c  A4      : 0x00000000  A5      : 0xc49ee551  
A6      : 0x00000005  A7      : 0x3ffce624  A8      : 0x00000001  A9      : 0x00000000  
A10     : 0x3fff0fd8  A11     : 0x3ffc561c  A12     : 0x3ffc5750  A13     : 0x00000000  
A14     : 0x3ffce624  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xfffffffd  

Backtrace: 0x400d47c0:0x3ffce5b0 0x400d4f9d:0x3ffce600 0x400d33ba:0x3ffce650 0x400d34d3:0x3ffce670 0x400d345a:0x3ffce690 0x400d1d2a:0x3ffce6b0 0x4019fd44:0x3ffce6d0 0x40094d01:0x3ffce6f0

Rebooting...

romankc, 2018-12-17:
Here are two other reports of the same problem:
#1661 
#602 - see comment from @borolaci

beegee-tokyo, 2018-12-30:
@romankc 
Having the same problem with painlessMesh.

From my logs I see that it happens when a connection change is detected.
``` LOG
[MES_TSK] Changed connection
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060530  A0      : 0x800ee018  A1      : 0x3ffdea40  
A2      : 0xcd4c7027  A3      : 0xcd4c7023  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x8000beca  A9      : 0x3ffde9f0  
A10     : 0x3ffdeee0  A11     : 0x3ffbaa18  A12     : 0xffffffff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xcd4c7027  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400014dc:0x3ffdea40 0x400ee015:0x3ffdea50 0x400ee131:0x3ffdea70 0x400e4e02:0x3ffdea90 0x40097239:0x3ffdede0

Rebooting...
ets Jun  8 2016 00:22:57
```
The exception decoder unfortunately doesn't help, the only output I get is 
```
PC: 0x400014dc
EXCVADDR: 0xcd4c7027

Decoding stack results
0x40097239: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
Which doesn't help at all.

I think this issue should be posted in the painlessMesh repo.

romankc, 2019-02-19:
This is now resolved: https://gitlab.com/painlessMesh/painlessMesh/issues/221

closed 2019-02-19 by romankc

DenniZr, 2020-10-28:
that's chinese spyware