espressif/arduino-esp32 #2206: How can i pass error emac: Timed out waiting for PHY register?

smacyas, 2018-12-17:
I use wroom32 + lan8720 + wifi

When lan8720 is broken (emac: Timed out waiting for PHY register) the program stops and error loops. How can i pass this error end contunue program, exclude LAN and use wifi?

me-no-dev, 2018-12-17:
how do you hit this issue?

smacyas, 2018-12-17:
I have some boards in which the lan chip is clocked GPIO0, more recently such opportunities with 50 MHz from this pin not supported, and the device after some upgrade of firmware do not work, even those who are not using lan, want his error were ignored and not fixate survey lan.

It is very bad, since it is impossible to support older devices with 50 Mhz GPIO0 pin.

Now it is loop:
E (5556) emac: Timed out waiting for PHY rdgister 0x2 to have value 0x0022 (mask 0xffff). Current value 0xffff
E (6556) emac: Timed out waiting for PHY register 0x3 to have value 0x1430 (mask 0xfff0). Current value 0xffff
E (7556) emac: Timed out waiting for PHY register 0x2 to have value 0x0022 (mask 0xffff). Current value 0xffff
E (8556) emac: Timed out waiting for PHY register 0x3 to have value 0x1430 (mask 0xfff0). Current value 0xffff
E (9556) emac: Timed oet waiting for PHY register 0x2 to have value 0x0022 (mask 0xffff). Current value 0xffff
E (10556) emac: Timed out waiting for PHY register 0x3 to have value 0x1430 (mask 0xfff0). Current value 0xffff
E (11556) emac: Timed out waiting for PHY register 0x2 to have value 0x0022 (mask 0xffff). Current value 0xffff
E (12556) emac: Timed out waiting for PHY register 0x3 to have value 0x1430 (mask 0xfff0). Current value 0xffff

me-no-dev, 2018-12-18:
@igrr looking at the idf code, the default setting is 1000. What should we adjust it to to get this working?

closed 2019-08-15 by stale[bot]