espressif/arduino-esp32 #2207: Sensor VEML770  not work with last 1.0.1 rc2

asetyde, 2018-12-17:
After some correction and disabling ALS algorithm , with 1.0.1 rc1 all working, today I test rc2 sensor return only 0 and 1511.5199 value .. something wrong ... @stickbreaker  something change from rc1 to rc2 ?

stickbreaker, 2018-12-17:
yes, we decided to remove the I2C_ERROR_CONTINUE.

change any code that looks for `err!=7` back to `err != 0`

```c++
// 1.0.1Rc2
uint8_t err = Wire.endTransmission(false);
if(err != 0 ) { // problem
   return err;
} else { // everything good
  Wire.requestFrom(id,len);
  while ( Wire.available()){
     Wire.read();
  }
}
```

Chuck.

asetyde, 2018-12-18:
I use your variant https://github.com/stickbreaker/arduino-VEML7700

asetyde, 2018-12-18:
[image]
Also seems it's not correct loaded from compiler, esp32 part of code is obfuscated as unused

asetyde, 2018-12-18:
Also I'm not sure that change influence a result , at now library result only 151.519...

asetyde, 2018-12-18:
at now value is unusable, I came back to rc1

stickbreaker, 2018-12-18:
cut out that #ifdef ,  The `Wire()` coding has changed, there is no longer any difference between ESP32 coding and AVR Arduino coding.

Just change it to:
```
if(Wire.endTransmission(false)){ // NB: don't send stop here
    return STATUS_ERROR;
}
```

Chuck.

asetyde, 2018-12-19:
Ok as I've done . Thanks Chuck , your re best

closed 2018-12-19 by asetyde