espressif/arduino-esp32 #2211: Delay(x)  delays 3 times more than expected while running esp32 at 80MHz clock speed

rohitsam, 2018-12-18:
**Hardware:**
Board: ESP-WROOM-32 with a break out board.
Core Installation date:  04/09/2018
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Windows 10

**### Description:**
the delay function delays 3 times more than expected while running at 80Mhz clock speed please kindly help me with the solution for this problem?

**code:**
#include <soc/rtc.h>
void setup() {

  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
  Serial.begin(115200);
  
  Serial.println("starting_2_sec_delay");
  delay(2000);
  Serial.println("2sec_delay_ended");
  
  Serial.println("starting_5_sec_delay");
  delay(5000);
  Serial.println("5sec_delay_ended");
  
  Serial.println("starting_10_sec_delay");
  delay(10000);
  Serial.println("10sec_delay_ended");

  Serial.println("starting_20_sec_delay");
  delay(20000);
  Serial.println("20sec_delay_ended");

}

void loop() {
  // put your main code here, to run repeatedly:

}

### Debug Messages:
[image]

me-no-dev, 2018-12-18:
Arduino does not handle well cpu clock switching yet. You drop the clock, it's normal the time to be wrong as well. It's expecting 240 clocks for each microsecond, yet it's getting only 80.

rohitsam, 2018-12-18:
#define delay(x) delay(x/3)

this is one workaround.

me-no-dev, 2018-12-18:
I should fix this...

me-no-dev, 2018-12-18:
adding new API :)

https://github.com/espressif/arduino-esp32/commit/66d33f792c1ee44555d6b812bba77d653ad31220

```
2018-12-18 20:00:00	ets Jun  8 2016 00:22:57
2018-12-18 20:00:00	
2018-12-18 20:00:00	rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2018-12-18 20:00:00	configsip: 0, SPIWP:0xee
2018-12-18 20:00:00	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2018-12-18 20:00:00	mode:DIO, clock div:1
2018-12-18 20:00:00	load:0x3fff0018,len:4
2018-12-18 20:00:00	load:0x3fff001c,len:1100
2018-12-18 20:00:00	load:0x40078000,len:9296
2018-12-18 20:00:00	load:0x40080400,len:6364
2018-12-18 20:00:00	entry 0x400806a4
2018-12-18 20:00:00	
2018-12-18 20:00:00	starting_2_sec_delay
2018-12-18 20:00:02	2sec_delay_ended
2018-12-18 20:00:02	starting_5_sec_delay
2018-12-18 20:00:07	5sec_delay_ended
2018-12-18 20:00:07	starting_10_sec_delay
2018-12-18 20:00:17	10sec_delay_ended
2018-12-18 20:00:17	starting_20_sec_delay
2018-12-18 20:00:37	20sec_delay_ended
```

closed 2018-12-18 by me-no-dev