espressif/arduino-esp32 #2212: Occasional OTA crash

poengAlex, 2018-12-18:
### Hardware:
Board:							ESP32-wroom-32D (custom board)
Core Installation/update date:			1.0.0?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
Hi

I have build a simple test program to check the OTA functionality on ESP32. There are 2 versions of a bin fil I download with on integer value different in the two. Both programs start OTA and download the other bin file. If the same bin fil starts after a reset the OTA must have failed somehow. I also have a timer that restarts the MCU after 5 minutes.
After running the program over the weekend I have 6722 successful OTA and 162 fails. 2.4% fail rate. There is a few instances (<5) where the MCU simply freeze and do the restart failsafe kicks in. But the most of the failures there is a LoadStoreError. The full exception you can find bellow.
Do anyone else have this problem? Any solutions?

### Sketch: 
```cpp

	Update.onProgress(OnProgress); 
	ESPhttpUpdate.rebootOnUpdate(false);
	t_httpUpdate_return ret = ESPhttpUpdate.update(bin);

	switch (ret) {
	case HTTP_UPDATE_FAILED:
		Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
		vTaskDelay(1000);
		ESP.restart();
		break;

	case HTTP_UPDATE_NO_UPDATES:
		Serial.println("HTTP_UPDATE_NO_UPDATES");
		vTaskDelay(1000);
		ESP.restart();
		break;

	case HTTP_UPDATE_OK:
		Serial.println("HTTP_UPDATE_OK");
		vTaskDelay(1000);
		ESP.restart();
		break;
	}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d8f3c  PS      : 0x00060930  A0      : 0x800d8fdc  A1      : 0x3ffb1d00  
A2      : 0x3ffbc01c  A3      : 0x3ffb1d4f  A4      : 0x00000001  A5      : 0x40000000  
A6      : 0x00043814  A7      : 0x04d8e925  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x3ffbc7ec  A12     : 0x00000050  A13     : 0x3ffb1c10  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Decoding 12 results
0x400d8f3c: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi\src/WiFiClient.cpp line 507
0x400d8fd9: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi\src/WiFiClient.cpp line 507
0x40161442: WiFiClient::operator bool() at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi\src/WiFiClient.cpp line 507
0x400db219: Print::println(int, int) at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Print.cpp line 227
0x400db245: Stream::timedRead() at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Stream.cpp line 279
0x400d8bb2: UpdateClass::writeStream(Stream&) at C:\Users\test\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Update\src/Updater.cpp line 308 (discriminator 2)
0x400da391: ESP32HTTPUpdate::runUpdate(Stream&, unsigned int, String, int) at C:\Program Files (x86)\Arduino\libraries\esp32-http-update\src/ESP32httpUpdate.cpp line 33
0x400da713: ESP32HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at C:\Program Files (x86)\Arduino\libraries\esp32-http-update\src/ESP32httpUpdate.cpp line 33
0x400da896: ESP32HTTPUpdate::update(String const&, String const&) at C:\Program Files (x86)\Arduino\libraries\esp32-http-update\src/ESP32httpUpdate.cpp line 33
0x400d202e: SimpleOta(char const*, char const*) at C:\atechcomp\OneDrive - atechcomp.no\SW\Arduino\ota_test/ButtonHandling.ino line 47
0x400d234d: setup() at C:\atechcomp\OneDrive - atechcomp.no\SW\Arduino\ota_test/ButtonHandling.ino line 47
0x40145f7f: _GLOBAL__sub_I__ZNSt12ctype_bynameIwEC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEj at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_facets_nonio.h line 754
:  (inlined by) _GLOBAL__sub_I__ZNSt12ctype_bynameIwEC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEj at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/src/c++11/locale-inst.cc line 380

```

poengAlex, 2018-12-18:
App core (from setup()).

which version of the core are you running?

poengAlex, 2018-12-19:
How can I check that?

joseitor, 2018-12-19:
I got a similar error
 `Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400da9e8  PS      : 0x00060230  A0      : 0x800daa8c  A1      : 0x3ffcf320  
A2      : 0x3ffe4998  A3      : 0x3ffcf36f  A4      : 0x00000001  A5      : 0x40000000  
A6      : 0x0001b3ec  A7      : 0x2c2f2978  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x3ffe4c14  A12     : 0x00000050  A13     : 0x3ffcf220  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400da9e8:0x3ffcf320 0x400daa89:0x3ffcf340 0x4021478b:0x3ffcf360 0x400e5384:0x3ffcf390 0x400e53b1:0x3ffcf3b0 0x400e47c2:0x3ffcf3d0 0x400e39be:0x3ffcf3f0 0x400e3fae:0x3ffcf430 0x400e420d:0x3ffcf520 0x400d63ab:0x3ffcf5c0 0x400d3c8a:0x3ffcf620 0x400d7259:0x3ffcfaf0 0x401e663c:0x3ffcfb40 0x40097535:0x3ffcfb60`

But it happens to me all the time, and I code is just the excample code introduccing my SSID, PASSWORD and URL where the .bin file is.

Thanks!

nikuz, 2018-12-31:
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/AWS_S3_OTA_Update
```bash
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d25c4  PS      : 0x00060a30  A0      : 0x800d2667  A1      : 0x3ffb1df0  
A2      : 0x3ffbb2fc  A3      : 0x3ffb1e3f  A4      : 0x00000001  A5      : 0x40000000  
A6      : 0x00044b60  A7      : 0xff000000  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x00548f70  A11     : 0x3ffbb4a0  A12     : 0x8008eb2c  A13     : 0x3ffb1d00  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000b  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d25c4:0x3ffb1df0 0x400d2664:0x3ffb1e10 0x40142c6e:0x3ffb1e30 0x400d4009:0x3ffb1e60 0x400d4035:0x3ffb1e80 0x400d39fa:0x3ffb1ea0 0x400d205d:0x3ffb1ec0 0x400d232f:0x3ffb1f90 0x400d4e21:0x3ffb1fb0 0x4008e919:0x3ffb1fd0

0x400d25c4: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x400d2664: WiFiClient::read(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x40142c6e: WiFiClient::read() at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x400d4009: Stream::timedRead() at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.cpp:279
0x400d4035: Stream::readBytes(char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.cpp:279
0x400d39fa: Stream::readBytes(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.h:103
 (inlined by) UpdateClass::writeStream(Stream&) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/Update/src/Updater.cpp:342
0x400d205d: execOTA() at /Users/nikolay/Documents/Arduino/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino:158
0x400d232f: loop() at /Users/nikolay/Documents/Arduino/AWS_S3_OTA_Update/AWS_S3_OTA_Update.ino:221
0x400d4e21: loopTask(void*) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp:25
0x4008e919: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```

https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdate/examples/httpUpdate/httpUpdate.ino
```bash
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2024  PS      : 0x00060e30  A0      : 0x800d20c7  A1      : 0x3ffb1ca0  
A2      : 0x3ffbb684  A3      : 0x3ffb1cef  A4      : 0x00000001  A5      : 0x40000000  
A6      : 0x000390f4  A7      : 0xff000000  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x00e118af  A11     : 0x3ffc6f0c  A12     : 0x8008eb2c  A13     : 0x3ffb1ba0  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000000b  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d2024:0x3ffb1ca0 0x400d20c4:0x3ffb1cc0 0x40145992:0x3ffb1ce0 0x400d5ffd:0x3ffb1d10 0x400d6029:0x3ffb1d30 0x400d570a:0x3ffb1d50 0x400d4956:0x3ffb1d70 0x400d4f4a:0x3ffb1db0 0x400d51a9:0x3ffb1ea0 0x400d1c9e:0x3ffb1f40 0x400d7359:0x3ffb1fb0 0x4008e919:0x3ffb1fd0

0x400d2024: WiFiClientRxBuffer::read(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x400d20c4: WiFiClient::read(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x40145992: WiFiClient::read() at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/WiFi/src/WiFiClient.cpp:521
0x400d5ffd: Stream::timedRead() at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.cpp:76
0x400d6029: Stream::readBytes(char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.cpp:76
0x400d570a: Stream::readBytes(unsigned char*, unsigned int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/Stream.h:103
 (inlined by) UpdateClass::writeStream(Stream&) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/Update/src/Updater.cpp:342
0x400d4956: HTTPUpdate::runUpdate(Stream&, unsigned int, String, int) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/HTTPUpdate/src/HTTPUpdate.cpp:45
0x400d4f4a: HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/HTTPUpdate/src/HTTPUpdate.cpp:45
0x400d51a9: HTTPUpdate::update(WiFiClient&, String const&, String const&) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/libraries/HTTPUpdate/src/HTTPUpdate.cpp:45
0x400d1c9e: loop() at /var/folders/wt/t_3rnhg91bs7_m_l448f7fp80000gn/T/arduino_modified_sketch_699172/httpUpdate.ino:53 (discriminator 3)
0x400d7359: loopTask(void*) at /Users/nikolay/Library/Arduino15/packages/esp32/hardware/esp32/1.0.0/cores/esp32/main.cpp:25
0x4008e919: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
```

Looks like it always related to the WiFiClientRxBuffer::read.

stickbreaker, 2018-12-31:
LoadStoreFailure is a Buffer location error, the referenced data location is not within the Cpu's RAM. Set Core Debug Level to ERROR or WARNING and rerun your test. See if the RAM allocation error is listed in the console log.

Chuck.

nikuz, 2018-12-31:
> LoadStoreFailure is a Buffer location error, the referenced data location is not within the Cpu's RAM. Set Core Debug Level to ERROR or WARNING and rerun your test. See if the RAM allocation error is listed in the console log.
> 
> Chuck.

@stickbreaker I have tried it already. There were nothing in terminal output. I have tried those both debug levels.

stickbreaker, 2018-12-31:
@nikuz Reading through HttpClient. it looks like it is a heap fragmentation error. `Strings` are used extensively. 
Try adding a heapfree call before and after the update()
``` 
Serial.printf("Free internal heap Before update %u\n", ESP.getFreeHeap());

    t_httpUpdate_return ret = httpUpdate.update(client, "http://server.com/public_html/V1.0/sketch.bin");

Serial.printf("Free internal heap After update %u\n", ESP.getFreeHeap());

```

Chuck.

atanisoft, 2018-12-31:
String and Stream both had a few bugs in them when exactly 64 bytes was printed, these were fixed on master a week or so ago. There may be other heap leaks in there still though.

nikuz, 2018-12-31:
@stickbreaker same behaviour even with the heapfree call. I have added the additional log for progress monitoring. Looks like LoadStoreError always happens on the 237568 bytes of stream download. I have tried to update to two different *.bin sketches, which has a not the same size..

nikuz, 2018-12-31:
One time I got this error with debug log ERROR enabled, right before LoadStoreError:
```bash
(34214) esp_image: invalid segment length 0xfefefefe
[E][HTTPUpdate.cpp:408] runUpdate(): Update.end failed! (Could Not Activate The Firmware)
```

nikuz, 2018-12-31:
Sometimes updates happens. I'll check, maybe something wrong in my environment. Maybe I'll move the update file somewhere closer to the esp itself.

lbernstone, 2018-12-31:
I am able to reproduce with a minimal sketch. I'll dig in and take a look.
```
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

#define MYSSID "ssid"
#define MYPWD "password"
#define SKETCH_URL "http://s3.amazonaws.com/esp32-lb/wifikittest.ino.bin"

void setup() {
  Serial.begin(115200);
  WiFi.begin(MYSSID,MYPWD);
  if (WiFi.waitForConnectResult() == ESP_OK) {
    log_e("Connection Failed!");
    return;
  }

  WiFiClient client;
  t_httpUpdate_return ret = httpUpdate.update(client, SKETCH_URL);
}

void loop() {}
```
Monitor:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9296
load:0x40080400,len:6364
entry 0x400806a4
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:386] _eventCallback(): STA IP: 192.168.1.125, MASK: 255.255.255.0, GW: 192.168.1.1
[V][HTTPClient.cpp:224] beginInternal(): url: http://s3.amazonaws.com/esp32-lb/wifikittest.ino.bin
[D][HTTPClient.cpp:265] beginInternal(): host: s3.amazonaws.com port: 80 url: /esp32-lb/wifikittest.ino.bin
[E][WiFiClient.cpp:236] setSocketOption(): 1006 : 9
[D][HTTPClient.cpp:977] connect():  connected to s3.amazonaws.com:80
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'x-amz-id-2: vXsnIvsyrS3x0zpMwzBeUncbgBSczQLMG1L1iNqtP7JSwF0dT6m+VElbrhtRHVTut4Vb0QLLlOA='
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'x-amz-request-id: BA8EDAE43EC93485'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Date: Mon, 31 Dec 2018 16:44:55 GMT'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Last-Modified: Mon, 31 Dec 2018 16:43:34 GMT'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'ETag: "0d6937abaf803897bffdd8ba37347ed4"'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Content-Type: application/x-sega-cd-rom'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Content-Length: 892208'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Server: AmazonS3'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: 'Connection: close'
[V][HTTPClient.cpp:1072] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1102] handleHeaderResponse(): code: 200
[D][HTTPClient.cpp:1105] handleHeaderResponse(): size: 892208
[D][HTTPUpdate.cpp:224] handleUpdate(): Header read fin.

[D][HTTPUpdate.cpp:225] handleUpdate(): Server header:

[D][HTTPUpdate.cpp:226] handleUpdate():  - code: 200

[D][HTTPUpdate.cpp:227] handleUpdate():  - len: 892208

[D][HTTPUpdate.cpp:233] handleUpdate(): ESP32 info:

[... 46 lines cut ...]

[D][HTTPUpdate.cpp:227] handleUpdate():  - len: 892208

[D][HTTPUpdate.cpp:233] handleUpdate(): ESP32 info:

[D][HTTPUpdate.cpp:234] handleUpdate():  - free Space: 2097152

[D][HTTPUpdate.cpp:235] handleUpdate():  - current Sketch Size: 748496

[D][HTTPUpdate.cpp:288] handleUpdate(): runUpdate flash...

[D][Updater.cpp:136] begin(): OTA Partition: app1
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1fd0  PS      : 0x00060e30  A0      : 0x800d2073  A1      : 0x3ffb1c90  
A2      : 0x3ffbb6b8  A3      : 0x3ffb1cdf  A4      : 0x00000001  A5      : 0x40000000  
A6      : 0x0003737c  A7      : 0x8488faf6  A8      : 0x00000000  A9      : 0xffffffff  
A10     : 0x00000000  A11     : 0x3ffc8c84  A12     : 0x00000050  A13     : 0x3ffb1ba0  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d1fd0:0x3ffb1c90 0x400d2070:0x3ffb1cb0 0x401451fa:0x3ffb1cd0 0x400d5db1:0x3ffb1d00 0x400d5ddd:0x3ffb1d20 0x400d55c2:0x3ffb1d40 0x400d4652:0x3ffb1d60 0x400d4d79:0x3ffb1da0 0x400d5045:0x3ffb1ea0 0x400d1ca0:0x3ffb1f40 0x4012609b:0x3ffb1fb0 0x4008eed9:0x3ffb1fd0

Rebooting...
```

lbernstone, 2018-12-31:
Looks like @Jeroen88 has been working on HTTPUpdate.  Looping him in.

lbernstone, 2018-12-31:
The stream is failing during a read at inconsistent times.  The exact failure point is here:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/Updater.cpp#L339
But, it seems like this is probably a result of passing the stream around.  The stream is defined [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPUpdate/src/HTTPUpdate.cpp#L274) and then passed over to Updater.cpp.  Maybe a localization problem?

nikuz, 2019-01-01:
HTTPUpdate works just fine if I'm doing update from my local network.

poengAlex, 2019-01-01:
Here is the full code if someone wants to try reproducing.

```
/*
    Name:       ota_test2.ino
    Created:	18-Dec-18 15:21:32
    Author:     poengAlex
*/

/*
	Old test results
	ota_test.ino: (bin file apx 1.2Mb)
	[otaTest:133] successes: 7191
	[otaTest:134] failes: 175
*/

#define TIMEMS (uint32_t) esp_timer_get_time() / 1000
#include <Preferences.h>
#include <FS.h>
#include <SPIFFS.h>
#include <esp_partition.h>

#include <HTTPClient.h>
#include <WiFiClientSecure.h>
#include <Update.h>
#include <Wifi.h>
#include <ESP32httpUpdate.h>

#define BIN1_URL "http://YOURURL.com/test1.bin"
#define BIN2_URL "http://YOURURL.com/test2.bin"

void SimpleOta(const char *newFwVersion, const char *bin);
void OnProgress(int progress, int totalt);

int otaVersion = 2; //Make 2 bin files. One with otaVersion = 1 and one with otaVersion = 2
int progressPercent = 0;

const char PRINTTAG[] = "otaTest";

int loopTestRestart = 5 * 60 * 1000; //Max 5 min to do OTA or restart
uint32_t startTime = 0;

/*
	Restart if freeze
*/
void debugReset(void * ignore) {
	for (;;) {
		int usedTime = TIMEMS - startTime;
		Serial.print("usedTime: ");
		Serial.println(usedTime);
		Serial.print("TIMEMS: ");
		Serial.println(TIMEMS);

		if (usedTime > loopTestRestart && true) {
			Serial.println("\n\n\nRESTARTING!\n\n");
			ESP.restart();
		}
		vTaskDelay(60000);
	}
}

void setup()
{
	Serial.begin(115200);
	Serial.setDebugOutput(true);
	Serial.println("Starting!");

	Serial.print("otaVersion: ");Serial.println(otaVersion);
	startTime = TIMEMS;

	xTaskCreatePinnedToCore(
		debugReset,		/* function to implement the task */
		"debugReset",	/* name of the task */
		8000,	/* stack size in words */
		NULL,					/* task input parameter */
		200,     /* priority of the task*/
		NULL,					/* task handle. */
		1);		/* core where the task should run Core 1 is loop */

	//Connect to wifi
	WiFi.mode(WIFI_STA);
	WiFi.enableSTA(true);
	WiFi.begin("esp32", "12345678");
	Serial.println("Connecting to STA");
	int startTimeConnect = TIMEMS;
	bool timeOutConnect = false;
	while (WiFi.status() != WL_CONNECTED && !timeOutConnect && true) {
		int used = (TIMEMS - startTimeConnect); //[ms]
		timeOutConnect = used > 5000;
		vTaskDelay(100);
		Serial.print(".");
	}
	if (WiFi.status() == WL_CONNECTED) {
		Serial.print("Connected! IP address: ");
		Serial.println(WiFi.localIP());
	}
	else {
		Serial.println("\n\n\nERROR:\nCannot connect to wifi\n\n\n");
	}

	//Keep track of versions and successes
	Preferences preference;
	preference.begin("otaTest");
	int previousVersion = preference.getInt("version", -1);
	int successes = preference.getInt("successes", 0);
	int failes = preference.getInt("failes", 0);

	if (previousVersion != otaVersion) {
		Serial.println("OTA successfull!");
		successes++;
	}
	else if (previousVersion != -1) { //Do not count first time
		Serial.println("OTA failed!");
		failes++;
	}

	Serial.print("successes: "); Serial.println(successes);
	Serial.print("failes: "); Serial.println(failes);

	preference.putInt("version", otaVersion);
	preference.putInt("successes", successes);
	preference.putInt("failes", failes);

	if (true) { //Reset
		preference.putInt("version", -1);
		preference.putInt("successes", 0);
		preference.putInt("failes", 0);
	}

	preference.end();

	String otaBin = "";
	//Start OTA
	if (otaVersion == 1) {
		otaBin = BIN2_URL;
	}
	else {
		otaBin = BIN1_URL;
	}
	SimpleOta("otaTest1", otaBin.c_str());
}

int minuts = 0;
void loop()
{
	vTaskDelay(60000);
	minuts++;
	Serial.print("minuts: "); Serial.println(minuts);
}

void SimpleOta(const char *newFwVersion, const char *bin) {
	Update.onProgress(OnProgress);
	ESPhttpUpdate.rebootOnUpdate(false);
	t_httpUpdate_return ret = ESPhttpUpdate.update(bin);

	switch (ret) {
	case HTTP_UPDATE_FAILED:
		Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
		vTaskDelay(1000);
		ESP.restart();
		break;

	case HTTP_UPDATE_NO_UPDATES:
		Serial.println("HTTP_UPDATE_NO_UPDATES");
		vTaskDelay(1000);
		ESP.restart();
		break;

	case HTTP_UPDATE_OK:
		Serial.println("HTTP_UPDATE_OK");
		vTaskDelay(1000);
		ESP.restart();
		break;
	}
}

int last = 0;
void OnProgress(int progress, int totalt) {
	progressPercent = (100 * progress) / totalt;
	if (last != progressPercent && progressPercent % 10 == 0) {
		//print every 10%
		Serial.print(progressPercent);
		Serial.print(" ");
	}
	last = progressPercent;
}
```

nikuz, 2019-01-01:
> HTTPUpdate works just fine if I'm doing update from my local network.

Still not stable.

Jeroen88, 2019-01-01:
@poengAlex, @joseitor, @nikuz and @lbernstone, could you please test PR #2259? I could reproduce the error with the sketch posted by @lbernstone and it ran without error after this fix. I think it will solve your issues (and possibly a lot more!)

nikuz, 2019-01-01:
@Jeroen88 for me it works very well! Thank you very much! I have tested on three different devices. And update successfully passed every time.

Jeroen88, 2019-01-01:
@nikuz that's nice, thnx for the quick testing!

lbernstone, 2019-01-05:
This issue was almost certainly fixed with https://github.com/espressif/arduino-esp32/pull/2263.  Please pull the new version of WiFiClient.cpp and see if that fixes your problem.  Otherwise, you will need to open a new issue.

closed 2019-08-15 by stale[bot]

umartechboy, 2020-01-12:
I can see that the issue is closed. It didn't help me, but I helped myself and want to share the resolve here.

In my case, I used ESP32 core downloaded in Dec 2019. It gave me random issues while OTA, mostly, when calling Upload.end, sometimes, it gave me MD5 mismatch error, sometime, it said it couldn't read from the flash, and most of the times it said that it couldn't activate the boot partition.

Since I was observing brown out detection during several high power tasks in the code, I first tried disabling brown out detection in the startup code using WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector. This solved my code issues but not the OTA issues.

On my ESP32-Cam, the brown out was occurring most probably since I was using an old Arduino mega to power it up which used a over current protection resistor in series to the main USB supply. I shortened it and viola! problem solved.... 

Disabling brown out isn't good, in my opinion. It should never occur. The cam and the WiFi requires quite a lot of power that needs to be provided otherwise it even reduces WiFi transfer speeds.

ketan, 2021-01-28:
I can confirm that this issue is not fixed yet. I'm using a version with this patch applied and see the following occassional stacktrace:

```
t:63700ms) (loop)(C1) esp_get_free_heap_size=14196,esp_get_minimum_free_heap_size=1216,ESP.getFreeHeap=35152,ESP.getMinFreeHeap=22152,ESP.getHeapSize=240008,ESP.getMaxAllocHeap=20956
abort() was called at PC 0x401b7307 on core 1

Backtrace: 0x400942d4:0x3ffcfaa0 0x40094505:0x3ffcfac0 0x401b7307:0x3ffcfae0 0x401b734e:0x3ffcfb00 0x401b73fb:0x3ffcfb20 0x401b747e:0x3ffcfb40 0x401b7495:0x3ffcfb60 0x401e83a3:0x3ffcfb80 0x401bc3de:0x3ffcfbc0 0x400d9361:0x3ffcfbe0 0x400d1e54:0x3ffcfc00 0x400dcc89:0x3ffcfc40 0x40090835:0x3ffcfc60
  #0  0x400942d4:0x3ffcfaa0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x40094505:0x3ffcfac0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x401b7307:0x3ffcfae0 in __cxxabiv1::__terminate(void (*)()) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #3  0x401b734e:0x3ffcfb00 in std::terminate() at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
  #4  0x401b73fb:0x3ffcfb20 in __cxa_throw at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
  #5  0x401b747e:0x3ffcfb40 in operator new(unsigned int) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
  #6  0x401b7495:0x3ffcfb60 in operator new[](unsigned int) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32
  #7  0x401e83a3:0x3ffcfb80 in WiFiUDP::parsePacket() at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp:129
  #8  0x401bc3de:0x3ffcfbc0 in ArduinoOTAClass::handle() at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/libraries/ArduinoOTA/src/ArduinoOTA.cpp:89
  #9  0x400d9361:0x3ffcfbe0 in OTAHelper::handleOTA() at lib/ota_helper/ota_helper.cpp:54
  #10 0x400d1e54:0x3ffcfc00 in loop() at src/main.cpp:165
  #11 0x400dcc89:0x3ffcfc40 in loopTask(void*) at /Users/ketanpadegaonkar/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
  #12 0x40090835:0x3ffcfc60 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

mbilsky, 2021-02-18:
I too am having the same issue. It happens when I start streaming video from the ov2640 (using the example code). If I disable the OTA handle function, everything else works OK. I'm not trying to update code and stream at the same time. ESP32V1 - D0WD

Before the call there is about 44K of heap available.

Stack trace:
0x40091da0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x40091fd1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4017027b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x401702c2: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4016f913: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x4016fb3a: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x4016faf1: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc line 32
0x40170c87: WiFiUDP::parsePacket() at C:\Users\matt\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiUdp.cpp line 210
0x400f90d8: ArduinoOTAClass::handle() at C:\Users\matt\.platformio\packages\framework-arduinoespressif32\libraries\ArduinoOTA\src\ArduinoOTA.cpp line 367
0x400dd91d: loop() at src\main.cpp line 378
0x400fc2bd: loopTask(void*) at C:\Users\matt\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x4008e4b5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

me-no-dev, 2021-02-18:
I would guess that the heap could be so fragmented that it can not allocate a continuous buffer for 1460 bytes. What I did not expect is `abort` to be called from the toolchain if `new` failed. Maybe using `malloc` and `free` instead could get around the crash. Can you try and replace `int WiFiUDP::parsePacket()` with the following code?

```cpp
int WiFiUDP::parsePacket(){
  if(rx_buffer)
    return 0;
  struct sockaddr_in si_other;
  int slen = sizeof(si_other) , len;
  char * buf = (char *)malloc(1460);
  if(!buf){
    return 0;
  }
  if ((len = recvfrom(udp_server, buf, 1460, MSG_DONTWAIT, (struct sockaddr *) &si_other, (socklen_t *)&slen)) == -1){
    free(buf);
    if(errno == EWOULDBLOCK){
      return 0;
    }
    log_e("could not receive data: %d", errno);
    return 0;
  }
  remote_ip = IPAddress(si_other.sin_addr.s_addr);
  remote_port = ntohs(si_other.sin_port);
  if (len > 0) {
    rx_buffer = new cbuf(len);
    rx_buffer->write(buf, len);
  }
  free(buf);
  return len;
}
```

mbilsky, 2021-02-18:
Oddly it took me a while to re-create the bug this morning after having changed the way the camera worked to utilitize the RTOS approach from here (which seemed to fix everything and increase stability): https://www.hackster.io/anatoli-arkhipenko/multi-client-mjpeg-streaming-from-esp32-47768f

In both tests I am using the latest version of the camera code directly integrated into my source code in PlatformIO.

After reverting my code back and flashing a few times, I was able to get the bug again after launching the camera stream like before.

The fix seems to have prevented the hard crash. Now I sometimes get the following error in the serial debug console: 
[E][WiFiClient.cpp:392] write(): fail on fd 62, errno: 11, "No more processes"

I will keep this branch open in my code if you want me to play with some other things although I'm going to continue my dev using the technique listed above.

mbilsky, 2021-03-09:
The issue started to happen again, even with the new freeRTOS code.

After troubleshooting, I realized that updating to the espressif32 3.1 release broke the code. After reverting back to 3.0, the "No more processes" and/or hard crashes (depending on if I caught a failed malloc rather than new char).

It seems like it is somehow running out of ram despite the available heap size being greater than the desired allocation.

Still not sure what is actually causing it, only how to avoid pissing it off.

mbilsky, 2021-05-17:
@me-no-dev I tried upgrading to the new espressif32 3.2 release and the same bug appears to be there too :(