espressif/arduino-esp32 #2215: esp rebooting   need help for fix

flyballdogs, 2018-12-18:
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9220
load:0x40080400,len:6300
entry 0x400806a4
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
Saving settings
INFO: [WiFi] Starting in mode 0
assertion "Invalid mbox" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c", line 267, function: tcpip_callback_with_block
abort() was called at PC 0x400f94d7 on core 1

Backtrace: 0x4008d408:0x3ffb1d20 0x4008d639:0x3ffb1d40 0x400f94d7:0x3ffb1d60 0x40130c32:0x3ffb1d90 0x40142f71:0x3ffb1db0 0x4012dcdd:0x3ffb1df0 0x400dd0e9:0x3ffb1e30 0x400e6f3d:0x3ffb1e60 0x400e703d:0x3ffb1e90 0x400e7099:0x3ffb1ec0 0x400d2ce5:0x3ffb1f10 0x400d2fdf:0x3ffb1f60 0x4015f16e:0x3ffb1fb0 0x4008f655:0x3ffb1fd0

atanisoft, 2018-12-18:
please decode the backtrace

me-no-dev, 2018-12-18:
and post code... how could we know what you are doing?

flyballdogs, 2018-12-18:
you find The hole software on github 

https://github.com/vyruz1986/FlyballETS-Software/tree/feature/SyncStarting

lbernstone, 2018-12-18:
You need to isolate the code which is causing the error.

flyballdogs, 2018-12-18:
> 
> 
> please decode the backtrace

Decoding stack results
0x4008d408: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008d639: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400f94d7: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40130c32: tcpip_callback_with_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 267
0x40142f71: netconn_gethostbyname_addrtype at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/api_lib.c line 1054
0x4012dcdd: lwip_gethostbyname at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/netdb.c line 100
0x400dd0e9: WiFiUDP::beginPacket(char const*, unsigned short) (C:\Program Files at x86)\Arduino\libraries\WiFi\src\WiFiUdp.cpp line 168
0x400e6f3d: Syslog::_sendLog(unsigned short, char const*) at C:\Users\kevin\Documents\Arduino\libraries\Syslog-master\src\Syslog.cpp line 217
0x400e703d: Syslog::vlogf_P(unsigned short, char const*, __va_list_tag) at C:\Users\kevin\Documents\Arduino\libraries\Syslog-master\src\Syslog.cpp line 130
0x400e7099: Syslog::logf_P(char const*, ...) at C:\Users\kevin\Documents\Arduino\libraries\Syslog-master\src\Syslog.cpp line 172
0x400d2ce5: SetupWiFi() at C:\Users\kevin\Desktop\FlyballETS-Software-feature-SyncStarting working\FlyballETS-Software-feature-SyncStarting\FlyballETS-Software/WifiManager.ino line 32
0x400d2fdf: setup() at C:\Users\kevin\Desktop\FlyballETS-Software-feature-SyncStarting working\FlyballETS-Software-feature-SyncStarting\FlyballETS-Software/FlyballETS-Software.ino line 240
0x4015f16e: loopTask(void*) at C:\Users\kevin\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15
0x4008f655: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

closed 2018-12-24 by flyballdogs

kbickham, 2019-08-09:
Why is this closed? I cannot discern here what the issue was.

mvermand22, 2021-10-05:
Why did you close it? How did you solve it?

tablatronix, 2021-12-30:
wtf is this, man this is annoying ,

Sitting here writing code and all of a sudden this same thing 

No way to undo it, no idea whats causing it..