espressif/arduino-esp32 #2218: change pin on SD card adapter

steve-heslouin, 2018-12-18:
### Hardware:
Board:  TTGO Lora32 v1

### Description:
According this schema:
https://github.com/espressif/arduino-esp32/tree/master/libraries/SD/

But on this board i have, 
https://cdn-images-1.medium.com/max/1600/1*t9c7j4894YWCy5bBWkyhlQ.jpeg
there **is no GPIO 18, 19 and 5**, is there any way i can modify it to adapt to my free available GPIO, like **25, 34, 35** etc..?

For CS pin, its written SD.begin(CSpin), but for the others, i'm not quite sure

Thanks a lot

steve-heslouin, 2018-12-18:
editing type

atanisoft, 2018-12-18:
you can specify whichever pins you want by passing in a local SPI instance:
```
SPIClass sdSPI;
sdSPI.begin(SCK, MISO, MOSI, SS);
SD.begin(CSpin, sdSPI);
```

steve-heslouin, 2018-12-18:
Thanks a lot for your help, that's really a good news! 👍

closed 2018-12-18 by steve-heslouin