espressif/arduino-esp32 #2219: HTTPS Not working on ESP32

wezleysherman, 2018-12-18:
Hello, how do I properly do an HTTP GET request on an https link?

I've tried the following with no avail -- (I keep getting a response of -1)

http.begin("https url", "SHA1 fingerprint");
and
http.begin("https url", "raw rsa certificate");

I've even tried utilizing the tutorial here (following it exactly) with the same issue:
https://techtutorialsx.com/2017/11/18/esp32-arduino-https-get-request/

I know the tutorial is old -.. has something in the library changed? I've even tried having it connect to Google. Performing GET requests on regular http links work fine.

copercini, 2018-12-19:
This is probably a non enough heap problem...
You can get a board with PSram, and this will be magically solved or try optimize BLE and mbedtls to save resources and work together (requires moving to Arduino as IDF component)

wezleysherman, 2018-12-19:
Is there a quick and dirty fix for this? Can I adjust the heap size within the partition table?

What I’m doing right now is, I’m enabling BLE and polling for a few seconds, then I disable BLE and free the memory. After that I’m making my https request without any issues. The problem is enabling BLE again after I’ve made the HTTPS request — everything works fine up to where I’m creating the BLE Service (it hangs up here). Is there a way I can free the memory used by TLS after the https request is returned? 

I’ve just been resetting the ESP32 module via software afterwards to restore BLE.

chegewara, 2018-12-19:
@wezleysherman In case you described there may be problem with free heap (ram) or with ble library. What you can do is to check free internal memory just before your app stuck/crash.

czuvich, 2018-12-19:
@wezleysherman I run into the same issues on the WROVER module with PSRAM (4MB) when invoking HTTPS.  I think it's an issue with the internal mbed or wifi libraries.  I think BLE is actually fairly well-behaved (but RAM intense).  I really don't think extra heap will fix your problem.  Someone needs to dive in and diagnose the issue.

czuvich, 2018-12-19:
@wezleysherman Also, in regards to a dirty fix... if you can get away with resetting your board between BLE and WiFi, that should work.  In other words, when it's time to run WiFi set a flag and reset your board and run WiFi.  Once finished with WiFi, reset the flag and run BLE again.

wezleysherman, 2018-12-19:
@czuvich Thank you! That's the current way I'm doing it. It seems to work fine -.. My only concern is the 1-2 second delay before BLE is available again. 

This'll be something I look into when I move my code to use the ESP-IDF instead.

closed 2019-08-15 by stale[bot]